fwupd/fwupd #8833: [WIP Description] Unable to update Intel ME firmware of T14 Gen 1 (type 20S0, 20S1) Laptop (ThinkPad) - Type 20S1

ionutnechita, 2025-05-24:
- Unable to update Intel ME firmware of T14 Gen 1 (type 20S0, 20S1) Laptop (ThinkPad) - Type 20S1

Try to use to update ME firmware; this version: 14.1.75.2420.

Windows report:
```
#  ./csme-version-detection-tool-console.exe
Intel(R) CSME Version Detection Tool
Application Version: 7.0.2.0
Computer Name: BGD-THINK
Scan date: 24.05.2025 17:36:30

*** Host Computer Information ***
Manufacturer: LENOVO
Model: 20S1S5MX1J
Processor Name: Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz
OS Version: Microsoft Windows 11 Pro

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 12.0.93.2331
*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.

For more information refer to the Intel(R) CSME Version Detection Tool User Guide or the related Intel Security Advisory list at: https://www.intel.com/content/www/us/en/support/articles/000031784/technologies.html
Copyright(C) 2017-2022, Intel Corporation, All rights reserved.

# ./FwCapUpdate_v43.exe
FW Update Utility couldn't identify the system's SKU (Consumer or Corporate)

# ./FWDetect.exe
# echo $?
False

# ./MEFirmwareSKUCheckTool.exe
MEFW SKU: Error

# ./MEInfoWin64.exe
Intel (R) MEInfo Version: 14.1.75.2423
Copyright (C) 2005 - 2024, Intel Corporation. All rights reserved.

Error 624: Unsupported hardware platform. HW: Cannonlake Platform. Supported HW: Cometlake Platform.

```

hughsie, 2025-07-03:
What happens exactly when you try and update?

hughsie, 2025-07-04:
@mrhpearson have you ever seen this error before? It's a new one on me... Maybe image is refering to the ux-capsule perhaps?

jcharaoui, 2025-08-11:
I don't think my system is vPro but I admit I don't know how to check.

Here's the output of `fwupdmgr get-updates` in case that's helpful: https://paste.lib3.net/lavamind/2025-08-11-GmyhjVQZ30P6Xs5gn85HftX0znIgtKkObpJR0Mtl_t4/stdin.txt

jcharaoui, 2025-08-11:
I have to admit I haven't retried the upgrade in a while, and I see now there seems to have been a new release on 2025-07-25, so maybe I should just retry it now?

jcharaoui, 2025-08-11:
Tried again and it failed again.

The message was the same as I remembered, something about a "failed SBAT verification", I didn't have time to jot it down.

mrhpearson, 2025-08-11:
OK - I think this is a different problem. Please create a ticket and tag me on it