fwupd/fwupd #8834: Unable to update UEFI dbx on Dell Precision 5750 mobile workstation

tedli, 2025-05-25:
**Describe the bug**
I faced a very similar issue like this one https://github.com/fwupd/fwupd/issues/8710, on a Dell Precision 5750 mobile workstation with Ubuntu 24.04.02 LTS. Does the BaseboardProduct=0RV9F4 need to be added as well?

**Steps to Reproduce**

1. Tried using the firmware updater (the Gnome app), it said it just need to reboot but then the updates didn't get installed correctly.
2. Also tried to update by using the terminal command sudo fwupdmgr update --force -y, which was also said to reboot, but then the UEFI dbx were not updated as well, and the gnome notification (for updates) appeared again.

**Expected behavior**
UEFI dbx to be updated / installed correctly.

**fwupd version information**

```shell
fwupdtool get-report-metadata
```

```text
BatteryLevel:           100
BatteryThreshold:       10
BootTime:               1748140783
CompileVersion(com.hughsie.libjcat):0.2.0
CompileVersion(com.hughsie.libxmlb):0.3.18
CompileVersion(org.freedesktop.fwupd):1.9.29
CompileVersion(org.freedesktop.gusb):0.4.8
CpuArchitecture:        x86_64
CpuModel:               Intel Xeon W-10885M CPU @ 2.40GHz
DisplayState:           connected
DistroId:               ubuntu
DistroName:             Ubuntu
DistroPrettyName:       Ubuntu 24.04.2 LTS
DistroVersion:          24.04
FwupdSupported:         True
HostBaseboardManufacturer:Dell Inc.
HostBaseboardProduct:   RV9F4
HostBiosMajorRelease:   01
HostBiosMinorRelease:   23
HostBiosVendor:         Dell Inc.
HostBiosVersion:        1.35.0
HostEnclosureKind:      a
HostFamily:             Precision
HostFirmwareMajorRelease:ff
HostFirmwareMinorRelease:ff
HostProduct:            Precision 5750
HostSku:                990
HostVendor:             Dell Inc.
KernelName:             Linux
KernelVersion:          6.11.0-26-generic
LidState:               open
PowerState:             ac-fully-charged
RuntimeVersion(com.hughsie.libjcat):0.2.0
RuntimeVersion(com.hughsie.libxmlb):0.3.x
RuntimeVersion(org.freedesktop.fwupd):1.9.29
RuntimeVersion(org.freedesktop.fwupd-efi):1.4
RuntimeVersion(org.freedesktop.gusb):0.4.8
RuntimeVersion(org.kernel):6.11.0-26-generic
DeviceId:               863e54366e53dd2e565b5165100b745ac8b59171
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x0
DeviceId:               bcb9e7b6f2d29fd470f490246c15e277a384c33a
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
[... 855 lines cut ...]
│       └── zstd.mod
├── initrd.img -> initrd.img-6.11.0-26-generic
├── initrd.img-6.11.0-26-generic
├── initrd.img.old -> initrd.img-6.11.0-26-generic
├── lost+found
├── memtest86+ia32.bin
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── System.map-6.11.0-26-generic
├── vmlinuz -> vmlinuz-6.11.0-26-generic
├── vmlinuz-6.11.0-26-generic
└── vmlinuz.old -> vmlinuz-6.11.0-26-generic

12 directories, 311 files
```

**Additional questions**

- Operating system and version: Ubuntu 24.04.2 LTS
- Have you tried rebooting? Yes.
- Is this a regression? Not sure, got this machine recently.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? Yes.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

elvisdukaj, 2025-06-02:
Same issue with Dell Precision 7670 and Ubuntu 24.04. My understanding is how Intel changed the versioning scheme for the dbx.

Similar issue can be also found [here](https://github.com/fwupd/fwupd/issues/8696)

hughsie, 2025-06-02:
I'd be surprised if it failed for Dell; are you using the distro package or the snap build? I wonder if it would work for you using something like Fedora.

tedli, 2025-06-02:
Hi @hughsie ,

Thanks for reply.

> are you using the distro package or the snap build?

Both tried, without succees. (Actually shown succeeded, but after reboot, It still notices that need to update.)

> I wonder if it would work for you using something like Fedora.

Neither. (I tried fedora first. Difference from ubuntu, fedora shows failure directly.)

---

**TL; DR**

I've been using windows for years. Disk full. I bought new disk. And turn to use linux. I noticed that, **Fedora** recently released Workstation 42, so I tried Fedora first. However, with fedora 42, cuda and nvidia driver didn't get happy with it.

[image]

So, I turned to use Ubuntu, every thing works as expect, except this dbx uptating alert.

And, there was a said story, I lost my laptop last friday 😭. So it's not possible for me to confirm this issue anymore. If others didn't hit this same issue, feel free to close.