fwupd/fwupd #8844: DBX Update fails on HP Elitedesk G6

sicherist, 2025-05-28:
**Describe the bug**
For the last few months I keep getting a message "UEFI dbx can be upgraded from version 20230501 to 20241101". If I install, the message re-appears after the next reboot.

**Steps to Reproduce**
In my case: Open KDE-updater-app and install latest UEFI dbx update.

**Expected behavior**
Update should be installed or not offered if not applicable.

**fwupd version information**
compile   org.freedesktop.fwupd         1.9.29
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.29
runtime   org.kernel                    6.11.0-26-generic

```shell
fwupdtool get-report-metadata
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       50
BootTime:               1748416218
CompileVersion(com.hughsie.libjcat):0.2.0
CompileVersion(com.hughsie.libxmlb):0.3.18
CompileVersion(org.freedesktop.fwupd):1.9.29
CompileVersion(org.freedesktop.gusb):0.4.8
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 5 PRO 4650GE with Radeon Graphics
DisplayState:           connected
DistroId:               ubuntu
DistroName:             Ubuntu
DistroPrettyName:       Ubuntu 24.04.2 LTS
DistroVersion:          24.04
FwupdSupported:         True
HostBaseboardManufacturer:HP
HostBaseboardProduct:   872C
HostBiosMajorRelease:   12
HostBiosMinorRelease:   00
HostBiosVendor:         HP
HostBiosVersion:        S29 Ver. 02.18.00
HostEnclosureKind:      23
HostFamily:             103C_53307F HP EliteDesk
HostFirmwareMajorRelease:09
HostFirmwareMinorRelease:08
HostProduct:            HP EliteDesk 805 G6 Desktop Mini PC
HostSku:                21L52EA#ABD
HostVendor:             HP
KernelName:             Linux
KernelVersion:          6.11.0-26-generic
LidState:               unknown
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat):0.2.0
RuntimeVersion(com.hughsie.libxmlb):0.3.x
RuntimeVersion(org.freedesktop.fwupd):1.9.29
RuntimeVersion(org.freedesktop.fwupd-efi):1.4
RuntimeVersion(org.freedesktop.gusb):0.4.8
RuntimeVersion(org.kernel):6.11.0-26-generic
DeviceId:               b25c7b6ea3573c266acd835f8bb581071663baa8
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
[... 599 lines cut ...]
├── System.map-6.8.0-58-generic
├── System.map-6.8.0-59-generic
├── System.map-6.8.0-60-generic
├── vmlinuz -> vmlinuz-6.11.0-26-generic
├── vmlinuz-6.11.0-25-generic
├── vmlinuz-6.11.0-26-generic
├── vmlinuz-6.8.0-54-generic
├── vmlinuz-6.8.0-55-generic
├── vmlinuz-6.8.0-56-generic
├── vmlinuz-6.8.0-57-generic
├── vmlinuz-6.8.0-58-generic
├── vmlinuz-6.8.0-59-generic
├── vmlinuz-6.8.0-60-generic
└── vmlinuz.old -> vmlinuz-6.11.0-25-generic

```

**Additional questions**

- Operating system and version: Kubuntu 24.04.2
- Have you tried rebooting? Yes
- Is this a regression?
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no

hughsie, 2025-06-06:
It looks like updating the dbx is broken on that model -- could you prepare a pull request like https://github.com/fwupd/fwupd/pull/8824 that add it to the ignore list please?

sicherist, 2025-06-07:
@hughsie is there a way to let fwupdmgr ignore this specific update? I'm afraid it will take years for this patch to reach Ubuntu LTS.

sicherist, 2025-06-12:
Just for future reference. To disable dbx plugin in ubuntu, edit `/etc/fwupd/fwupd.conf` under the section `[fwupd] ` add the line ` DisabledPlugins=uefi_dbx`