fwupd/fwupd #8861: 🐞 Bug: UEFI dbx Update Fails on Dell Latitude 3500 (Even with Latest BIOS and Secure Boot Enabled)

Anjeho, 2025-06-03:
**Describe the bug**  
The Secure Boot dbx update (`20241101`) from LVFS is repeatedly offered but never successfully applies on a Dell Latitude 3500, even after updating to the latest BIOS (`1.38.0`) and enabling Secure Boot. After each attempted update and reboot, the system continues to report dbx version `20230501`, and GNOME Software and `fwupdmgr` keep re-offering the update.

**Steps to Reproduce**  
1. Install openSUSE Aeon on a Dell Latitude 3500.
2. Update BIOS to latest (1.38.0) via Dell's F12 boot flash menu.
3. Enable Secure Boot in UEFI setup.
4. Run `fwupdmgr update` to apply Secure Boot dbx update (`20241101`).
5. Reboot.
6. Check `fwupdmgr get-history`.

**Expected behavior**  
The dbx version should be upgraded to `20241101` after reboot, and the update should no longer be offered.

**fwupd version information**  
Installed by default on openSUSE Aeon.

```shell
fwupdtool get-report-metadata
compile   info.libusb                   1.0.28
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.9
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.6-1-default
runtime   org.freedesktop.fwupd         2.0.9
```

Install method: bundled with openSUSE Aeon

<details>
<summary>**fwupd device information**</summary>

```shell
System Firmware
  Current version: 1.38.0
  Update State: Success

UEFI dbx:
  Current version: 20230501
  Update Error: failed to run update on reboot: expected 20241101 and got 20230501
```
</details>

**System UEFI configuration**

```shell
efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0001
Boot0000* openSUSE Boot Manager	HD(...) /File(\EFI\systemd\shim.efi)
Boot0001* UEFI BC501 NVMe SK hynix 256GB ... /File(\EFI\Boot\BootX64.efi){auto_created_boot_option}
```

```shell
efivar -l | grep -i fw
efivar: command not found on openSUSE Aeon by default
```

```shell
ls -R /boot
/boot:
efi  mbrid

/boot/efi:
aeon  EFI  loader

/boot/efi/aeon:
6.14.6-1-default

/boot/efi/aeon/6.14.6-1-default:
initrd-...  linux-...

/boot/efi/EFI:
BOOT  dell  systemd

/boot/efi/EFI/BOOT:
BOOTX64.EFI  fallback.efi  MokManager.efi

/boot/efi/EFI/dell/bios/recovery:
BIOS_CUR.rcv

/boot/efi/EFI/systemd:
shim.efi  grub.efi  MokManager.efi ...

/boot/efi/loader/entries:
aeon-6.14.6-1-default-1.conf
aeon-6.14.6-1-default-2.conf
```

**Additional questions**

- Operating system and version: openSUSE Aeon (current as of June 2025)
- Have you tried rebooting? Yes, multiple times
- Is this a regression? Unknown — issue exists persistently
- Are you using an NVMe disk? Yes, SK hynix BC501 NVMe 256GB
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No, this is a Dell Latitude 3500

hughsie, 2025-06-03:
If you clear your firmware settings back to factory defaults does the dbx then apply?

Anjeho, 2025-06-04:
I’ve reset the BIOS to factory defaults as suggested, re-enabled Secure Boot, and re-applied the update, but the dbx update still fails to apply. It continues to report version 20230501 after every reboot. At this point, I believe it’s a firmware issue with this Dell model.

stargladius, 2025-06-27:
Hello.
I have the same issue on my Dell Latitude 5401. My OS is Ubuntu 22.04.
Will be glad to add any additional info if needed, just don't know what exactly.
As you can see Firmware for UEFI get updated today without any problems with reboot and flashing. But this DBX is still in avaliable.

**sudo fwupdmgr get-history** 
```

Latitude 5401
│
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Previous version:   1.39.0
│ │   Update State:       Success
│ │   Last modified:      2025-06-27 08:58
│ │   GUID:               0eecff0c-95b8-4ade-9717-8f4f3edc9e09
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─Latitude 5X01/Precision 3541:
│       New version:      1.40.1
│       Remote ID:        lvfs
│       Release ID:       114456
│       Summary:          Firmware for the Dell Latitude 5X01/Precision 3541
│       License:          Proprietary
│       Size:             23,1 МБ
│       Created:          2025-04-20
│       Urgency:          Critical
│       Vendor:           Dell
│       Description:      
│       This stable release fixes the following issues:
│       
│       • This release contains security updates as disclosed in the Dell Security Advisories
│     
└─UEFI dbx:
  │   Device ID:          362301da643102b9f38477387e2193e57abaa590
  │   Previous version:   77
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2025-06-27 09:05
  │   GUID:               00fe3755-a4d8-5ef7-ba5f-47979fbb3423
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │ 
  └─Secure Boot dbx:
        New version:      20241101
        Remote ID:        lvfs
        Release ID:       108324
        Summary:          UEFI Secure Boot Forbidden Signature Database
        Variant:          x64-compat
        License:          Proprietary
        Size:             23,3 кБ
        Created:          2023-05-09
        Urgency:          High
        Vendor:           Linux Foundation
        Duration:         1 second
        Release Flags:    • Is upgrade
        Description:      
        This updates the list of forbidden signatures (the "dbx") to the latest release from Microsoft.
        
        An insecure version of Howyar's SysReturn software was added, due to a security vulnerability that allowed an attacker to bypass UEFI Secure Boot.
```

**sudo fwupdmgr get-updates**
```
 
Devices with no available firmware updates: 
 • 0000:00:1f.5
 • TPM 2.0
Devices with the latest available firmware version:
 • Micron 2200S NVMe 256GB
 • System Firmware
Latitude 5401
│
└─UEFI dbx:
  │   Device ID:          362301da643102b9f38477387e2193e57abaa590
  │   Summary:            UEFI revocation database
  │   Current version:    466
  │   Minimum Version:    466
  │   Vendor:             UEFI:Linux Foundation
  │   Install Duration:   1 second
  │   GUID:               00fe3755-a4d8-5ef7-ba5f-47979fbb3423 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9
  │                       4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
  │                       c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
  │                       f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Only version upgrades are allowed
  │                       • Signed Payload
  │ 
  └─Secure Boot dbx:
        New version:      20241101
        Remote ID:        lvfs
        Release ID:       108324
        Summary:          UEFI Secure Boot Forbidden Signature Database
        Variant:          x64-compat
        License:          Proprietary
        Size:             23,3 кБ
        Created:          2023-05-09
        Urgency:          High
        Vendor:           Linux Foundation
        Duration:         1 second
        Release Flags:    • Is upgrade
        Description:      
        This updates the list of forbidden signatures (the "dbx") to the latest release from Microsoft.
        
        An insecure version of Howyar's SysReturn software was added, due to a security vulnerability that allowed an attacker to bypass UEFI Secure Boot.
```

stargladius, 2025-07-25:
This was resolved in a newer version
And we still have old versions in our distributions.
So I suggest closing this request and addressing it to the distro maintainers.
For example with Ubuntu:
https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/2028548