fwupd/fwupd #8879: FATAL-WARNING: no PlatformArchitecture in metadata

davidlt, 2025-06-06:
**Describe the bug**
Building fwupd-2.0.11 on Fedora/RISCV (riscv64) and one test fails.

```
[..]
# DEBUG: PlatformArchitecture=(null)
# DEBUG: PowerState=unknown
# DEBUG: RuntimeVersion(com.hughsie.libjcat)=0.2.3
# DEBUG: RuntimeVersion(com.hughsie.libxmlb)=0.3.22
# DEBUG: RuntimeVersion(org.freedesktop.fwupd)=2.0.11
# DEBUG: RuntimeVersion(org.kernel)=6.6.93-200.eswin.el10.riscv64
# DEBUG: SELinux=permissive
not ok /fwupd/engine{report-metadata} - FATAL-WARNING: no PlatformArchitecture in metadata
Bail out!
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
34/34 fu-self-test                   OK               0.04s
Summary of Failures:
33/34 fu-self-test                   FAIL             8.02s   killed by signal 5 SIGTRAP
Ok:                 33  
Expected Fail:      0   
Fail:               1   
Unexpected Pass:    0   
Skipped:            0   
Timeout:            0   
Full log written to /builddir/build/BUILD/fwupd-2.0.11-build/fwupd-2.0.11/redhat-linux-build/meson-logs/testlog.txt
```

**Steps to Reproduce**
Run tests (`/usr/bin/meson test -C redhat-linux-build --num-processes 4 --print-errorlogs`)

**Expected behavior**
The test should not fail. I would assume PlatformArchitecture should not be null.

**fwupd version information**
2.0.11

</details>

**Additional questions**

- Operating system and version: Fedora 42.
- Have you tried rebooting? No, but it was built (and failed) on multiple boards.
- Is this a regression? Yes. The tests used to pass on riscv64.

davidlt, 2025-06-06:
Looking around, and I almost forgot that on RISC-V `AT_BASE_PLATFORM` and `AT_PLATFORM` are NULL.

davidlt, 2025-06-06:
Note  that the man page (getauxval(3)) says `Not all type values are present on all architectures.`

hughsie, 2025-06-06:
> There is LPC talk from 2021 that just confirms that those are NULL for riscv

Right; I think we need some way to set the PlatformArchitecture -- otherwise other things will break. I think it would even be fine to have a _ifdef RISCV //add metadata manually endif_ block.