fwupd/fwupd #8890: Unable to upgrade Thunderbolt host controller firmware 20->23 on ThinkPad T480

eloydegen, 2025-06-07:
**Describe the bug**
`fwupdmgr upgrade` prints `release version format 'pair' but no device version format` instead of upgrading my Thunderbolt host controller firmware, even after multiple reboots. I use a Lenovo ThinkPad T480 with Fedora Linux 42, initially installed version 36.

**Steps to Reproduce**
`fwupdmgr upgrade`

**Expected behavior**
I expected to be able to update my Thunderbolt host controller firmware and not get the error `release version format 'pair' but no device version format`

**fwupd version information**
Please provide the version of the daemon and client.

```shell
Loading…                 [******************                     ]12:11:11.913 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/host3/target3:0:0/3:0:0:0/block/sdb: failed to subclass open: failed to open /dev/sdb: No medium found
12:11:11.920 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/host3/target3:0:0/3:0:0:0/block/sdb: failed to subclass open: failed to open /dev/sdb: No medium found
Loading…                 [*******************                    ]12:11:11.935 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:16.0/mei/mei0: failed to setup: generic failure [0xb]
Waiting…                 [*************************************  ]
BatteryLevel:           75
BatteryThreshold:       25
BootTime:               1749297256
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.Passim): 0.1.10
CompileVersion(org.freedesktop.fwupd): 2.0.11
CpuArchitecture:        x86_64
CpuModel:               Intel Core™ i5-8350U CPU @ 1.70GHz
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 42 (Workstation Edition)
DistroVariant:          workstation
DistroVersion:          42
FwupdSupported:         True
HostBaseboardManufacturer: LENOVO
HostBaseboardProduct:   20L6SA5Q00
HostBiosMajorRelease:   01
HostBiosMinorRelease:   36
HostBiosVendor:         LENOVO
HostBiosVersion:        N24ET79W (1.54 )
HostEnclosureKind:      a
HostFamily:             ThinkPad T480
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 16
HostProduct:            20L6SA5Q00
HostSku:                LENOVO_MT_20L6_BU_Think_FM_ThinkPad T480
HostVendor:             LENOVO
KernelName:             Linux
KernelVersion:          6.14.9-300.fc42.x86_64
LidState:               open
PassimDownloadSaving:   0
PlatformArchitecture:   x86_64
PowerState:             ac
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.11
RuntimeVersion(org.freedesktop.fwupd-efi): 1.6
RuntimeVersion(org.kernel): 6.14.9-300.fc42.x86_64
SELinux:                enforcing
DeviceId:               f2a0ebaf16bf5280a057b882965430a4770b70cf
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               5d3115dfafc3359e6be9e1954a28e39b7030db24
  pre:
[... 282 lines cut ...]
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
      Device Requests:    • Message
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • Intel Management Engine (184.93.4323 → 184.94.4494)
 • System Firmware (0.1.46 → 0.1.50)
 • UEFI dbx (20220801 → 20241101)
 • KEK CA (2011 → 2023)
 • Intel Management Engine (184.94.4494 → 184.96.4657)
 • System Firmware (0.1.50 → 0.1.54)

```

</details>

**Additional questions**

- Operating system and version: Fedora Linux 42
- Have you tried rebooting? Yes
- Is this a regression? Not sure

hughsie, 2025-06-19:
2.0.12 was released today -- and I've built into the fedora and the copr.

closed 2025-06-19 by hughsie

efirshik, 2025-08-16:
same problem on debian trixie, any chance update to 2.0.12 ?

hughsie, 2025-08-17:
@efirshik you need to ask Debian to backport.