fwupd/fwupd #8908: Lenovo T14 Gen2i updates don't complete

pprkut, 2025-06-11:
**Describe the bug**

I can't get fwupd to complete any firmware update to complete (of the 3 that it offers me currently). I run `fwupdmgr update` and that seems to all work fine, but something with the handover to fwupd-efi doesn't seem to be working. When I reboot I see this:

```
fwupd-efi version 1.6
WARNING: Could not get variable "LastMemoryMapInformation": Not Found
WARNING: Could not populate update info for "LastMemoryMapInformation"
WARNING: Could not find updates: Not Found
```
followed by
```
error: no suitable video mode found
```

I also tried with fwupd-efi 1.7, but that only says that there is no update to process and continues without doing anything after 10 seconds.

Possibly the same as/similar to https://github.com/fwupd/firmware-lenovo/issues/505

**Steps to Reproduce**
I really only run `fwupdmgr update` and then reboot.

**Expected behavior**
No error messages from fwupd-efi and completed firmware update

**fwupd version information**
Please provide the version of the daemon and client.

<details>

<summary>fwupdtool get-report-metadata</summary>

```
BatteryLevel:           95
BatteryThreshold:       25
BootTime:               1749665685
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.29
CompileVersion(org.freedesktop.fwupd): 2.0.11
CpuArchitecture:        x86_64
CpuModel:               Intel 11th Gen Intel Core™ i7-1185G7 @ 3.00GHz
DisplayState:           connected
DistroId:               slackware
DistroName:             Slackware
DistroPrettyName:       Slackware 15.0 x86_64 (post 15.0 -current)
DistroVersion:          15.0
FwupdSupported:         True
HostBaseboardManufacturer: LENOVO
HostBaseboardProduct:   20W0CTO1WW
HostBiosMajorRelease:   01
HostBiosMinorRelease:   39
HostBiosVendor:         LENOVO
HostBiosVersion:        N34ET57W (1.57 )
HostEnclosureKind:      a
HostFamily:             ThinkPad T14 Gen 2i
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 2a
HostProduct:            20W0CTO1WW
HostSku:                LENOVO_MT_20W0_BU_Think_FM_ThinkPad T14 Gen 2i
HostVendor:             LENOVO
KernelCmdline:          psmouse.synaptics_intertouch=1 ipv6.disable=1
KernelName:             Linux
KernelVersion:          6.12.33
LidState:               open
PlatformArchitecture:   x86_64
PowerState:             ac
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.11
[... 971 lines cut ...]
├── make-kernel-backup
├── onlyblue.bmp
├── onlyblue.dat
├── remove-orphaned-initrds
├── slack.bmp
├── tuxlogo.bmp
├── tuxlogo.dat
├── vmlinuz -> vmlinuz-6.12.33
├── vmlinuz-5.14.17
├── vmlinuz-6.12.33
└── vmlinuz-generic -> vmlinuz-6.12.33

15 directories, 447 files
```

</details>

**Additional questions**

- Operating system and version: Slackware64 current
- Have you tried rebooting? Yes
- Is this a regression? Yes, although I can't remember when it was last working
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Not sure whether that is turned on

hughsie, 2025-06-11:
Can you try installing 1.7 and _then_ reschedule the update please.

hughsie, 2025-06-12:
No updates to process means that the UEFI variable we created was cleared -- perhaps by the system firmware. We don’t do any testing on Slackware -- does the update work correctly if you use Ubuntu or Fedora?

pprkut, 2025-06-12:
Yes, that was my assumption too. I'm not really in a position where I could easily/quickly test with Ubuntu/Fedora. I would have to try that on the weekend perhaps.

I checked our init scripts to see if there's anything obvious touching the UEFI variables, but I can't see anything. I'll check with the rest of the slackware team if anyone perhaps knows something. Would you have any pointers to look out for? Or suggestions on what could help me debug this?

mariolpantunes, 2025-10-15:
Using debian I am getting the same result.