fwupd/fwupd #8909: failed to write data to efivarfs:... invalid argument...

DjDiabolik, 2025-06-11:
**Describe the bug**
It's appear a notification need to be update my dbx 20241101 -> 20250507

**Steps to Reproduce**
from terminal:
```
diabolik@NUC-Ubuntu:~$ fwupdmgr update
Devices with no available firmware updates: 
 • HS-SSD-E100 512G
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 20241101 to 20250507?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ Some insecure versions of BiosFlashShell and Dtbios by DT Research Inc were  ║
║ added, due to a security vulnerability that allowed an attacker to bypass    ║
║ UEFI Secure Boot.                                                            ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Scrittura…               [                                       ]
failed to write data to efivarfs: Errore nello scrivere sul descrittore di file: Argomento non valido
diabolik@NUC-Ubuntu:~$ 
```

**Expected behavior**
i don't have idea.. probably request to restart and successfull install ?

**fwupd version information**
```
diabolik@NUC-Ubuntu:~$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         1.9.29
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-60-generic
runtime   org.freedesktop.fwupd         1.9.29
```

```shell
fwupdtool get-report-metadata:
```
diabolik@NUC-Ubuntu:~$ fwupdtool get-report-metadata
Caricamento…             [*                                      ]
Failed to load AppStream data: cannot create metadata.xmlb: Errore nell'aprire il file «/var/cache/fwupd/metadata.xmlb»: Permesso negato
WARNING: This program may only work correctly as root
```
I need to run it as sudo ?

fwupdtool hwids:
```
diabolik@NUC-Ubuntu:~$ fwupdtool hwids
Caricamento…             [*                                      ]
Computer Information
--------------------
BiosVendor: Intel Corp.
BiosVersion: JYGLKCPX.86A.0071.2023.0825.1034
Manufacturer: Intel(R) Client Systems
Family: JY
ProductName: NUC7PJYHN
ProductSku: BOXNUC7PJYHN
EnclosureKind: 35
BaseboardManufacturer: Intel Corporation
BaseboardProduct: NUC7JYB
[... 195 lines cut ...]
Comando «tree» non trovato, ma può essere installato con:
sudo snap install tree  # version 2.1.3+pkg-5852, or
sudo apt  install tree  # version 2.1.1-2
Per ulteriori versioni, consultare «snap info tree».
diabolik@NUC-Ubuntu:~$ 
```

```

**Additional questions**

- Operating system and version: 
Ubuntu 24.04.2 LTS
- Have you tried rebooting? 
Yes... also after sudo apt update and upgrade procedure.
- Is this a regression? 
I don't have idea
- Are you using an NVMe disk? 
No.. It's a SSD installed inside a NUC Intel
- Is secure boot enabled? 
I don't have idea if yes.. how i can check this ?
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? 
No.. It's a INTEL NUC.. probably info of hardware can found upper here

Thanks in advance for fix or suggestion...

hughsie, 2025-06-12:
Please see this: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-dbx-efivarfs-IO-error

DjDiabolik, 2025-06-12:
Also because honestly my mistake seems different:
Here it's say "Invalid argument" and not "Error writing to file descriptor"

hughsie, 2025-06-12:
> It's say "resetting the system firmware" but what does it mean exactly what is there to reset?

I think you need to do this yourself dude; there are 20,000 dbx updates going to end users per hour at the moment.

DjDiabolik, 2025-06-12:
I have, right now also tryed to disable "Secure boot" from bios... reboot my PC but same error remain.... 🤦

robhancocksed, 2025-06-12:
@hughsie Out of a sample size of 3 Linux machines I have tried these recent UEFI DBX/KEK updates on, 2 (one Asus, one Lenovo) have run into these "invalid argument" errors. So if this is due to firmware bugs, they have to be pretty widespread. It's annoying to the user as it keeps popping up with update notifications which then fail to install. Seems like this needs some more investigation..

DjDiabolik, 2025-06-12:
I mean friends let's understand each other... in march 2025 i have, on same exact pc, need to use fwupdmgr to update something similar and speak inside this italian forum here:
https://forum.ubuntu-it.org/viewtopic.php?p=5387118#p5387118

on that precise occasion I had installed the version.. 20241101

DjDiabolik, 2025-06-13:
on ubuntu.it forum they suggested me to try reporting the problem in launchpad and here it is :) :

https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/2114511

TC1977, 2025-06-14:
I was able to clear this error just now by going into my machine's UEFI screen at boot, going to the Key Management menu, and restoring the factory `dbx` only.

After exiting and booting, I was then able to use `sudo fwupdmgr update` to update from the original factory `dbx` all the way up to the newest version. (I also disabled Secure Boot initially, and the re-enabled it, but not sure that was necessary.)

I'm running Ubuntu 24.04.2 on an Intel Celeron N4000. No NvME drive. Hope this helps.

```
AMI Intel
│
└─UEFI dbx:
  │   Device ID:          362301da643102b9f38477387e2193e57abaa590
  │   Previous version:   20160809
  │   Install Duration:   3 seconds
  │   Update State:       Success
  │   Last modified:      2025-06-14 17:59
  │   GUID:               c4d67ee2-05f1-56dd-8ea3-ff7a558a375d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │                       • Only version upgrades are allowed
  │                       • Signed Payload
  │ 
  └─Secure Boot dbx Configuration Update:
        New version:      20250507
        Remote ID:        lvfs
        Release ID:       115586
        Summary:          UEFI Secure Boot Forbidden Signature Database
        Variant:          x64
        License:          Proprietary
        Size:             24.0 kB
        Created:          2025-01-17
        Urgency:          High
        Tested by Linux Foundation:
          Tested:         2025-06-11
          Distribution:   fedora 42 (workstation)
          Old version:    20241101
          Version[fwupd]: 2.0.11
        Vendor:           Linux Foundation
        Duration:         1 second
        Release Flags:    • Trusted metadata
                          • Tested by trusted vendor
        Description:      
        This updates the list of forbidden signatures (the "dbx") to the latest release from Microsoft.
        
        Some insecure versions of BiosFlashShell and Dtbios by DT Research Inc were added, due to a security vulnerability that allowed an attacker to bypass UEFI Secure Boot.
        Issues:           806555
                          CVE-2025-3052
        Checksum:         8b1efdd1ae2ae86b7a3d611570a4c02d644710e527b6b78917e8782aa3453166
```

TC1977, 2025-06-15:
Quite possible. I'll note that like OP, I only had about 25k free in my `/sys/firmware/efi/efivars` partition. And I believe the `dbx` is about 23k.

DjDiabolik, 2025-06-16:
@superm1 e @TC1977 ok but how i can free the space in the efivars ??? the actual dbx it's about 32kb and i have posted the info picture also here:
so currently it's:
```
diabolik@NUC-Ubuntu:~$ df -h
File system     Dim. Usati Dispon. Uso% Montato su
efivarfs        128K  102K     22K  83% /sys/firmware/efi/efivars

```

Available it's 22K.... the cab it's 30.5K... how i can fix ?

Also on italian ubuntu forum another user has resolve the issue resetting the secureboot keys in bios.
On bios i have this setting:

[image]

If you have some suggestion thanks in advance..

DjDiabolik, 2025-06-16:
And why my efivars it's only 128K instead 192K ? When i install ubuntu i need to increase this size manually ?

hughsie, 2025-06-16:
> And why my efivars it's only 128K instead 192K

It depends on how the OEM configured the NVRAM space for the platform. I think resetting the firmware to factory defaults might help, as that seems to "defragment" the NVRAM store. Also note, some software like systemd can store things like crash dumps in the NVRAM, so maybe check for those in /sys/firmware/efi/efivars/

DjDiabolik, 2025-06-16:
@hughsie as already explained in the past I can't understand what you mean when you talk about restoring the "factory settings" without specifying where and how.

So i have to ask ? It's the OS the issue ? I have to made a new install of ubuntu or what else ? I've never done any weird stuff in the OS though.

@TC1977 touching on BIOS setting i have made this right now:

[image]

This cause this at next time of ubuntu:
diabolik@NUC-Ubuntu:~$ df -h
File system     Dim. Usati Dispon. Uso% Montato su
efivarfs        128K   74K     50K  60% /sys/firmware/efi/efivars

So right now i the space on efivars it's bigger than before. But right now `fwupdmgr` cannot find any updates to install.

I ask if we "did" the damage on entire os :) right now.. i hope no.

TC1977, 2025-06-16:
So the questions you’re asking are related to the BIOS (actually UEFI) on your particular computer. They don’t have anything to do with fwupd or with Ubuntu. I suggest you google your particular motherboard model, or the manufacturer, and look for BIOS settings and how to reset Secure Boot. 

If I were to guess blindly, my next step would be to check “Force Secure Boot Defaults” (which may be the equivalent of restoring factory keys on my computer), uncheck “Secure Boot Mode”, and proceed.

The good news is none of this has anything to do with Ubuntu or fwupd. Once you get some kind of dbx installed, if what I did works for your computer, then simply running fwupdmgr refresh —force and then fwupdmgr update will do the job.

DjDiabolik, 2025-06-16:
> So the questions you’re asking are related to the BIOS (actually UEFI) on your particular computer. They don’t have anything to do with fwupd or with Ubuntu. I suggest you google your particular [...]

@TC1977 yes that the fault was not of fwupd in itself I think it was clear for a while. and I think in any case I also understood what was happening and I'm also sure that the same problem will happen again with the next update.

So look at that... i have resolved my issue right now look the output of this command:
```
diabolik@NUC-Ubuntu:~$ fwupdmgr get-devices
Intel(R) Client Systems NUC7PJYHN
│
├─GeminiLake [UHD Graphics 605]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    06
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               6f398ef1-6999-553b-822a-e5798e7d3809 ← PCI\VEN_8086&DEV_3184
│                         7e675f13-a2d7-52fa-b886-05280bb915a6 ← PCI\VEN_8086&DEV_3184&SUBSYS_80862072
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─HS-SSD-E100 512G:
│     Device ID:          c46944fe84f2b057692258cc87a812cc86364881
│     Summary:            ATA drive
│     Current version:    030fAA20
│     Serial Number:      30094593359
│     Update Error:       No vendor ID set
│     GUID:               0dc6e288-2ed0-5ef3-8c9f-f2eded1c7c1b ← IDE\HS-SSD-E100_512G________________________030fAA20
│                         9022c257-5da3-5c5a-a070-e717ec4b185d ← IDE\0HS-SSD-E100_512G________________________
│                         372f0676-1fad-5401-b708-00c85066c469 ← HS-SSD-E100 512G
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─Pentium Silver J5040 CPU @ 2.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000026
│     Vendor:             Intel
│     GUID:               54817722-b948-55a7-8324-3d4736e90181 ← CPUID\PRO_0&FAM_06&MOD_7A
│                         f02ec022-4f79-53c2-a68f-08ea48563cb4 ← CPUID\PRO_0&FAM_06&MOD_7A&STP_8
│     Device Flags:       • Internal device
│   
├─System Firmware:
│ │   Device ID:          b4ffc2818515d7df26477b55caa461b59dee8205
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    71
│ │   Minimum Version:    71
│ │   Vendor:             Intel(R) Client Systems (DMI:Intel Corp.)
│ │   Update State:       Success
│ │   GUID:               856758ee-a971-40c7-8ee4-26d67b4f6704
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database 6b4328eb
│       Current version:  20250507
│       Minimum Version:  20250507
│       Vendor:           UEFI:Microsoft
│       Install Duration: 1 second
│       GUID:             f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
[... 30 lines cut ...]

at that point, however, if secureboot was active from Bios, from Bios it was not working or not active.
Reading information online I found that I had to, once again, return to BIOS and complete the installation of the Intel keys using the option "Install Intel Platform Key" PKPub ! Save again bios setting and booting again on ubuntu! Now secureboot correctly enabled but again the efivarfs partition it has returned to being essentially full.

So.... i have also an possibly explain..... it's a suggestion by ChatGPT/CoPilot info.

Copilot states that bios prior to 2023 in most cases had this limit of 128Kb for this efivarfs variable.

Many bios developed after 2024 instead present in the changelogs that have increased this sort of cache to 192Kb and therefore this precisely to guarantee the installation of systems that require more space for that partition.

So infact look at right now:
diabolik@NUC-Ubuntu:~$ df -h
File system     Dim. Usati Dispon. Uso% Montato su
efivarfs        128K  101K     23K  82% /sys/firmware/efi/efivars

On next update this BUG/LIMIT it's be back on many system like me Intel NUC and so many other mainboard.....

Expecially because it's not an invalid argument but simply NO AVAILABLE SPACE...... Need to fix.

The question it's:
Fwupgr can be fix to be made a better installation/upgrade ? Why i need to reset the space manually from bios and not fwupgr not overwrite correctly the old uefi dbx database ?

So other question it's... if this eufi dbx become most bigger then my current 128Kb it's pratically unusable on most device like my current Intel NUC model.

So.. for my Intel NUC there's no exist a newest bios over my current and already installed.... if you know try to suggest like subreddit or fanmade custom bios whit manual hotfix or similar.

robhancocksed, 2025-06-16:
One of the machines that is having issues with these UEFI updates is a Lenovo ThinkCentre M720s. I was able to successfully upgrade the UEFI dbx on this machine, but there is also a Lenovo UEFI KEK update that refuses to install with the "failed to write data to efivarsfs: Error writing to file descriptor: Invalid argument" error. That machine seems to have plenty of efivarfs space available:

```
Filesystem      Size  Used Avail Use% Mounted on
efivarfs        320K   74K  242K  24% /sys/firmware/efi/efivars

```

DjDiabolik, 2025-06-16:
@robhancocksed but you can check, for a purpose test, the filesize of .cab ? It's have a bigger size than 242K after decompressed ?

robhancocksed, 2025-06-16:
For that update, the .cab file contains a bunch of .bin files, but each of them seems to be at most 3.8KB, so there should be plenty of room?

hughsie, 2025-06-17:
> no one seems to know how to help for each specific situation

The IBVs don't tell us anything and we can't see the source code. We get back a value of "I/O Error" -- nothing more. Trust me it's just as frustrating for _us_ as _you_.

DjDiabolik, 2025-06-17:
> The IBVs don't tell us anything and we can't see the source code. We get back a value of "I/O Error" -- nothing more. Trust me it's just as frustrating for _us_ as _you_.

@hughsie exactly the opposite.. I understand the situation perfectly. Also because I myself have been able to verify that a second NUC (at my brother's house and specifically an ASUS VM40B much older than my current intel nuc) there the same procedure did not give any kind of problem.
Then the same evening I go home I try to do it on my intel nuc and here is the problem.
Really it could be anything:
Tomorrow I go back to my brother and try to see, from a shell, like if the efivars partition is also 128K or some other technical detail but I'm sure that the bios is older than mine as far as release date.

DjDiabolik, 2025-07-21:
> I see this issue is closed, but I have sort of the same problem. However, I managed to manually update db, dbx, and KEK in UEFI settings with the .bin files provided by the microsoft secureboot_objects repo, so there is apparently enough space? Now, `fwupdmgr` still reports that there is an update available but that there is even less free space.

without any addictional information about your system this feedback it's like useless..

DjDiabolik, 2025-07-22:
@wheeheee Aggressive what? When I reported and opened the issue, I was asking for information, and as you can see, to the best of my knowledge, I've always cooperated.
Up until now, you've only mentioned the problem, which, as you yourself said, doesn't even seem to be the same. So, am I the aggressive one?

DjDiabolik, 2025-07-22:
Anyway, to solve my problem. In the end, I didn't understand if the real problem was the lack (on efivars or similar) of space that was making it impossible to install the update itself.
Resetting the Secure Boot keys from the BIOS settings still seems to be the thing that was blocking everything.
I'll find out only in the next update if I'll need to go back and see the information in this ticket again.