fwupd/fwupd #8910: Does not how to handle URI | No URIs to download

jck-ifeu, 2025-06-12:
**Describe the bug**
if I start `fwupdmgr -v upgrade`
the following (verbose) output is generated with no installation of the update, because it can't handle the file or find the URI:

(fwupdmgr:5235): GLib-GIO-DEBUG: 14:51:11.360: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.360: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:5235): dconf-DEBUG: 14:51:11.361: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:5235): GLib-GIO-DEBUG: 14:51:11.361: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
(fwupdmgr:5235): pxbackend-DEBUG: 14:51:11.362: Active config plugins:
(fwupdmgr:5235): pxbackend-DEBUG: 14:51:11.362:  - config-env
(fwupdmgr:5235): pxbackend-DEBUG: 14:51:11.362:  - config-xdp
(fwupdmgr:5235): pxbackend-DEBUG: 14:51:11.362:  - config-kde
(fwupdmgr:5235): pxbackend-DEBUG: 14:51:11.362:  - config-gnome
(fwupdmgr:5235): pxbackend-DEBUG: 14:51:11.362:  - config-sysconfig
(fwupdmgr:5235): GLib-GIO-DEBUG: 14:51:11.362: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
(fwupdmgr:5235): GLib-GIO-DEBUG: 14:51:11.364: _g_io_module_get_default: Found default implementation networkmanager (GNetworkMonitorNM) for ‘gio-network-monitor’
(fwupdmgr:5235): pxbackend-DEBUG: 14:51:11.364: px_manager_constructed: Up and running
(fwupdmgr:5235): GLib-GIO-DEBUG: 14:51:11.364: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:5235): Fwupd-DEBUG: 14:51:11.366: Emitting ::status-changed() [idle]
(fwupdmgr:5235): FuMain-DEBUG: 14:51:11.373: current version is 0.15.2515: 0.15.2515=same, 0.5.2141=older
(fwupdmgr:5235): FuMain-DEBUG: 14:51:11.376: current version is 0.1.17: 0.1.17=same, 0.1.16=older, 0.1.15=older, 0.1.14=older, 0.1.12=older, 0.1.6=older
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 20241101 to 20250507?                                                                                                                              ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest                                                                                          ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ Some insecure versions of BiosFlashShell and Dtbios by DT Research Inc were  ║
║ added, due to a security vulnerability that allowed an attacker to bypass    ║
║ UEFI Secure Boot.                                                            ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
(fwupdmgr:5235): FuMain-DEBUG: 14:51:13.114: ignoring: No releases found
(fwupdmgr:5235): FuMain-DEBUG: 14:51:13.115: ignoring: No releases found
(fwupdmgr:5235): FuMain-DEBUG: 14:51:13.119: ignoring: No releases found
(fwupdmgr:5235): FuMain-DEBUG: 14:51:13.119: ignoring: No releases found
(fwupdmgr:5235): FuMain-DEBUG: 14:51:13.119: ignoring: No releases found
(fwupdmgr:5235): FuMain-DEBUG: 14:51:13.119: ignoring: No releases found
(fwupdmgr:5235): Fwupd-DEBUG: 14:51:13.120: do not how to handle URI 8b1efdd1ae2ae86b7a3d611570a4c02d644710e527b6b78917e8782aa3453166-DBXUpdate-20250507-x64.cab
No URIs to download

**Steps to Reproduce**
I assume: start the the update procedure (`fwupdmgr  upgrade`) on a computer/notebook where the update applies to. 

**Expected behavior**
The provided upgrade should install (get the right file from the server or an correct URI) or if not applicable to the system should not be displayed.

**fwupd version information**

compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.11
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libjcat           0.2.3
runtime   com.hughsie.libxmlb           0.3.22
[... 730 lines cut ...]

**Additional questions**

- Operating system and version: 
NAME="openSUSE Tumbleweed"
# VERSION="20250610"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20250610"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
# CPE 2.3 format, boo#1217921
CPE_NAME="cpe:2.3:o:opensuse:tumbleweed:20250610:*:*:*:*:*:*:*"
#CPE 2.2 format
#CPE_NAME="cpe:/o:opensuse:tumbleweed:20250610"
BUG_REPORT_URL="https://bugzilla.opensuse.org"
SUPPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo-Tumbleweed"

- Have you tried rebooting?
Yes
- Is this a regression?
I think, no...

hughsie, 2025-06-12:
We just saw this on OpenQA in Fedora too -- do you by any chance have a custom fwupd.conf without FirmwareBaseURI set?

jck-ifeu, 2025-06-12:
> Sie erhalten nicht häufig E-Mails von ***@***.*** Erfahren Sie, warum dies wichtig ist<https://aka.ms/LearnAboutSenderIdentification>

Hi Richard,

not that I know of. I have not ‘manually’ edited the configuration files at least in any way.

See output of sudo less /etc/fwupd/fwupd.conf > ~/Downloads/fwupd.conf attached. If the file only overwrites the settings, it may be OK and unchanged.

Thanks!
Jan

Mit freundlichen Grüßen

Jan Kräck

Mobilität

[X]<https://www.ifeu.de/>

ifeu - Institut für Energie- und Umweltforschung Heidelberg gGmbH

Wilckensstraße 3
69120 Heidelberg
Tel.: +49 6221 47 67 - 60 (direkt) - 0 (zentral)
***@***.***
www.ifeu.de<http://www.ifeu.de/>

Follow us: Twitter<https://twitter.com/ifeu> | Linkedin<https://www.linkedin.com/company/ifeu-ggmbh/mycompany/?viewAsMember=true>

Follow FlyingLess: Twitter<https://twitter.com/FlyingLess_de> | Linkedin<https://www.linkedin.com/showcase/flyingless/?viewAsMember=true>

ifeu-Newsletter: Anmelden<https://www.ifeu.de/> (ganz unten)

Presse: Kontakt und Presseverteiler<https://www.ifeu.de/presse/>

Geschäftsführung: Andreas Detzel, Lothar Eisenmann und Dr. Martin Pehnt

Handelsregister: Amtsgericht Mannheim HRB 334263
Sitz der Gesellschaft: Heidelberg

jck-ifeu, 2025-06-12:
Yes, of course.

[lvfs.conf.txt](https://github.com/user-attachments/files/20712631/lvfs.conf.txt)

In addition this information might help: a firmware update was running directly beforehand, which went through without any problems. Forgot to mention, sorry.

hughsie, 2025-06-12:
Actually cancel that, i worked it out:

[image]

hughsie, 2025-06-12:
I'm building and deploying https://gitlab.com/fwupd/lvfs-website/-/commit/45c7da1c0f725eb9cd094d099b6cdd90c461eacc then will regenerate metadata. Give me ~5 mins.

jck-ifeu, 2025-06-13:
Now it has worked. Thank you very much! Yesterday about 30...45 minutes after your last message it still didn't work. Thought I might wait until everything had synchronized. Just now, this morning (European local time) it worked after the refresh command.