fwupd/fwupd #8913: Prompted for update and reboot after updating and rebooting

hamirmahal, 2025-06-12:
**Describe the bug**
I keep seeing an option to update **Device Firmware** `Secure Boot dbx Configuration Update` on startup.

If I click it in `Ubuntu Software`, I get a popup saying `An update has been installed` and `A restart is required for it to take effect.`

Then, after restarting, I get prompted for the same update, again.

**Expected behavior**
The updates messages go away after updating.

**Additional questions**

- Operating system and version:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy
```
- Have you tried rebooting? **yes**

hamirmahal, 2025-06-12:
[image]

hughsie, 2025-06-13:
What hardware is this please?

hamirmahal, 2025-06-13:
> What hardware is this please?

<details>

<summary>**fwupd device information**</summary>

```
$  fwupdmgr get-devices --show-all-devices
System Product Name
│
├─Core™ i7-7700K CPU @ 4.20GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f8
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         809a0b93-8a12-5338-a571-ad5583acf896
│                         d0f754d5-1395-5573-bc83-85ba955da70a
│     Device Flags:       • Internal device
│   
├─GP107 [GeForce GTX 1050]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              1c0abc56-a884-57b2-8fc3-9d4786305ef1
│                         8cd24c9d-195b-5f2a-b3b9-370543564f9d
│                         21ed0b8c-dd28-51bc-abe1-0add082a1b9a
│                         ef3ab717-7332-5616-94e6-ffe243d75f69
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─SHGP31-1000GM:
│     Device ID:          310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:            NVM Express solid state drive
│     Current version:    41062C20
│     Vendor:             SK hynix (NVME:0x1C5C)
│     GUIDs:              34f9d6fa-e1d6-5d7d-909e-0406644298c5
│                         34e06406-f620-5b9d-a9cd-7f3a4e648633
│                         4406b8dc-20c5-5f3a-bc9c-8afdcee48684
│                         0ee57466-ca9d-5f10-a005-ec6bacf6fb82
│                         c555ec26-0c1b-5724-97ac-01bf179af935
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    4865
  │   Minimum Version:    4865
  │   Vendor:             System manufacturer (DMI:American Megatrends Inc.)
  │   Update State:       Success
  │   GUIDs:              8fcb46b9-2409-54ea-9b5f-aa04d310d9b3
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI revocation database
        Current version:  468
        Minimum Version:  468
        Vendor:           UEFI:Linux Foundation
        Install Duration: 1 second
        Update State:     Needs reboot
        Last modified:    2025-06-12 23:23
        GUIDs:            fda6234b-adcb-5105-8515-9af647d29775
                          f8ff0d50-c757-5dc3-951a-39d86e16f419
                          c6682ade-b5ec-57c4-b687-676351208742
                          f8ba2887-9411-5c36-9cee-88995bb39731
                          7d5759e5-9aa0-5f0c-abd6-7439bb11b9f6
                          0c7691e1-b6f2-5d71-bc9c-aabee364c916
        Device Flags:     • Internal device
                          • Updatable
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Only version upgrades are allowed
                          • Signed Payload
```
</details>

hughsie, 2025-06-13:
No, what hardware. e.g. what's the make and model of the computer. e.g. `sudo fwupdtool hwids` might give you sensible results, but the label on the box will be most accurate.

hughsie, 2025-06-13:
Oof, the last firmware update was from 2018 -- are you on BIOS 1301 already?

superm1, 2025-06-13:
> @majamee you should read https://fwupd.github.io/libfwupdplugin/uefi-dbx-README.html which explains the version format change. If you use a newer fwupd version in Ubuntu it'll show a date-based release in all cases.

A few ways to do that:
* Switch to newer Ubuntu release 
* Use the snap
* Use the flatpak
* Compile from source and run in fwupd development environment.

We can't force canonical to update this in older Ubuntu releases. It's up to them to decide what to do.

hamirmahal, 2025-06-13:
Would updating to the latest LTS version of Ubuntu fix this on my end?

superm1, 2025-06-13:
I would say "probably". And that's because it has a newer fwupd. But I don't want to assert that you won't run into a different bug. This dbx update has brought a lot of bugs from under the woodwork.

hamirmahal, 2025-06-13:
Would switching to a new motherboard and CPU help? They're both close to 10 years old now.

Are there any you recommend for running consumer-grade Ubuntu LTS versions?

superm1, 2025-06-13:
Newer motherboards usually have newer firmware and generally speaking newer firmware handles a lot of this stuff more gracefully.

If you're going to use Ubuntu yes I encourage using the latest LTS.