fwupd/fwupd #8914: No releases found for UEFI dbx (missing current version)

CragW, 2025-06-13:
**Describe the bug**
The `csum` from the last entry of DBX didn't get a match from the metadata, thus cannot get a `current version` to install new releases.

**Steps to Reproduce**
$ fwupdmgr install {device-id-dbx}

**Expected behavior**
Available updates are seen.

**fwupd version information**
Please provide the version of the daemon and client.

<details>

```shell
$ fwupdtool get-report-metadata
Loading…                 [*********************                  ]02:23:51.252 FuPluginIntelMchi    failed to get public key using file-id 0x40002300: generic failure [0xb]
Loading…                 [*************************************  ]
WARNING: This package has not been validated, it may not work properly.
Waiting…                 [*************************************  ]
BatteryLevel:           100
BatteryThreshold:       10
BootTime:               1749774144
CompileVersion(com.hughsie.libjcat): 0.2.0
CompileVersion(com.hughsie.libxmlb): 0.3.21
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.fwupd): 2.0.12
CompileVersion(org.freedesktop.fwupd.source): 2.0.11-37-ga4620cb97
CpuArchitecture:        x86_64
CpuModel:               Intel CoreTM Ultra 7 165U
DisplayState:           connected
DistroId:               ubuntu
DistroName:             Ubuntu
DistroPrettyName:       Ubuntu 24.04.2 LTS
DistroVersion:          24.04
FwupdSupported:         False
HostBaseboardManufacturer: Dell Inc.
HostBaseboardProduct:   UA010
HostBiosMajorRelease:   01
HostBiosMinorRelease:   0e
HostBiosVendor:         Dell Inc.
HostBiosVersion:        1.14.1
HostEnclosureKind:      1f
HostFamily:             Latitude
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 11
HostProduct:            Latitude 7350
HostSku:                CBE
HostVendor:             Dell Inc.
KernelName:             Linux
KernelVersion:          6.14.0-rc2-next-20250217+
LidState:               open
PlatformArchitecture:   x86_64
PowerState:             ac
RuntimeVersion(com.hughsie.libjcat): 0.2.0
RuntimeVersion(com.hughsie.libxmlb): 0.3.21
RuntimeVersion(org.freedesktop.fwupd): 2.0.12
RuntimeVersion(org.freedesktop.fwupd-efi): 1.4
RuntimeVersion(org.kernel): 6.14.0-rc2-next-20250217+
DeviceId:               da09124945f291ef6e9e9a50ef324a067716a70a
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               62eaeac7e8b975f47f5e95f824a40c1914a84dab
  pre:
[... 540 lines cut ...]
│     Install Duration:   30 seconds
│     GUIDs:              aa995882-da42-574a-8338-c8dfdca447e3 ← UFY\VID_046D&PID_C52B
│                         279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│                         9d131a0c-a606-580f-8eda-80587250b8d6 ← USB\VID_046D&PID_AAAA
│     Device Flags:       • Updatable
│                         • Supported on remote server
│                         • Signed Payload
│                         • Can tag for emulation
│
└─YJ3JK:
      Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
      Vendor:             PNP:AUO
      GUID:               340848af-fb81-553f-a1d4-9b81b5539548 ← DRM\VEN_AUO&DEV_CFA4
      Device Flags:       • Internal device
                          • Can tag for emulation

```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2025-06-13:
So your last hash is https://github.com/microsoft/secureboot_objects/blob/main/PreSignedObjects/DBX/dbx_info_msft_06_10_25.json#L4457 -- if so, I'm a bit confused -- those are all ia32 hashes -- what are they doing on your (presumably 64 bit) system?! Maybe this is a super-old system that had the pre-arch-split dbx? I think it's just fine to add it.

hughsie, 2025-06-13:
@CragW if you want to fix this "in the field" we could also upload the 939AEEF4F5FA51E23340C3F2E49048CE8872526AFDF752C3A7F3A3F2BC9F6049 release to the LVFS (and, presumably, all the other interim versions too) but I'd need you to find the dbxupdate.bin for 20180401 and upload it to https://github.com/fwupd/dbx-firmware/ -- it might be more of an archeological dig than you first planned.

closed 2025-06-16 by hughsie

CragW, 2025-06-17:
Thank you for the fix. Successfully updated from `20180401` to `20250507`.