fwupd/fwupd #8915: "Blocked executable in the ESP, ensure grub and shim are up to date": What to do with outdated .efi entries in recovery partitions?

pglpm, 2025-06-13:
**Describe the question**
The latest `fwupdmgr update` gave this:

```shell
[...]
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 267 to 20241101?                                       ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ An insecure version of Howyar's SysReturn software was added, due to a       ║
║ security vulnerability that allowed an attacker to bypass UEFI Secure Boot.  ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Decompressing…           [***************************************]
Blocked executable in the ESP, ensure grub and shim are up to date: /media/root/PQSERVICE/efi/boot/shimx64.efi Authenticode checksum [2ea4cb6a1f1eb1d3dce82d54fde26ded243ba3e18de7c6d211902a594fe56788] is present in dbx
```
The partition `/media/root/PQSERVICE/efi/boot/` seems to be a recovery partition preinstalled in my Thinkpad (which was shipped with Ubuntu alone). From what I understand, the `shimx64.efi` file there must now be outdated.

Now I've seen two suggestions to solve this problem:

1. Run `sudo fwupdmgr update --force -y`.
2. Mount `/media/root/PQSERVICE/efi/boot/` and then delete the `shimx64.efi` file there. Probably this renders that partition useless.

***Which of these two suggestions is safer? Are there better solutions?***

**fwupd version information**

```shell
$ fwupdtool get-report-metadata

Command not found

Use fwupdtool --help for help
```
Also run this, if it's of any help:
```shell
$ fwupdmgr get-devices --show-all-devices
20Y5CTO1WW
│
├─ELAN067A:00 04F3:3197:
│
[...]
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.29
│ │   Minimum Version:    0.1.23
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              f1918178-1ab6-4d4e-b82c-26e4d0e3869c
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  267
│       Minimum Version:  267
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       Update State:     Transient failure
│       Update Error:     Blocked executable in the ESP, ensure grub and shim are up to date: /media/root/PQSERVICE/efi/boot/shimx64.efi Authenticode checksum [2ea4cb6a1f1eb1d3dce82d54fde26ded243ba3e18de7c6d211902a594fe56788] is present in dbx
│       Last modified:    2025-06-13 04:38
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649
│                         5971a208-da00-5fce-b5f5-1234342f9cf7 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
[...]
```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,001A,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0001
Boot0000* ubuntu        HD(1,GPT,54b819f0-e0bf-4635-83c1-7aa52cf1e2eb,0x800,0x1b7801)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux-Firmware-Updater        HD(1,GPT,54b819f0-e0bf-4635-83c1-7aa52cf1e2eb,0x800,0x1b7801)/File(\EFI\ubuntu\fwupdx64.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Asset Information     FvFile(da465b87-a26f-4c12-b78a-0361428fa026)
Boot0015  Regulatory Information        FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0016  ThinkShield secure wipe       FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
Boot0017  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0018  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0019  MEBx Hot Key  FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot001A* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot001B* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001C* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001D* NVMe1 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a402)
Boot001E* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001F* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0020* PXE BOOT      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0021* LENOVO CLOUD  VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri(https://download.lenovo.com/pccbbs/cdeploy/efi/boot.efi)
Boot0022* ON-PREMISE    VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri()
Boot0023  Other CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35400)
Boot0024  Other HDD     VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot0025* IDER BOOT CDROM       PciRoot(0x0)/Pci(0x14,0x0)/USB(15,1)
Boot0026* IDER BOOT Floppy      PciRoot(0x0)/Pci(0x14,0x0)/USB(15,0)
Boot0027* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0028* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

**Additional questions**

- Operating system and version: Ubuntu 20.04.6 LTS
- Machine: Thinkpad X1 Extreme gen 4
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No idea, how do I check this?

hughsie, 2025-06-13:
> Mount /media/root/PQSERVICE/efi/boot/ and then delete the shimx64.efi file there. Probably this renders that partition useless.

The best solution is to get Lenovo to somehow update that recovery partition, but I don't think that's going to happen. We do try and avoid the recovery partitions -- I think that if you update fwupd to something newer this bug will go away. e.g. https://github.com/fwupd/fwupd/blob/a4620cb97cd3a0b58754f21db00c02f8449a0c3d/libfwupdplugin/fu-volume.c#L980

Can you open a bug against Ubuntu and ask them to update fwupd please?

closed 2025-06-13 by hughsie

pglpm, 2025-06-13:
I'll do that. Yet, 20.04 is now out of support; probably my best choice is to upgrade to a new Ubuntu version. Thank you for the help!