fwupd/fwupd #8923: Lenovo Thinkstation P8 Firmware upgrade failed

Zhitao-Li, 2025-06-15:
**Describe the bug**
Firmware version 1.54 failed to upgrade from 1.50, error message: Firmware version formats were different, device was 'pair' and release is 'triplet'

**Steps to Reproduce**
fwupdmgr install --allow-reinstall 6ec7d5c111ad7163fa544d687dce39e24c4e5afe824c5dfc9f1c2913057fed3e-Lenovo-ThinkStation-P8-System-Firmware-S0GKT36A.cab

**Expected behavior**
Should not have any errors.

**fwupd version information**
`fwupdmgr --version`

```
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.11
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.10
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.9-300.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.11
```

```shell
fwupdtool get-report-metadata

Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       25
BootTime:               1749967513
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.Passim): 0.1.10
CompileVersion(org.freedesktop.fwupd): 2.0.11
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen Threadripper PRO 7955WX 16-Cores
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 42 (KDE Plasma Desktop Edition)
DistroVariant:          kde
DistroVersion:          42
FwupdSupported:         True
HostBaseboardManufacturer: LENOVO
HostBaseboardProduct:   105E
HostBiosMajorRelease:   01
HostBiosMinorRelease:   32
HostBiosVendor:         LENOVO
HostBiosVersion:        S0GKT32A
HostEnclosureKind:      3
HostFamily:             ThinkStation P8
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 0f
HostProduct:            30HHCTO1WW
HostSku:                LENOVO_MT_30HH_BU_Think_FM_ThinkStation P8
HostVendor:             LENOVO
KernelCmdline:          no5lvl rd.driver.blacklist=nouveau modprobe.blacklist=nouveau
KernelName:             Linux
KernelVersion:          6.14.9-300.fc42.x86_64
LidState:               unknown
PassimDownloadSaving:   0
PlatformArchitecture:   x86_64
PowerState:             ac
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.11
RuntimeVersion(org.freedesktop.fwupd-efi): 1.6
[... 363 lines cut ...]
├── initramfs-6.14.6-300.fc42.x86_64.img
├── initramfs-6.14.9-300.fc42.x86_64.img
├── loader
│   └── entries  [error opening dir]
├── lost+found  [error opening dir]
├── symvers-6.14.4-300.fc42.x86_64.xz -> /lib/modules/6.14.4-300.fc42.x86_64/symvers.xz
├── symvers-6.14.6-300.fc42.x86_64.xz -> /lib/modules/6.14.6-300.fc42.x86_64/symvers.xz
├── symvers-6.14.9-300.fc42.x86_64.xz -> /lib/modules/6.14.9-300.fc42.x86_64/symvers.xz
├── System.map-6.14.4-300.fc42.x86_64
├── System.map-6.14.6-300.fc42.x86_64
├── System.map-6.14.9-300.fc42.x86_64
├── vmlinuz-0-rescue-396416b48e0e40e9abf1c618d059cba6
├── vmlinuz-6.14.4-300.fc42.x86_64
├── vmlinuz-6.14.6-300.fc42.x86_64
└── vmlinuz-6.14.9-300.fc42.x86_64
```

**Additional questions**

- Operating system and version: Fedora 42
- Have you tried rebooting? Yes
- Is this a regression? Not sure
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Not sure

Zhitao-Li, 2025-08-02:
No, still the same bug, trying to upgrade from 1.66 to 1.67.

hughsie, 2025-08-04:
Where did you get 6ec7d5c111ad7163fa544d687dce39e24c4e5afe824c5dfc9f1c2913057fed3e from? I can't find it on the LVFS at all...

Zhitao-Li, 2025-08-08:
I am using fedora 42, this is from the system's fwupd.

Zhitao-Li, 2025-08-08:
> Where did you get 6ec7d5c111ad7163fa544d687dce39e24c4e5afe824c5dfc9f1c2913057fed3e from? I can't find it on the LVFS at all...

Wait, this one is the one I downloaded from Lenovo US website, after the system's fwupd failed with the reported error. This one also produced the same error.