fwupd/fwupd #8929: UEFI db considered "invalid" after UEFI update

Jackaed, 2025-06-16:
**Describe the bug**
`fwupdmgr security` shows `UEFI db` as "Invalid" after update to 20250507

**Steps to Reproduce**
Update UEFI dbx from 20241101 to 20250507

**Expected behavior**
`fwupdmgr security` gives "Valid" `UEFI db`

**fwupd version information**
```
compile   info.libusb                   1.0.28
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.9
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.12.32
runtime   org.freedesktop.fwupd         2.0.9
```

```shell
fwupdtool get-report-metadata
fwupdtool hwids
```

```
Loading…                 [*****************                      ]11:46:26.272 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:33:00.0: failed to setup: cannot parse datastream
Waiting…                 [*************************************  ]11:46:26.899 FuDeviceLocker       failed to close device: cannot close device, refcount already zero

BatteryLevel:           101
BatteryThreshold:       25
BootTime:               1750000699
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.fwupd): 2.0.9
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 7 PRO 6850U with Radeon Graphics
DisplayState:           connected
DistroId:               nixos
DistroName:             NixOS
DistroPrettyName:       NixOS 25.11 (Xantusia)
DistroVariant:
DistroVersion:          25.11
FwupdSupported:         True
HostBaseboardManufacturer: LENOVO
HostBaseboardProduct:   21CQCTO1WW
HostBiosMajorRelease:   01
HostBiosMinorRelease:   2d
HostBiosVendor:         LENOVO
HostBiosVersion:        R22ET75W (1.45 )
HostEnclosureKind:      a
HostFamily:             ThinkPad T14s Gen 3
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 21
HostProduct:            21CQCTO1WW
HostSku:                LENOVO_MT_21CQ_BU_Think_FM_ThinkPad T14s Gen 3
HostVendor:             LENOVO
KernelCmdline:          lsm=landlock,yama,bpf workqueue.power_efficient=true vt.default_red=0x20,0xea,0xa9,0xd8,0x7d,0xd3,0x89,0xdd,0x5a,0xea,0xa9,0xd8,0x7d,0xd3,0x89,0xfb init=/nix/store/xjpczf9lgl5vjvn0h24yf2slfg9bc004-nixos-system-nixos-25.11.20250605.d3d2d80/init vt.default_grn=0x20,0x69,0xb6,0xa6,0xae,0x86,0xb4,0xc7,0x52,0x69,0xb6,0xa6,0xae,0x86,0xb4,0xf1 vt.default_blu=0x20,0x62,0x65,0x57,0xa3,0x9b,0x82,0xa1,0x4c,0x62,0x65,0x57,0xa3,0x9b,0x82,0xc7
KernelName:             Linux
KernelVersion:          6.12.32
LidState:               unknown
PlatformArchitecture:   x86_64
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.9
RuntimeVersion(org.kernel): 6.12.32
[... 539 lines cut ...]
│   │   ├── initrd-6.12.31-bpihno7o5yvdtffloxq6uiidivemtz275myejfkmbocjt5xrrlra.efi
│   │   ├── initrd-6.12.32-xzucts736vpro5umysqdp4g35pwa5dcdxuavhpkwzw6h5jwzbboq.efi
│   │   ├── kernel-6.12.30-2affc54ptzm5fly7x7jmrg2q7ei3v6l3edbf3xsbied4ups6gawa.efi
│   │   ├── kernel-6.12.31-mro2y6tu7emmih5a2p76nxrrcmwgmha3q4xiv2pp5xopi6pd3fvq.efi
│   │   └── kernel-6.12.32-htaz3gnvmtgiu6zmkujyrq57imfiqi5dq3zmudnzlgfi3sxcm5oa.efi
│   └── systemd
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    │   ├── nixos-generation-26.conf
    │   └── nixos-generation-27.conf
    ├── entries.srel
    ├── loader.conf
    └── random-seed

8 directories, 22 files
```
**Additional questions**

- Operating system and version: NixOS unstable
- Have you tried rebooting? Yes
- Is this a regression? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? It is currently, disabled it for the update
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? It is a lenovo system, but boot order lock is disabled

hughsie, 2025-06-16:
I think the key thing here is that db is a different file to dbx -- I'm waiting for Microsoft to make the db file public -- which is filed in https://github.com/microsoft/secureboot_objects/issues/203

Once that is deployed, the invalid will change to valid.

hughsie, 2025-11-11:
db is available now, and has been downloaded  586K times...