fwupd/fwupd #8945: ESP partition not being found although it is specified in the fwupd.conf (No udisk)

heitbaum, 2025-06-21:
**Describe the bug**
ESP partition not being found although it is specified in the fwupd.conf
No udisk as this is an embedded system

**Steps to Reproduce**
Steps to reproduce the behavior.
```
nuc12:~ # cat /etc/fwupd/fwupd.conf 
[fwupd]
EspLocation=/flash
nuc12:~ # tree /flash
/flash
├── EFI
│   └── BOOT
│       ├── bootia32.efi
│       ├── bootx64.efi
│       ├── grub.cfg
│       └── ldlinux.e64
├── KERNEL
├── SYSTEM
├── WSADL357.0096.CAP
├── WSADL357.cap
├── ldlinux.c32
├── ldlinux.sys
└── syslinux.cfg

3 directories, 11 files
nuc12:~ # fwupdtool esp-list --verbose
16:35:07.561 FuDebug              verbose to info (on console 1)
Loading…                 [*                                      ]16:35:07.562 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
16:35:07.562 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Loading…                 [*                                      ]16:35:07.563 FuContext            SMBIOS Manufacturer=Intel(R) Client Systems
16:35:07.563 FuContext            SMBIOS EnclosureKind=23
16:35:07.563 FuContext            SMBIOS Family=WS
16:35:07.563 FuContext            SMBIOS ProductName=NUC12WSKi7
16:35:07.563 FuContext            SMBIOS ProductSku=NUC12WSKi7000
16:35:07.563 FuContext            SMBIOS BiosVendor=Intel Corp.
16:35:07.563 FuContext            SMBIOS BiosVersion=WSADL357.0096.2025.0115.1051
16:35:07.563 FuContext            SMBIOS BiosMajorRelease=05
16:35:07.563 FuContext            SMBIOS BiosMinorRelease=1a
16:35:07.563 FuContext            SMBIOS FirmwareMajorRelease=52
16:35:07.563 FuContext            SMBIOS FirmwareMinorRelease=00
16:35:07.563 FuContext            SMBIOS BaseboardManufacturer=Intel Corporation
16:35:07.563 FuContext            SMBIOS BaseboardProduct=NUC12WSBi7
16:35:07.563 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
16:35:07.565 FuContext            Failed to get block devices: failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): The name org.freedesktop.UDisks2 was not provided by any .service files
16:35:07.565 FuContext            added udev subsystem watch of firmware-attributes
Loading…                 [**                                     ]16:35:07.565 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
16:35:07.565 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
Loading…                 [***                                    ]16:35:07.565 FuContext            added udev subsystem watch of drm
16:35:07.566 FuContext            added udev subsystem watch of block
16:35:07.566 FuContext            added udev subsystem watch of block:disk for plugin algoltek_usbcr
16:35:07.566 FuContext            added udev subsystem watch of i2c
16:35:07.566 FuContext            added udev subsystem watch of platform
16:35:07.566 FuContext            added udev subsystem watch of pci
16:35:07.566 FuContext            added udev subsystem watch of block:partition for plugin android_boot
16:35:07.566 FuContext            added udev subsystem watch of hidraw
16:35:07.566 FuContext            added udev subsystem watch of drm_dp_aux_dev for plugin bnr_dp
16:35:07.566 FuContext            added udev subsystem watch of i2c-dev
16:35:07.566 FuContext            added udev subsystem watch of gpio for plugin gpio
16:35:07.566 FuContext            added udev subsystem watch of mei
16:35:07.566 FuContext            added udev subsystem watch of iommu for plugin iommu
16:35:07.566 FuContext            added udev subsystem watch of video4linux
16:35:07.566 FuContext            added udev subsystem watch of msr for plugin msr
16:35:07.566 FuContext            added udev subsystem watch of mtd for plugin mtd
16:35:07.566 FuContext            added udev subsystem watch of nvme for plugin nvme
Loading…                 [***                                    ]16:35:07.566 FuContext            added udev subsystem watch of wwan
16:35:07.567 FuContext            added udev subsystem watch of serio
16:35:07.567 FuContext            added udev subsystem watch of thunderbolt
16:35:07.569 FuPluginUefiDbx      cannot find default ESP: failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): The name org.freedesktop.UDisks2 was not provided by any .service files
[... 581 lines cut ...]
│   
└─UEFI Device Firmware:
      Device ID:          aa57281a0d8785aa3ead30ef2ef4336bddaf3949
      Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)
      Current version:    1
      Vendor:             DMI:Intel Corp.
      Update State:       Success
      Update Error:       Not updatable as UEFI ESP partition not detected
      GUID:               2fe2cbfc-b9aa-4a93-ab5b-40173b581c42
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
      Device Requests:    • Message

```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04 / LibreELEC 13.0
- Have you tried rebooting? Yes
- Is this a regression? NA

superm1, 2025-06-21:
We used to support a case with no udisks, but not anymore. Udisks is used for more purposes now.

closed 2025-06-21 by superm1

heitbaum, 2025-06-22:
> We used to support a case with no udisks, but not anymore. Udisks is used for more purposes now.

Thanks for the clarification. Sad to hear. I’ll try looking for something else or another way to do updates, as udisks2 is not an option.