fwupd/fwupd #8960: UEFI dbx update from 466 to 20241101 fails

CompileNix, 2025-06-25:
**Describe the bug**

After updating UEFI dbx (466 → 20241101) via `fwupdmgr update` and a reboot, the update is not installed.

NOTE: During the reboot I see the following output from the firware updater: WARNING: No updates to process, exiting in 10 seconds

NOTE: The following fwupd files are present in `/sys/firmware/efi/efivars/`, after I performed the update via `fwupdmgr update` and before the reboot:
- `fwupd-6b5ce143-7c0e-4ca3-aa72-a620dba1c41d-0-0abba7dc-e516-4167-bbf5-4d9d1c739416`

**Steps to Reproduce**

1. `fwupdmgr update`
2. perform the update
3. reboot, as prompted
4. run `fwupdmgr update` shows the same update due

**Expected behavior**

The UEFI dbx to be installed in version 20241101.

**fwupd version information**
Please provide the version of the daemon and client.

<details>

<summary>`fwupdtool get-report-metadata`</summary>

```txt
Command not found

Use fwupdtool --help for help
```

</details>

<details>

<summary>`fwupdtool --version`</summary>

```txt
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.8.0-60-generic
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd         1.7.9
```

</details>

<details>

<summary>`fwupdtool hwids`</summary>

```txt
Computer Information
--------------------
BiosVendor: LENOVO
BiosVersion: R2LET33W (1.14 )
BiosMajorRelease: 1
BiosMinorRelease: 14
FirmwareMajorRelease: 01
FirmwareMinorRelease: 05
Manufacturer: LENOVO
Family: ThinkPad T14 Gen 5
ProductName: 21MC0058GE
ProductSku: LENOVO_MT_21MC_BU_Think_FM_ThinkPad T14 Gen 5
EnclosureKind: a
BaseboardManufacturer: LENOVO
BaseboardProduct: 21MC0058GE
[... 724 lines cut ...]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.15.0-142-generic
├── System.map-6.8.0-59-generic
├── System.map-6.8.0-60-generic
├── vmlinuz -> vmlinuz-5.15.0-142-generic
├── vmlinuz-5.15.0-142-generic
├── vmlinuz-6.8.0-59-generic
├── vmlinuz-6.8.0-60-generic
└── vmlinuz.old -> vmlinuz-6.8.0-60-generic

10 directories, 318 files
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 22.04.5 LTS
- Have you tried rebooting? Yes
- Is this a regression? N/A
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2025-06-25:
You need fwupd 2.0.12 for the dbx now I'm afraid. Does `sudo -i flatpak run org.freedesktop.fwupd refresh` and then `sudo -i flatpak run org.freedesktop.fwupd update` work better?

hughsie, 2025-06-26:
> Also it prompted me to update from 20241101 to 20250507 and not from 466 to 20241101 (or 20250507).

The 20241101 to 20250507 is what I expected -- see https://github.com/fwupd/fwupd/blob/main/plugins/uefi-dbx/README.md#comparing-versions for the why.

> but there is still the "WARNING: No updates to process, exiting in 10 seconds" message on reboot

dbx updates don't need to run the RT->UpdateCapsule helper so I don't know why that's being run.

> sudo -i flatpak run org.freedesktop.fwupd get-devices --show-all-devices shows 20250507

That looks like it succeeded. If you're using fwupdmgr 1.7.9 that's end of life and we don't support that any more.