fwupd/fwupd #8964: TPM PCR0 Invalid when SHA384 checksums in TPM enabled

lunarlattice0, 2025-06-26:
**Describe the bug**
Enabling SHA384 in conjunction with SHA256 in TPM options in my BIOS causes fwupdmgr to invalidate my TPM PCR0 value. This behavior does not occur when only SHA256 mode is enabled. 

**Steps to Reproduce**
Enable SHA384 in TPM2 and run `fwupdmgr security`

**Expected behavior**
The TPM PCR0 should validate regardless of what TPM type is used.

fwupdtool get-report-metadata
<details>
```
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1750928944
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.Passim): 0.1.10
CompileVersion(org.freedesktop.fwupd): 2.0.12
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 5 9600X 6-Core Processor
DisplayState:           unknown
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 42 (Server Edition)
DistroVariant:          server
DistroVersion:          42
EfivarsNvramFree:       78628
EfivarsNvramUsed:       71873
FwupdSupported:         True
HostBaseboardManufacturer: ASRockRack
HostBaseboardProduct:   B650D4U
HostBiosMajorRelease:   05
HostBiosMinorRelease:   23
HostBiosVendor:         American Megatrends International, LLC.
HostBiosVersion:        21.08
HostEnclosureKind:      11
HostFamily:             To be filled by O.E.M.
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            To be filled by O.E.M.
HostSku:                To be filled by O.E.M.
HostVendor:             AsrockRack
KernelName:             Linux
KernelVersion:          6.14.11-300.fc42.x86_64
LidState:               unknown
PassimDownloadSaving:   0
PlatformArchitecture:   x86_64
PowerState:             ac
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.12
RuntimeVersion(org.kernel): 6.14.11-300.fc42.x86_64
SELinux:                enforcing
  linux_lockdown:
      LinuxLockdown:    integrity
  tpm:
      Pcr0_SHA256:      4e088c83ecd38cc237afcb9dd1a14dbd831164c4d4852806c1a3eb73744d01db
      TpmEventLog:      0x00000008 d681837b3b8f5fd685bfad042ca78e6884e10e1c8f272dbccd2fd10be738a7fa [BSsQp8fZZUGBQCrd6Ur2PA==]
      0x8000000a e8ced075e13eaa8eecb285568b6066b55a75645087bec4b56706e6587f03c044 [D0ZWX01BSU5fV1JBUFBFUgAAAAEAAAAAAAC7AAAAAAA=]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: 4e088c83ecd38cc237afcb9dd1a14dbd831164c4d4852806c1a3eb73744d01db
      TpmFamily:        2.0
  uefi_pk:
      UefiPkKeyId:      073a97f61c424d169e88ad7cdc17e51ade4f0504
  uefi_capsule:
      SecureBoot:       Enabled
      UEFIUXCapsule:    Enabled
```
[... 251 lines cut ...]
      Serial Number:      WD-WX72A417DYZ6
      GUIDs:              69090f9a-662d-53f1-82fb-cf275607cac1 ← IDE\WDC_WD10SPZX-22Z10T1____________________04.01A04
                          beb1f617-07aa-557b-9c0d-ce6155df7097 ← IDE\0WDC_WD10SPZX-22Z10T1____________________
                          14206da8-7a48-5363-b1c1-d45a28d1025e ← WDC WD10SPZX-22Z10T1
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Can tag for emulation
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • UEFI dbx (20241101 → 20250507)
```
</details>

**Additional questions**

- Operating system and version:
Fedora 42
- Have you tried rebooting?
Yes
- Is this a regression? 
Unknown

hughsie, 2025-06-26:
I wonder what 'sudo fwupdtool security -vv' says about the reconstruction?

hughsie, 2025-06-27:
> However, sudo tpm2_pcrread sha384:0 returns nothing

Doesn't this indicate the TPM isn't actually calculating a sha384 PCR0?

lunarlattice0, 2025-06-28:
It seems that way, though I'm not sure why the AMD internal tpm is not calculating for sha384.

superm1, 2025-07-15:
It sounds like a firmware bug.  You should report it to your manufacturer.  Once they reproduce and acknowledge it, if it's a problem in AGESA they'll work with AMD to fix it.

EpicVon2468, 2026-10-03:
also experiencing this on intel, same setup; enable sha384 in tpm2, same result