fwupd/fwupd #8965: Enabling DMA remap support without actual DMAr support still allows Pre-Boot DMA protection to pass

lunarlattice0, 2025-06-26:
**Describe the bug**
Enabling the setting in AMD CBS listed as "DMA remap support in IVRS IVinfo Field.", but not the actual "Enable DMAr system protection during POST." setting still allows the respective field in `fwupdmgr security` to pass.

**Steps to Reproduce**
Enable "DMA Remap support in IVRS", but keep "DMAr system protection during POST" setting disabled.

**Expected behavior**
Pre-boot DMA protection should fail, since pre-boot protection was not actually enabled.

fwupdtool get-report-metadata
<details>

```BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1750928944
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.Passim): 0.1.10
CompileVersion(org.freedesktop.fwupd): 2.0.12
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 5 9600X 6-Core Processor
DisplayState:           unknown
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 42 (Server Edition)
DistroVariant:          server
DistroVersion:          42
EfivarsNvramFree:       78628
EfivarsNvramUsed:       71873
FwupdSupported:         True
HostBaseboardManufacturer: ASRockRack
HostBaseboardProduct:   B650D4U
HostBiosMajorRelease:   05
HostBiosMinorRelease:   23
HostBiosVendor:         American Megatrends International, LLC.
HostBiosVersion:        21.08
HostEnclosureKind:      11
HostFamily:             To be filled by O.E.M.
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            To be filled by O.E.M.
HostSku:                To be filled by O.E.M.
HostVendor:             AsrockRack
KernelName:             Linux
KernelVersion:          6.14.11-300.fc42.x86_64
LidState:               unknown
PassimDownloadSaving:   0
PlatformArchitecture:   x86_64
PowerState:             ac
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.12
RuntimeVersion(org.kernel): 6.14.11-300.fc42.x86_64
SELinux:                enforcing
  linux_lockdown:
      LinuxLockdown:    integrity
  tpm:
      Pcr0_SHA256:      4e088c83ecd38cc237afcb9dd1a14dbd831164c4d4852806c1a3eb73744d01db
      TpmEventLog:      0x00000008 d681837b3b8f5fd685bfad042ca78e6884e10e1c8f272dbccd2fd10be738a7fa [BSsQp8fZZUGBQCrd6Ur2PA==]
      0x8000000a e8ced075e13eaa8eecb285568b6066b55a75645087bec4b56706e6587f03c044 [D0ZWX01BSU5fV1JBUFBFUgAAAAEAAAAAAAC7AAAAAAA=]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: 4e088c83ecd38cc237afcb9dd1a14dbd831164c4d4852806c1a3eb73744d01db
      TpmFamily:        2.0
  uefi_pk:
      UefiPkKeyId:      073a97f61c424d169e88ad7cdc17e51ade4f0504
  uefi_capsule:
      SecureBoot:       Enabled
      UEFIUXCapsule:    Enabled
```
[... 253 lines cut ...]
      Serial Number:      WD-WX72A417DYZ6
      GUIDs:              69090f9a-662d-53f1-82fb-cf275607cac1 ← IDE\WDC_WD10SPZX-22Z10T1____________________04.01A04
                          beb1f617-07aa-557b-9c0d-ce6155df7097 ← IDE\0WDC_WD10SPZX-22Z10T1____________________
                          14206da8-7a48-5363-b1c1-d45a28d1025e ← WDC WD10SPZX-22Z10T1
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Can tag for emulation
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • UEFI dbx (20241101 → 20250507)
```
</details>

**Additional questions**

- Operating system and version:
Fedora 42
- Have you tried rebooting?
Yes
- Is this a regression? 
Unknown

superm1, 2025-06-26:
This sounds like a BIOS bug to me.

lunarlattice0, 2025-06-26:
> fwupd looks at the IVRS for that HSI variable. What's "Enable DMAr system protection during POST" actually doing?

I think what is happening is that the BIOS may be showing support, but not actually enabling the pre-boot protection. I'm not sure if a lying BIOS is something worth detecting

superm1, 2025-06-26:
> I think what is happening is that the BIOS may be showing support, but not actually enabling the pre-boot protection. I'm not sure if there is anything that can be done with a lying BIOS

Yeah that's why I think this is a BIOS bug.  This isn't a combination that should be allowed.

lunarlattice0, 2025-06-27:
Enabling DMAr system protection during POST on my system actually causes the entire system to slow to a crawl; I can't even get the system to boot into Linux and POST takes upwards of 10 minutes. Just loading the BIOS setup takes a solid 5 minutes, and I can see the background draw while it loads.

I'm not sure if this feature is not supported on non-pro AMD SKUs, I'm using a 9600X currently.

superm1, 2025-06-27:
This sounds like memory training to me. Is it perhaps just one time and needing to get cached?

lunarlattice0, 2025-06-27:
I don't think it is memory training, since it eventually boots into the EFI shell.

superm1, 2025-06-27:
> I don't think it is memory training, since it eventually boots into the EFI shell. 

I think you should report both of these issues to the OEM. I don't believe either is normal behavior.

closed 2025-07-24 by superm1