fwupd/fwupd #8969: modem-manager without udev

lvoegl, 2025-06-27:
**Is your feature request related to a problem? Please describe.**
Plugins relying on `fu_udev_device` (e.g. modem-manager) require udev Netlink events to detect replugs, etc.
When switching from `libgudev` to the udev Netlink socket, compatability for linux systems without systemd/udev or replacements seems limited, as they mostly do _not_ emit events to Netlink's group 0x2 (udev).

**Describe the solution you'd like**
Detect uevents on systems without systemd/udev.

One possible solution that I thought would be possible is adding something like a `fu_kernel_backend` using kernel events like
```
...
[KOBJ_ADD] =		"add",
[KOBJ_REMOVE] =		"remove",
[KOBJ_CHANGE] =		"change",
...
```
directly to substitute `fu_udev_backend` in that case.
As far as I can see, other than that, `fu_udev_device` and `fu_udev_backend` don't rely on udev specific behaviour or APIs.

I'm happy to help with any other ideas, implementations or whatsoever.

hughsie, 2025-07-04:
So you get kernel events, but not udev broadcast events? e.g. this would work:

            struct sockaddr_nl nls = {
                .nl_family = AF_NETLINK,
                .nl_pid = getpid(),
    -           .nl_groups = FU_UDEV_MONITOR_NETLINK_GROUP_UDEV,
    +           .nl_groups = FU_UDEV_MONITOR_NETLINK_GROUP_KERNEL,
            };

I guess then we could parse the `add@/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0` events in something different to fu_udev_backend_netlink_cb -- but my main question is *why* -- what system are you trying to support? Using the kernel even source by default is going to be really fragile IIUC: https://github.com/systemd/systemd/blob/c7fc641ea349a08291dd3cd6db4a99262f834271/src/libudev/libudev-monitor.c#L212

lvoegl, 2025-07-04:
I'm mainly trying to support openwrt and similar systems with sysvinit/non-systemd init systems. Connecting and parsing kernel netlink events safely is arguably out of scope for fwupd, but having another option to systemd would be great, because a lot of embedded devices do not run systemd.
ModemManager e.g. allows appending events via cli or their library to integrate with other init systems.
```
--report-kernel-event=["key=value,..."]                                                      Report kernel event
```

hughsie, 2025-07-04:
> Connecting to and processing kernel netlink events safely is arguably out of scope for fwupd

Ohh it's not too much code at all; e.g. does something like https://github.com/fwupd/fwupd/pull/9006 work?

lvoegl, 2025-07-04:
> Ohh it's not too much code at all; e.g. does something like [#9006](https://github.com/fwupd/fwupd/pull/9006) work?

That was basically my initial thought. I don't see this having any race conditions, similar to udev, on openwrt and similar systems as they only provides very basic functionality (no managing, renaming devices, etc.). I'll check in more detail on monday.

lvoegl, 2025-07-07:
Generally works really well on OpenWrt with those changes. I'm still having issues with devices replugging, even though add/remove/change events seem to be processed correctly - I think the ID does not match for some reason, but I haven't looked further into it yet.

closed 2025-07-11 by lvoegl