fwupd/fwupd #8976: Device not removed after _detach puts device in bootloader mode

vcdmp, 2025-06-30:
**Describe the bug**
Sometimes fwupd fails to update the device firmware. 

**Steps to Reproduce**
Upgrade Logitech Tap Touch Screen firmware using  logitech-tap plugin
Issue doesn't happen all the time but easy to reproduce (approx 2 failures out of 10 attempts)

**Expected behavior**
The upgrade should succeed.

**fwupd version information**
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.6
compile   com.hughsie.libxmlb           0.3.19
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd.source  2.0.6-15-gfc156e4f5
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.kernel                    5.10.228-25683-gd898b260cb14
runtime   org.freedesktop.fwupd         2.0.6

<details>
FuBackend/FuEngine/FuDeviceList are not removing device when device goes into bootloader mode. Updater waits for removal and eventually times out.

In a working case, _detach() puts device in bootloader mode and seeing following removal entries in the logs:
FuBackend            UDEV /sys/devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8.4/3-8.4.2/3-8.4.2:1.0/0003:222A:0001.000C/hidraw/hidraw5 removed
FuEngine             udev removed /sys/devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8.4/3-8.4.2/3-8.4.2:1.0/0003:222A:0001.000C/hidraw/hidraw5
FuEngine             auto-removing backend device ILITEK ILITEK-TP [3b446ec32a6055ad2d690013ff41ea2118051f9d]
FuPlugin             emit removed from logitech_tap: 3b446ec32a6055ad2d690013ff41ea2118051f9d
FuHistory            modifying device ILITEK ILITEK-TP [3b446ec32a6055ad2d690013ff41ea2118051f9d]
FuDeviceList         waiting 120000ms for ILITEK ILITEK-TP device removal
FuEngine             resetting update motd timeout

[fwupd_vv_logs.zip](https://github.com/user-attachments/files/20987729/fwupd_vv_logs.zip)

For non working case, _detach() puts device in bootloader mode.
Both kernel and fwupd sees device came back in bootloader mode:
kernel: [   10.871273] usb 3-8.4.2: New USB device found, idVendor=222a, idProduct=ff51, bcdDevice= 0.00

But not seeing device removed entry from fwupd, instead seeing following:
FuDeviceList         found existing device 36df72f9e35eae8c53c35b53b52e4340805956dd, reusing item
FuDeviceList         ::changed 36df72f9e35eae8c53c35b53b52e4340805956dd [Logitech Tap Display]

Whenever failure happens, seeing following entry as well:
fwupd[3206]: new: 3b446ec32a6055ad2d690013ff41ea2118051f9d [0x55a5d4416150] WAIT_FOR_REPLUG
fwupd[3206]: old: 3b446ec32a6055ad2d690013ff41ea2118051f9d [0x55a5d43f7a50]
fwupd[3206]: FuEngine             resetting update motd timeout
fwupd[3206]: FuUsbDevice          get 0x101 bytes of HID descriptor on iface 0x1
fwupd[3206]: libusb: error [submit_control_transfer] submiturb failed, errno=16
fwupd[3206]: FuUsbDevice          ignoring: failed to get HID report descriptor: USB error: Input/Output Error [-1]

Verbose logs for both working/nonworking case attached. Full OS logs available, if needed for further troubleshooting 
</details>

**Additional questions**

- Operating system and version: ChromeOS
- Have you tried rebooting? Yes
- Is this a regression? No

[fwupd_vv_logs.zip](https://github.com/user-attachments/files/20987737/fwupd_vv_logs.zip)

vcdmp, 2025-07-11:
Any updates here! I have attached another set of logs that exhibits same behavior. Hope that helps.

[fwupd_vv_logs_reproduced.zip](https://github.com/user-attachments/files/21175027/fwupd_vv_logs_reproduced.zip)

hughsie, 2025-07-14:
@vcdmp if you open another tab, does `sudo udevadm monitor` report events when fwupd fails to see them? If so, what do they look like please.