fwupd/fwupd #9003: questions should be sent to the terminal, not to the standard output

vinc17fr, 2025-07-04:
**Describe the bug**
Questions like
```
Review and upload report now? (Requires internet connection) [Y|n]:
```
are sent to the standard output, which may be redirected to a file, so that they are not visible in such a case. Instead, they should be sent to the terminal (if there is one).

**Steps to Reproduce**
Assuming that one is in a state where an upload would be proposed (I don't know how to force this for the test):
```
fwupdmgr get-devices > out
```
and one gets nothing in the terminal, but the command does not terminate. A look at file `out` shows the above question at the end (waiting for an answer).

**Expected behavior**
The question (and possibly related information that would come before) should appear in the terminal, not in the output file `out`.

**fwupd version information**
Debian package fwupd 2.0.12-1 (from experimental).

```
qaa:~> fwupdmgr --version
compile   info.libusb                   1.0.28
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.12
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.12.33+deb13-amd64
runtime   org.freedesktop.fwupd         2.0.12
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
```
aptitude install -t experimental fwupd
```

<details>

<summary>**fwupd device information**</summary>

This issue is unrelated to the devices.

</details>

**Additional questions**

- Operating system and version: Debian/unstable (with fwupd from experimental)
- Have you tried rebooting? This was done just after a reboot.
- Is this a regression? I don't know.

The bug seems to be in `src/fu-console.c`: function `fu_console_input_bool` is called, which calls `fu_console_print_full` with `FU_CONSOLE_PRINT_FLAG_NONE` for the flags, and this function has
```
        if (flags & FU_CONSOLE_PRINT_FLAG_STDERR) {
                g_printerr("%s", str->str);
        } else {
                g_print("%s", str->str);
        }
```
In this function, `FU_CONSOLE_PRINT_FLAG_WARNING` will set `FU_CONSOLE_PRINT_FLAG_STDERR`, but since there are no flags, the output is done with `g_print` above.

superm1, 2025-07-04:
What are you _trying_ to do?  `fwupdmgr` is intended to be scripted only with `--json`.

vinc17fr, 2025-07-04:
This is not in a script. I just want to keep the output as a human-readable format, which `--json` doesn't do.