fwupd/fwupd #9010: Update does not complete after reboot on Dell Precision XPS 15 7590

knirirr, 2025-07-06:
**Describe the bug**

After running `fwupdmgr upgrade` and rebooting the system an the expected system firmware upgrade from 1.36.0 to 1.37.0 has not occurred.

**Steps to Reproduce**

```
sudo fwupdmgr upgrade
```

...followed by rebooting. 

**Expected behavior**

Successful upgrade of the system firmware.

**fwupd version information**

```shell
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.7
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.9
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.6-109.bazzite.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.7
```

```shell
$ sudo fwupdtool get-report-metadata
Loading…                 [********************                   ]20:38:44.121 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/drm_dp_aux0: failed to setup: failed to read device firmware header: failed to write 7 bytes to 17: Input/output error
Waiting…                 [*************************************  ]
BatteryLevel:           100
BatteryThreshold:       10
BootTime:               1751816735
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.Passim): 0.1.9
CompileVersion(org.freedesktop.fwupd): 2.0.7
CpuArchitecture:        x86_64
CpuModel:               Intel Core™ i9-9980HK CPU @ 2.40GHz
DisplayState:           connected
DistroId:               bazzite
DistroName:             Bazzite
DistroPrettyName:       Bazzite 42 (FROM Fedora Silverblue)
DistroVariant:          bazzite-gnome-nvidia
DistroVersion:          42
FwupdSupported:         True
HostBaseboardManufacturer: Dell Inc.
HostBaseboardProduct:   T8KGX
HostBiosMajorRelease:   01
HostBiosMinorRelease:   24
HostBiosVendor:         Dell Inc.
HostBiosVersion:        1.36.0
HostEnclosureKind:      a
HostFamily:             XPS
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            XPS 15 7590
HostSku:                905
HostVendor:             Dell Inc.
KernelCmdline:          modprobe.blacklist=nouveau nvidia-drm.modeset=1 bluetooth.disable_ertm=1 rd.driver.blacklist=nouveau preempt=full initcall_blacklist=simpledrm_platform_driver_init
KernelName:             Linux
KernelVersion:          6.14.6-109.bazzite.fc42.x86_64
LidState:               open
PassimDownloadSaving:   37877557
[... 286 lines cut ...]
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─Precision 5540 System Update:
        New version:      1.37.0
        Remote ID:        lvfs
        Release ID:       111139
        Summary:          Firmware for the Dell Precision 5540
        Licence:          Proprietary
        Size:             22.6 MB
        Created:          2025-04-02
        Urgency:          Critical
        Vendor:           Dell
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        This stable release fixes the following issues:
        
        • This release contains security updates as disclosed in the Dell Security Advisories.
        Issues:           DSA-2025-129
                          DSA-2025-016
        Checksum:         250f8c47360976b69b61e2f8b48fe29d21965f4d1e61cc064a5ee4cd2db7962d

```

knirirr, 2025-07-21:
Version 1.38 applies, so that should fix the issue:

```
└─Precision 5540 System Update:
        New version:      1.38.0
        Remote ID:        lvfs
        Release ID:       115519
        Summary:          Firmware for the Dell Precision 5540
        Licence:          Proprietary
        Size:             22.6 MB
        Created:          2025-06-10
        Urgency:          Critical
        Vendor:           Dell
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        This stable release fixes the following issues:
        
        • This release contains security updates as disclosed in the Dell Security Advisory.
        Issue:            DSA-2025-206
        Checksum:         6169111f9cae065d14a5098318352f33277cdf150721749007c429031bfa4809
```

closed 2025-07-21 by knirirr

janderssonse, 2025-09-20:
Dell has only released 1.35 for the Dell 7590. I also get prompts to update to 1.38 however, that is only for the Dell Precision 5540. Why does fwupd identify the 7590 with an unoffoffical bios, this smells a bug. https://www.dell.com/support/home/en-se/drivers/driversdetails?driverid=v62g2

knirirr, 2025-09-22:
@janderssonse thanks - do you think this worth re-opening? Is there any chance that Dell may intend to apply that particular update, even though it's shown as for a different model?