fwupd/fwupd #9014: UEFI ESP partition not detected or configured - but ESP files are listed

AntonLydike, 2025-07-07:
**Describe the bug**
Running `sudo fwupdmgr update` or `sudo fwupdtool esp-list` prints a warning "UEFI ESP partition not detected or configured" and no updates are performed. But `sudo fwupdtool esp-list` does list my ESP files.

**Steps to Reproduce**
Not sure to be honest.

**Expected behavior**
I would expect not to get the warning message if it is able to detect my ESP, or not to get any ESP files if it indeed does not detect my ESP.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
|>sudo fwupdtool get-report-metadata
Loading…                 [****                                   ]10:51:08.613 FuPluginUefiCapsule  cannot find default ESP: user specified ESP /boot/ not found
10:51:08.614 FuPluginUefiCapsule  cannot find default ESP: user specified ESP /boot/ not found
10:51:08.614 FuPluginUefiCapsule  cannot find default ESP: user specified ESP /boot/ not found
Loading…                 [*************************************  ]
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1751884777
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.fwupd): 2.0.9
CpuArchitecture:        x86_64
CpuModel:               Intel 12th Gen Intel Core™ i7-1260P
DisplayState:           connected
DistroId:               nixos
DistroName:             NixOS
DistroPrettyName:       NixOS 25.05 (Warbler)
DistroVariant:
DistroVersion:          25.05
FwupdSupported:         True
HostBaseboardManufacturer: Framework
HostBaseboardProduct:   FRANMACP06
HostBiosMajorRelease:   03
HostBiosMinorRelease:   05
HostBiosVendor:         INSYDE Corp.
HostBiosVersion:        3.05
HostEnclosureKind:      a
HostFamily:             13in Laptop
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            Laptop (12th Gen Intel Core)
HostSku:                FRANDACP06
HostVendor:             Framework
KernelCmdline:          init=/nix/store/mvkrlkd1gprmfq29ynsyhbjcw8awkn3h-nixos-system-rahmenwerk-25.05.805852.29e290002bff/init nvme.noacpi=1 lsm=landlock,yama,bpf
KernelName:             Linux
KernelVersion:          6.12.35
LidState:               unknown
PlatformArchitecture:   x86_64
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.9
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.12.35
DeviceId:               102e4f7fbf3503e5ee1ec49439e84f130fee7e12
  pre:
      CapsuleApplyMethod: nvram
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               b7f65e1b9eacf47d90d6aaeb91972ff28507cdc3
  pre:
[... 358 lines cut ...]
- Have you tried rebooting? Yes
- Is this a regression? Not sure
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

## Additional Informatio:

Running `fwupdtool esp-list` will print a warning that no ESP partition is found, and then print a bunch of files on the ESP...

```
|>sudo fwupdtool esp-list
Loading…                 [************************************** ]
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/grub/x86_64-efi/crypto.mod
/boot/grub/x86_64-efi/video_bochs.mod
/boot/grub/x86_64-efi/newc.mod
/boot/grub/x86_64-efi/part_acorn.mod
/boot/grub/x86_64-efi/ehci.mod
/boot/grub/x86_64-efi/spkmodem.mod
```

The boot partition is FAT32, with BOOT and ESP flags set. The disk has disklabel type GPT. The only other partition is a luks encrypted one.

AntonLydike, 2025-07-07:
Output from `>sudo fwupdtool esp-list -v` is:

<details>
<summary>show output</summary>

```
13:04:50.663 FuDebug              verbose to info (on console 1)
13:04:50.666 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [*                                      ]13:04:50.670 FuQuirks             loading quirks from /nix/store/ixhji4kis52fjyb77z42i0b3q3k9wklk-fwupd-2.0.9/share/fwupd/quirks.d
13:04:50.670 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Loading…                 [*                                      ]13:04:50.671 FuContext            SMBIOS Manufacturer=Framework
13:04:50.671 FuContext            SMBIOS EnclosureKind=a
13:04:50.671 FuContext            SMBIOS Family=13in Laptop
13:04:50.671 FuContext            SMBIOS ProductName=Laptop (12th Gen Intel Core)
13:04:50.671 FuContext            SMBIOS ProductSku=FRANDACP06
13:04:50.671 FuContext            SMBIOS BiosVendor=INSYDE Corp.
13:04:50.671 FuContext            SMBIOS BiosVersion=03.05
13:04:50.671 FuContext            SMBIOS BiosMajorRelease=03
13:04:50.671 FuContext            SMBIOS BiosMinorRelease=05
13:04:50.671 FuContext            SMBIOS FirmwareMajorRelease=ff
13:04:50.671 FuContext            SMBIOS FirmwareMinorRelease=ff
13:04:50.671 FuContext            SMBIOS BaseboardManufacturer=Framework
13:04:50.671 FuContext            SMBIOS BaseboardProduct=FRANMACP06
13:04:50.671 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
13:04:50.694 FuContext            added udev subsystem watch of firmware-attributes
Loading…                 [**                                     ]13:04:50.694 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
13:04:50.694 FuEngine             ignoring: Error opening directory “/nix/store/ixhji4kis52fjyb77z42i0b3q3k9wklk-fwupd-2.0.9/share/fwupd/local.d”: No such file or directory
Loading…                 [***                                    ]13:04:50.695 FuContext            added udev subsystem watch of drm
13:04:50.699 FuContext            added udev subsystem watch of block
13:04:50.699 FuContext            added udev subsystem watch of block:disk for plugin algoltek_usbcr
13:04:50.699 FuContext            added udev subsystem watch of i2c
13:04:50.699 FuContext            added udev subsystem watch of platform
13:04:50.699 FuContext            added udev subsystem watch of pci
13:04:50.699 FuContext            added udev subsystem watch of block:partition for plugin android_boot
13:04:50.699 FuContext            added udev subsystem watch of drm_dp_aux_dev for plugin bnr_dp
13:04:50.700 FuContext            added udev subsystem watch of i2c-dev
13:04:50.700 FuContext            added udev subsystem watch of hidraw
13:04:50.700 FuContext            added udev subsystem watch of gpio for plugin gpio
13:04:50.700 FuContext            added udev subsystem watch of mei
13:04:50.700 FuContext            added udev subsystem watch of iommu for plugin iommu
13:04:50.701 FuContext            added udev subsystem watch of video4linux
13:04:50.701 FuContext            added udev subsystem watch of msr for plugin msr
13:04:50.701 FuContext            added udev subsystem watch of mtd for plugin mtd
13:04:50.701 FuContext            added udev subsystem watch of nvme for plugin nvme
Loading…                 [***                                    ]13:04:50.701 FuContext            added udev subsystem watch of wwan
13:04:50.702 FuContext            added udev subsystem watch of serio
13:04:50.702 FuContext            added udev subsystem watch of thunderbolt
13:04:50.702 FuContext            added udev subsystem watch of tpm for plugin tpm
13:04:50.721 FuVolume             Looking for volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
13:04:50.724 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: crypto_LUKS
13:04:50.746 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
13:04:50.757 FuVolume             Looking for volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
13:04:50.759 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: crypto_LUKS
13:04:50.774 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
13:04:50.776 FuPluginUefiDbx      cannot find default ESP: user specified ESP /boot/ not found
13:04:50.776 FuEngine             plugins disabled: test, test_ble
Loading…                 [***                                    ]13:04:50.777 FuContext            battery threshold now 10
Loading…                 [************************************** ]13:04:50.777 FuEngine             resetting update motd timeout

WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/grub/x86_64-efi/crypto.mod
/boot/grub/x86_64-efi/video_bochs.mod
/boot/grub/x86_64-efi/newc.mod
```
</details>

AntonLydike, 2025-07-07:
More information on the partition:

fdisk says it's GPT disklabel, 
```
Disk /dev/nvme0n1: 1.82 TiB, 2000398934016 bytes, 3907029168 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt

Device           Start        End    Sectors  Size Type
/dev/nvme0n1p1    2048    2099199    2097152    1G EFI System
/dev/nvme0n1p2 ... Linux filesystem
```

```
>sudo file -s /dev/nvme0n1p1
/dev/nvme0n1p1: DOS/MBR boot sector, code offset 0x58+2, OEM-ID "mkfs.fat", sectors/cluster 8, Media descriptor 0xf8, sectors/track 63, heads 255, hidden sectors 2048, sectors 2097144 (volumes > 32 MB), FAT (32 bit), sectors/FAT 2048, reserved 0x1, serial number 0x30cd982a, unlabeled
```

Looks like FAT 32.

Interestingly, this states that it's MBR, but I think it's mistaken.

hughsie, 2025-07-07:
Try taking it out of fwupd.conf and let fwupd auto detect it with udisks perhaps?

AntonLydike, 2025-07-07:
> Try taking it out of fwupd.conf and let fwupd auto detect it with udisks perhaps?

It's not quite trivial on NixOS, the script that builds the config file does not allow omitting a value for this.

superm1, 2025-07-07:
This prompts me for another improvement we can do.  If someone puts the device node in the field, we could detect the prefix of `/dev` and tell them it's for mount points instead.

closed 2025-07-07 by superm1