fwupd/fwupd #9025: Upgrade UEFI dbx from 466 to 20241101 not working

bijink, 2025-07-10:
**Describe the bug**
On my Dell Latitude 5410 running Ubuntu, I'm unable to apply a UEFI Secure Boot dbx firmware update using fwupd.
Despite following all steps (boot order change, Secure Boot disabled, reinstall attempts), the update fails on reboot with the following error:

> fwupd-efi version 1.4
> WARNING: No updates to process, exiting in 10 seconds.

Followed by:
> start_image() returned Invalid Parameter, falling back to default loader

**Steps to Reproduce**
1. Run sudo fwupdmgr get-updates to list available updates.
2. Install the UEFI dbx update using sudo fwupdmgr update.
3. Reboot the system.
4. Observe the fwupd-efi failure screen.

Additional steps tried:
- Disabled Secure Boot in BIOS.
- Moved “Linux Firmware Update” to first boot priority.
- Attempted reinstall via sudo fwupdmgr reinstall 2.

**Expected behavior**
I expected the firmware update to apply successfully during the next boot, with fwupd-efi executing the capsule update.

**fwupd version information**
Please provide the version of the daemon and client.

```
$ fwupdtool get-report-metadata
Command not found
Use fwupdtool --help for help
NOTE: This program may only work correctly as root

$ sudo fwupdtool get-report-metadata
Command not found
Use fwupdtool --help for help
```
```
$ fwupdmgr --version

runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.8.0-60-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

```

```
$ fwupdtool hwids
Computer Information
--------------------
BiosVendor: Dell Inc.
BiosVersion: 1.36.1
Manufacturer: Dell Inc.
Family: Latitude
ProductName: Latitude 5410
ProductSku: 9A0
EnclosureKind: a
BaseboardManufacturer: Dell Inc.
BaseboardProduct: 

Hardware IDs
------------
not available as 'BiosMajorRelease' unknown
not available as 'BiosMajorRelease' unknown
not available as 'BiosMajorRelease' unknown
{c4762fdd-9943-556d-ba40-acd1733f5144}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
[... 445 lines cut ...]
├── initrd.img -> initrd.img-6.8.0-60-generic
├── initrd.img-6.8.0-59-generic
├── initrd.img-6.8.0-60-generic
├── initrd.img.old -> initrd.img-6.8.0-59-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-6.8.0-59-generic
├── System.map-6.8.0-60-generic
├── vmlinuz -> vmlinuz-6.8.0-60-generic
├── vmlinuz-6.8.0-59-generic
├── vmlinuz-6.8.0-60-generic
└── vmlinuz.old -> vmlinuz-6.8.0-59-generic

5 directories, 302 files
```

**Additional questions**

- Operating system and version: Linux Ubuntu 22.04.5 LTS
- Have you tried rebooting? Yes
- Is this a regression? Yes
- Are you using an NVMe disk? No
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2025-07-10:
1.7.x is EOL I'm afraid; you need to ask Ubuntu to update the fwupd version: https://github.com/fwupd/fwupd/blob/main/SECURITY.md#supported-versions

closed 2025-07-10 by hughsie

RichardAmes, 2025-07-19:
I have this issue also......  though my unapplied update is "487 --> 20241101".

jvcoelho, 2025-07-21:
I have this issue also in Dell Inc. Inspiron 15 5510

`$  lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy
`
`
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 466 to 20241101?                                       ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ An insecure version of Howyar's SysReturn software was added, due to a       ║
║ security vulnerability that allowed an attacker to bypass UEFI Secure Boot.  ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝
`