fwupd/fwupd #9050: Lenovo ThinkCentre M270s fwupdmgr says disc is encrypted

luisfcorreia, 2025-07-19:
**Describe the question**
This is a refurbish machine, where the nvme drive was wiped before installing Fedora Workstation 42 from scratch

**fwupd version information**
latest available as of today

```shell
fwupdtool get-report-metadata
```

BatteryLevel:           101
BatteryThreshold:       25
BootTime:               1752947124
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.Passim): 0.1.10
CompileVersion(org.freedesktop.fwupd): 2.0.12
CpuArchitecture:        x86_64
CpuModel:               Intel Core™ i5-8500 CPU @ 3.00GHz
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 42 (Workstation Edition)
DistroVariant:          workstation
DistroVersion:          42
EfivarsNvramFree:       250896
EfivarsNvramUsed:       79277
FwupdSupported:         True
HostBaseboardManufacturer: LENOVO
HostBaseboardProduct:   312A
HostBiosMajorRelease:   01
HostBiosMinorRelease:   77
HostBiosVendor:         LENOVO
HostBiosVersion:        M1UKT77A
HostEnclosureKind:      3
HostFamily:             ThinkCentre M720s
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 18
HostProduct:            10SUS1SF03
HostSku:                LENOVO_MT_10SU_BU_Think_FM_ThinkCentre M720s
HostVendor:             LENOVO
KernelName:             Linux
KernelVersion:          6.15.6-200.fc42.x86_64
LidState:               unknown
PassimDownloadSaving:   0
PlatformArchitecture:   x86_64
PowerState:             ac
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.12
RuntimeVersion(org.freedesktop.fwupd-efi): 1.6
RuntimeVersion(org.kernel): 6.15.6-200.fc42.x86_64
SELinux:                enforcing
DeviceId:               d5fa26ffd0f51f9730fc26ba044545bb0bca6a68
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x10077
DeviceId:               e68fe1402f9ac465f7713156fb286d14b29d7207
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0xc0230593
  linux_lockdown:
      LinuxLockdown:    none
  tpm:
      Pcr0_SHA1:        5921f7f6b8c20e9a72629847091296bd263a16e5
      Pcr0_SHA256:      7aced6366cb6543c967af66ec66bdb5a94dc19250218f307213fc7508ca5ea6c
      TpmEventLog:      0x00000008 c42fedad268200cb1d15f97841c344e79dae3320 [HvtrVAwdVUCkrU70vxe4Og==]
      0x00000001 a150ac108df098886c5084f7ef819448cd1ec519 [AGBL/wAAAAAAoHQAAAAAAA==]
      0x00000004 9069ca78e7450a285173431b3e52c5c25299e473 [AAAAAA==]
      PCR0: 5921f7f6b8c20e9a72629847091296bd263a16e5
      PCR0: 7aced6366cb6543c967af66ec66bdb5a94dc19250218f307213fc7508ca5ea6c
      TpmFamily:        2.0
  uefi_pk:
      UefiPkKeyId:      f6895759b0a357b240fdcfd8eafc3780265c91d9
  uefi_capsule:
      BootMgrDesc:      legacy
      SecureBoot:       Disabled
      UEFIUXCapsule:    Enabled

Applied the latest available firmware from Lenovo (M1UKT77A)
Out of ideas

JohnAZoidberg, 2025-08-21:
So you're not using LUKS?

fwupd gets this information from udisks, so you could you please run `udiskctl dump` and look for the information about your hard drive (or you can just paste the full output):

This is what it looks like for my drive:

```
/org/freedesktop/UDisks2/block_devices/nvme0n1:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1
    DeviceNumber:               66304
    Drive:                      '/org/freedesktop/UDisks2/drives/WD_PC_SN8000S_SDEPNRG_2T00_24434D800018'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-nvme-WD_PC_SN8000S_SDEPNRG-2T00_24434D800018
    IdLabel:                    
    IdType:                     
    IdUUID:                     
    IdUsage:                    
    IdVersion:                  
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1
    ReadOnly:                   false
    Size:                       2048408248320
    Symlinks:                   /dev/disk/by-diskseq/1
                                /dev/disk/by-id/nvme-WD_PC_SN8000S_SDEPNRG-2T00_24434D800018
                                /dev/disk/by-id/nvme-WD_PC_SN8000S_SDEPNRG-2T00_24434D800018_1
                                /dev/disk/by-id/nvme-eui.e8238fa6bf530001001b448b40d08f16
                                /dev/disk/by-path/pci-0000:c1:00.0-nvme-1
```

fwupd checks `CryptoBackingDevice:        '/'`
Because it's set to `/` that means there's no encryption.

luisfcorreia, 2025-08-21:
Yes, I did not install any encription nor activate LUKS

here's the requested log:

```

/org/freedesktop/UDisks2/block_devices/nvme0n1:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1
    DeviceNumber:               66304
    Drive:                      '/org/freedesktop/UDisks2/drives/INTEL_SSDPEKKF256G8L_BTHP923304F9256B'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-nvme-INTEL_SSDPEKKF256G8L_BTHP923304F9256B
    IdLabel:                    
    IdType:                     
    IdUUID:                     
    IdUsage:                    
    IdVersion:                  
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1
    ReadOnly:                   false
    Size:                       256060514304
    Symlinks:                   /dev/disk/by-diskseq/3
                                /dev/disk/by-id/nvme-INTEL_SSDPEKKF256G8L_BTHP923304F9256B
                                /dev/disk/by-id/nvme-INTEL_SSDPEKKF256G8L_BTHP923304F9256B_1
                                /dev/disk/by-id/nvme-eui.5cd2e42791a04f11
                                /dev/disk/by-path/pci-0000:01:00.0-nvme-1
```

luisfcorreia, 2025-08-21:
btw, using Fedora Workstation 42, the actual command I ran was

`udisksctl dump`

JohnAZoidberg, 2025-08-28:
9e61a75b48581e98982c8258ee3d151bae5ca54e

luisfcorreia, 2025-08-31:
it still does not work, now with 2.0.14

`│ ├─UEFI Key Exchange Key:
│ │ │   ID do dispositivo: 2a4c23bfb79b5dabe474cb7b1b3e604645d6f9c6
│ │ │   Opções do dispositivo: • Dispositivo interno
│ │ │ 
│ │ ├─KEK CA:
│ │ │     ID do dispositivo: b7a1d3d90faa1f6275d9a98da4fb3be7118e61c7
│ │ │     Versão atual:   2011
│ │ │     Fornecedor:     Microsoft (UEFI:Microsoft)
│ │ │     Estado da atualização: Falhou
│ │ │     Erro de atualização: failed to write-firmware: failed to write (null): failed to write data to efivarsfs: Erro ao escrever no descritor do ficheiro: Argumento inválido
│ │ │     Última modificação: 2025-08-31 15:56:06
│ │ │     GUIDs:          814e950f-1449-566a-a190-42c9d3a3a2df ← UEFI\VENDOR_Microsoft&NAME_Microsoft-KEK-CA
│ │ │                     dfa66406-6568-5bdf-bb8e-b53ddb4be4cf ← UEFI\CRT_9F402B1CC0243CBEDC58A525789816CCCA7687A9
│ │ │     Opções do dispositivo: • Dispositivo interno
│ │ │                     • Atualizável
│ │ │                     • Suporte no servidor remoto
│ │ │                     • Precisa de uma reinício após a instalação
│ │ │                     • O dispositivo pode ser usado durante a atualização
│ │ │                     • Signed Payload
│ │ │                     • Can tag for emulation
`

Erro de atualização: failed to write-firmware: failed to write (null): failed to write data to efivarsfs: Erro ao escrever no descritor do ficheiro: Argumento inválido

This roughly translates to:

Update error: failed to write-firmware: failed to write (null): failed to write data to efivarsfs: Error writing file descriptior, invalid argument