fwupd/fwupd #9056: Microsoft KEK and DB key not updateable

JohnAZoidberg, 2025-07-21:
**Describe the bug**
KEK and DB are listed as not having any updates available.

```
[I] (fwupd) zoid@fedora ~/c/fwupd (main) [2]> fwupdtool get-updates --plugins uefi-kek --plugins uefi-db
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • KEK CA
 • UEFI CA
 • Windows Production PCA
No updates available for remaining devices
```

```
[I] (fwupd) zoid@fedora ~/c/fwupd (main) [2]> fwupdtool get-devices --plugins uefi-kek --plugins uefi-db                                                                                                           
Loading…                 [************************************** ]                                                                                                                                                 
WARNING: This package has not been validated, it may not work properly.                                                                                                                                            
Framework Laptop 16 (AMD Ryzen 7040 Series)
│
├─UEFI Key Exchange Key:
│ │   Device ID:          2a4c23bfb79b5dabe474cb7b1b3e604645d6f9c6
│ │   Device Flags:       • Internal device
│ │ 
│ └─KEK CA:
│       Device ID:        b7a1d3d90faa1f6275d9a98da4fb3be7118e61c7
│       Current version:  2011
│       Vendor:           Microsoft (UEFI:Microsoft)
│       GUIDs:            814e950f-1449-566a-a190-42c9d3a3a2df ← UEFI\VENDOR_Microsoft&NAME_Microsoft-KEK-CA
│                         dfa66406-6568-5bdf-bb8e-b53ddb4be4cf ← UEFI\CRT_9F402B1CC0243CBEDC58A525789816CCCA7687A9
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Full disk encryption secrets may be invalidated when updating
│                         • Signed Payload
│                         • Can tag for emulation
│     
└─UEFI Signature Database:
  │   Device ID:          0352a8acc949c7df21fec16e566ba9a74e797a97
  │   Device Flags:       • Internal device
  │ 
  ├─UEFI CA:
  │     Device ID:        5bc922b7bd1adb5b6f99592611404036bd9f42d0
  │     Current version:  2011
  │     Vendor:           Microsoft (UEFI:Microsoft)
  │     GUIDs:            26f42cba-9bf6-5365-802b-e250eb757e96 ← UEFI\VENDOR_Microsoft&NAME_Microsoft-UEFI-CA
  │                       c34a7e6a-bd86-5244-8bd0-7db66fd3c073 ← UEFI\CRT_E30CF09DABEAB32A6E3B07A7135245DE05FFB658
  │     Device Flags:     • Internal device
  │                       • Updatable
  │                       • Needs a reboot after installation
  │                       • Full disk encryption secrets may be invalidated when updating
  │                       • Signed Payload
  │                       • Can tag for emulation
  │   
  └─Windows Production PCA:
        Device ID:        ad7e00ec37f005ae10492bdb7f73aef0d2e20488
        Current version:  2011
        Vendor:           Microsoft (UEFI:Microsoft) 
        GUIDs:            675d2184-6c9a-59f1-a6f1-3c229b5dbb79 ← UEFI\VENDOR_Microsoft&NAME_Microsoft-Windows-Production-PCA
                          0611d85d-99a4-5c50-8c17-fc5196226f85 ← UEFI\CRT_1A8B6903D64CC9AD09D12FCB355663A458A09EF0
        Device Flags:     • Internal device
                          • Updatable
                          • Needs a reboot after installation
                          • Full disk encryption secrets may be invalidated when updating
                          • Signed Payload
                          • Can tag for emulation
```

**Expected behavior**
I expect a Microsoft 2023 KEK and DB update.

**fwupd version information**

```
# Built from source
[I] (fwupd) zoid@fedora ~/c/fwupd (main)> fwupdmgr --version
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.13
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.10
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.fwupd.source  2.0.12-148-g9d908d10b
runtime   org.freedesktop.Passim        0.1.10
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.15.6-100.fc41.x86_64
runtime   org.freedesktop.fwupd         2.0.13
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**System UEFI configuration**
Please provide the output of the following commands:

```shell
efibootmgr -v
```

```shell
efivar -l | grep fw
```

```shell
tree /boot
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
- Are you using an NVMe disk?
- Is secure boot enabled?
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

JohnAZoidberg, 2025-07-24:
https://fwupd.org/lvfs/devices/com.microsoft.db-uefi-3p.firmware

This is only the Microsoft UEFI CA 2023 DB key, or also includes the KEK?
Extracting the cab file it looks like only "Microsoft UEFI CA 2023" and "Microsoft Option ROM UEFI CA 2023" are included.

But the lvfs page says "Updating the UEFI KEK allows updating the UEFI Signature Database."

JohnAZoidberg, 2025-07-24:
> I thought the new MS UEFI 2023 db was signed by the new KEK, but it's actually signed by the **old** KEK

That is correct. That's the point of the KEK :D It's only used when checking the signature of new KEK/DB/DBX keys.

Now it works!
I uploaded a report.

```
> sudo fwupdtool get-devices --plugins uefi_kek
Framework Laptop 13 (Intel Core Ultra Series 1)
│
└─UEFI Key Exchange Key:
  │   Device ID:          2a4c23bfb79b5dabe474cb7b1b3e604645d6f9c6
  │   Device Flags:       • Internal device
  │
  ├─KEK CA:
  │     Device ID:        b7a1d3d90faa1f6275d9a98da4fb3be7118e61c7
  │     Current version:  2011
  │     Vendor:           Microsoft (UEFI:Microsoft)
  │     GUIDs:            814e950f-1449-566a-a190-42c9d3a3a2df ← UEFI\VENDOR_Microsoft&NAME_Microsoft-KEK-CA
  │                       dfa66406-6568-5bdf-bb8e-b53ddb4be4cf ← UEFI\CRT_9F402B1CC0243CBEDC58A525789816CCCA7687A9
  │     Device Flags:     • Internal device
  │                       • Updatable
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │                       • Full disk encryption secrets may be invalidated when updating
  │                       • Signed Payload
  │                       • Can tag for emulation
  │
  └─Key Exchange Key:
        Device ID:        7227626d9209595851b3840e3d626601a375ce2d
        Current version:  2024
        Vendor:           Unknown (UEFI:UNKNOWN)
        GUID:             6b3f091b-067c-55c3-9c84-393aa4f44c05 ← UEFI\CRT_3BB7B137630F02EF8DA913573061E3226970B06C
        Device Flags:     • Internal device
                          • Updatable
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Full disk encryption secrets may be invalidated when updating
                          • Signed Payload
                          • Can tag for emulation

> fwupdmgr get-updates
└─UEFI CA:
  │   Device ID:          5bc922b7bd1adb5b6f99592611404036bd9f42d0
  │   Current version:    2011
  │   Vendor:             Microsoft (UEFI:Microsoft)
  │   GUIDs:              26f42cba-9bf6-5365-802b-e250eb757e96 ← UEFI\VENDOR_Microsoft&NAME_Microsoft-UEFI-CA
  │                       c34a7e6a-bd86-5244-8bd0-7db66fd3c073 ← UEFI\CRT_E30CF09DABEAB32A6E3B07A7135245DE05FFB658
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Full disk encryption secrets may be invalidated when updating
  │                       • Signed Payload
  │                       • Can tag for emulation
  │
  └─Secure Boot Signature Database Configuration Update:
        New version:      2023
        Remote ID:        lvfs
        Release ID:       116503
        Summary:          UEFI Secure Boot Signature Database
        License:          Proprietary
        Size:             10.0 kB
        Created:          2025-04-29
        Urgency:          High
        Vendor:           Linux Foundation
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:
        This updates the 3rd Party UEFI Signature Database (the "db") to the latest release from Microsoft. It also adds the latest OptionROM UEFI Signature Database update.
        Checksum:         0bc3c04d084462991ee96c2fed91bbf27ead45491b590e1eb20abdd73148fb12

> fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI CA from 2011 to 2023?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the 3rd Party UEFI Signature Database (the "db") to the latest  ║
║ release from Microsoft. It also adds the latest OptionROM UEFI Signature     ║
║ Database update.                                                             ║
║                                                                              ║
║ UEFI CA and all connected devices may not be usable while updating.          ║
╚══════════════════════════════════════════════════════════════════════════════╝
```

JohnAZoidberg, 2025-07-24:
Everything looks good now, maybe we could try to set the vendor information for our own DB/KEK keys.
I tried to, but it doesn't quite work.
The version I assume it gets from the certificate's issue data, that's ok.
But why does it still say unknown vendor?
This is with fwupd 2.0.12

```
> cat /var/lib/fwupd/quirks.d/test.quirk
[UEFI\CRT_AF5D1E0FC07FB462B5A1C59EF7E456A39CAC9AE1]               
VendorId = UEFI:FrameworkComputer                                                                                                                                                 

> sudo fwupdtool get-devices --plugins uefi_db                                                                                               
Loading…                 [************************************** ]                                                                                                                
Framework Laptop 13 (Intel Core Ultra Series 1)                                                                                                                                   
│                                                                                        
└─UEFI Signature Database:                                                               
  │   Device ID:          0352a8acc949c7df21fec16e566ba9a74e797a97                       
  │   Device Flags:       • Internal device                                              
  │                                         
  ├─Database Key:                                                                        
  │     Device ID:        0e3b4bfc8e1d26fa79855e17c4a13e4f4ddf316f
  │     Current version:  2024         
  │     Vendor:           Unknown (UEFI:UNKNOWN, UEFI:FrameworkComputer)
  │     GUID:             866f72e6-a32e-505c-bca6-f6981dc2a4e2 ← UEFI\CRT_AF5D1E0FC07FB462B5A1C59EF7E456A39CAC9AE1
  │     Device Flags:     • Internal device
  │                       • Updatable
  │                       • Needs a reboot after installation
  │                       • Full disk encryption secrets may be invalidated when updating
  │                       • Signed Payload
  │                       • Can tag for emulation
```

hughsie, 2025-07-25:
I guess your cert has no subject vendor set?

	fu_device_build_vendor_id(device,
				  "UEFI",
				  subject_vendor != NULL ? subject_vendor : "UNKNOWN");

Remind me, how are you setting the vendor ID for your cert? From the quirk file perhaps?

cgoesche, 2025-07-25:
After trying this on a HP Probook 445 G9 the UEFI seems to be stuck in a boot loop even after draining the battery and resetting the CMOS 🥲 . If needed I can create a new issue for this, just not to break the ongoing thread context and provide more information there.

JohnAZoidberg, 2025-07-25:
> I guess your cert has no subject vendor set?

We do have the subject field set, of course.
What's a subject vendor? `OU=`?
For example we have `Subject: CN=frame.work-LaptopKEK`.

JohnAZoidberg, 2025-07-26:
> fwupd is expecting to see `O=`

Right, I thought so. But why doesn't the quirk override that?

hughsie, 2025-07-26:
> But why doesn't the quirk override that?

It's appending a list -- maybe try ^^

theofficialgman, 2025-07-31:
@JohnAZoidberg The microsoft KEK update doesn't seem to have occured on your system even after resolving the issue with the DB update. I also have this problem

```
Devices with no available firmware updates: 
 • KEK CA
```

JohnAZoidberg, 2025-07-31:
I extracted the cab from https://fwupd.org/lvfs/devices/com.microsoft.kek.mcj.bf5a987e.firmware
And there are two metainfo files - why two?

They have two different binaries and they have two different firmware hashes in requires.
e8010bb5-29dc-5c38-ba6a-f7362ab8f4bc
and
6b760cae-6185-520a-b67c-2b8ba079d1c6

I don't have those, that's why it refuses to install:

```
> sudo fwupdtool get-devices --plugins uefi_kek
Framework Laptop 13 (Intel Core Ultra Series 1)
│                    
└─UEFI Key Exchange Key:                                                                 
  │   Device ID:          2a4c23bfb79b5dabe474cb7b1b3e604645d6f9c6
  │   Device Flags:       • Internal device                                              
  │                                                                                                                                                                               
  ├─KEK CA:                                
  │     Device ID:        b7a1d3d90faa1f6275d9a98da4fb3be7118e61c7
  │     Current version:  2011                                                           
  │     Vendor:           Microsoft (UEFI:Microsoft)                       
  │     GUIDs:            814e950f-1449-566a-a190-42c9d3a3a2df ← UEFI\VENDOR_Microsoft&NAME_Microsoft-KEK-CA
  │                       dfa66406-6568-5bdf-bb8e-b53ddb4be4cf ← UEFI\CRT_9F402B1CC0243CBEDC58A525789816CCCA7687A9
  │     Device Flags:     • Internal device                                              
  │                       • Updatable
  │                       • Needs a reboot after installation                                                                                                                     
  │                       • Device is usable for the duration of the update
  │                       • Full disk encryption secrets may be invalidated when updating 
  │                       • Signed Payload                                               
  │                       • Can tag for emulation                                                                                                                                 
  │   
  └─Key Exchange Key:
        Device ID:        7227626d9209595851b3840e3d626601a375ce2d
        Current version:  2024
        Vendor:           Unknown (UEFI:UNKNOWN, UEFI:FrameworkComputer)
        GUID:             6b3f091b-067c-55c3-9c84-393aa4f44c05 ← UEFI\CRT_3BB7B137630F02EF8DA913573061E3226970B06C
        Device Flags:     • Internal device
                          • Updatable
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Full disk encryption secrets may be invalidated when updating 
                          • Signed Payload
                          • Can tag for emulation
```

```
> sudo fwupdtool install 36ea945cf1cf357df5ff30096e228dcab4c83f9e3438ecb475a3e10af19
ff44f-KEK-MCJ.cab 
No other device e8010bb5-29dc-5c38-ba6a-f7362ab8f4bc found
```

JohnAZoidberg, 2025-08-07:
Ok you're right, confirmed with Microsoft. We'll work with them.

closed 2025-08-07 by JohnAZoidberg