fwupd/fwupd #9071: How to solve "failed to detach: failed to wait for detach replug: failed to wait for detach replug: device {id} did not come back" when fwupdmgr update to upgrade firmware?

1202jason, 2025-07-25:
Describe the bug
I want to use the local cab package to update or downgrade the firmware through fwupdmgr, but the output states: "failed to detach: failed to wait for detach replug: failed to wait for detach replug: device {id} did not come back".The current version is 2.0.9, and I successfully performed the same operation in version 2.0.8 to update or downgrade the firmware.

Steps to Reproduce
1.fwupdate --version
2.sudo fwupdmgr local-install 20250723-test-T99W696.FDE.F0.3.2.0.3.005.cab --allow-reinstall -v
3.fwupdate get-devices

sudo fwupdmgr --version 
[sudo] password for jbd: 
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.9
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.15.0-next-20250527-dirty
runtime   org.freedesktop.fwupd         2.0.9

sudo fwupdmgr local-install 20250723-test-T99W696.FDE.F0.3.2.0.3.005.cab --allow-reinstall -v
(pkttyagent:4957): GLib-GIO-DEBUG: 16:46:02.173: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:4955): GLib-GIO-DEBUG: 16:46:02.182: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.182: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.182: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.183: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.183: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.183: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:4955): GLib-GIO-DEBUG: 16:46:02.183: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:4955): GLib-GIO-DEBUG: 16:46:02.184: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:4955): GLib-GIO-DEBUG: 16:46:02.187: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.188: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.188: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.188: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.188: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:4955): dconf-DEBUG: 16:46:02.188: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:02.189: Emitting ::status-changed() [idle]
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:02.196: Emitting ::status-changed() [decompressing]
Decompressing…           [                            /          ](fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.357: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [                            /          ](fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.370: Emitting ::status-changed() [device-busy]
Waiting…                 [                            /          ](fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.370: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.370: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.371: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.371: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
Waiting…                 [                                  -    ](fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.641: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.641: Emitting ::status-changed() [decompressing]
Decompressing…           [                                       ](fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.845: Emitting ::status-changed() [device-restart]
Restarting device…       [                                       ](fwupdmgr:4955): Fwupd-DEBUG: 16:46:06.866: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:16.868: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:16.876: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:16.877: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:16.886: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:16.886: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:16.888: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)
(fwupdmgr:4955): Fwupd-DEBUG: 16:46:16.889: Emitting ::device-changed(3cc7caba52368ac6bf70fbdeb48187b7eed99fe9)

failed to detach: failed to wait for detach replug: device 3cc7caba52368ac6bf70fbdeb48187b7eed99fe9 did not come back

sudo fwupdmgr get-devices 
Dell Inc. Dell Pro 14 Plus PB14250
│
├─Core™ Ultra 7 268V:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000011b
│     Vendor:             Intel
│     GUIDs:              7e515fa5-57fb-557c-a30e-b27e38b18a30 ← CPUID\PRO_0&FAM_06&MOD_BD
│                         5f87d1bb-65ae-59e0-82cd-1c846105f12d ← CPUID\PRO_0&FAM_06&MOD_BD&STP_1
│     Device Flags:       • Internal device
│   
├─PC811 SED SK hynix 2048GB:
[... 82 lines cut ...]
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│                         • Can tag for emulation
│     
├─T99W696:
│     Device ID:          3cc7caba52368ac6bf70fbdeb48187b7eed99fe9
│     Summary:            Mobile broadband device
│     Current version:    FDE.F0.3.2.0.3.DF.001.005
│     Vendor:             foxconn (USB:0x17CB)
│     Update State:       Failed
│     Update Error:       failed to detach: failed to wait for detach replug: device 3cc7caba52368ac6bf70fbdeb48187b7eed99fe9 did not come back
│     Last modified:      2025-07-25
│     GUIDs:              dc406d8f-072e-511c-8725-459eb7a511e8 ← PCI\SSVID_105B&SSPID_E142&REV_0000&CARRIER_DEFAULT
│                         589094f5-ebb5-55c9-bd80-31cd278e2a3e ← PCI\VID_17CB&PID_0308&REV_0000&CARRIER_DEFAULT
│                         7f9b8503-2eb6-5369-83b9-b5a63564297c ← PCI\VID_17CB&PID_0308&REV_0000
│                         41b7a2f8-46e0-5ad5-b448-31594d4ea2ef ← PCI\VID_17CB&PID_0308
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Unsigned Payload
│                         • Can tag for emulation

SlarkXiao, 2025-07-28:
Hi @hughsie ,
It seems a regression issue from v2.0.8 to v2.0.9. We can do a normal firmware upgrade in v2.0.8, but always fail in v2.0.9.
Please take a look on this.

lvoegl, 2025-08-22:
```
13:38:24.509 Mbim                 [/dev/cdc-wdm0] opening device...
13:38:24.510 Mbim                 [/dev/cdc-wdm0] read max control message size from descriptors file: 4096
13:38:24.541 Mbim                 [/dev/cdc-wdm0] version information update reported: version 1.00, extended version 3.00
13:38:24.541 FuPlugin             emit added from modem_manager: 574f4e2ba7ee0c1eee6004bfc7ef87547039adea

13:38:24.544 Mbim                 [/dev/cdc-wdm0] closing device...
13:38:26.046 Mbim                 [/dev/cdc-wdm0] channel destroyed
13:38:26.049 FuDeviceLocker       failed to close device: Operation was cancelled
```
happens at every `fu_mm_mbim_device_close_sync` for me. Maybe has to do with the recently added `g_main_context` (a4f871d2142e62a6e8b3b8ce3b05d8187e96122f)?

lvoegl, 2025-08-25:
Would you be able to try out if #9155 fixes your issue?

SlarkXiao, 2025-08-27:
The root cause may come from chunks didn't do an update like version fwupd 2.0.8.
For each QDU write, it writes same data with the first one. No matter the second, the third or even the last.
So device would return an FAILURE for last QDU write since checksum failure in device side.

[image]

@lvoegl @hughsie Can you help check if the logical of chunks part ? It should have updated when QDU write completed.

SlarkXiao, 2025-08-27:
BTW, the marked data was the first 16 bytes of firmware images:

[image]

SlarkXiao, 2025-08-27:
https://github.com/fwupd/fwupd/pull/9165 could fix above issue. And I got a PASS with @lvoegl  's branch with this fix.

PR #9155 (2025-08-28): modem-manager: per-device main context
Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation

All MBIM message responses are sent to the main context the device was opened with.
If using a new context for each MBIM action (open/command/close), we are never going to see any response (except for the first time).
-> Use a single context per MBIM device and reuse it for every of its MBIM actions.

Fixes: #9071
closed 2025-08-28 by hughsie