fwupd/fwupd #9084: Failed update ()

jabcross, 2025-07-30:
**Describe the bug**
Firmware update fails with following message

```
bash-5.2$ fwupdmgr update
Devices with no available firmware updates: 
 • KINGSTON SNV2S1000G
 • Laptop Webcam Module (2nd Gen)
 • UEFI dbx
 • USB4 Retimer
Devices with the latest available firmware version:
 • Fingerprint Sensor
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 2141 to 2518?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ Update Intel CSME driver to 18.0.15.2518.                                    ║
║                                                                              ║
║ Laptop 13 (Intel Core Ultra Series 1) must remain plugged into a power       ║
║ source for the duration of the update to avoid damage.                       ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Decompressing…           [***************************************]
Specified firmware is older than installed '0x9d6 < 2141'
```

It looks like it's string-comparing the hex of 2518 and the decimal of 2141.

<details>
get-report-metadata
```
Loading…                 [********************                   ]01:12:08.090 FuPluginIntelMe      failed to get public key using file-id 0x40002300: generic failure [0xb]
Waiting…                 [*************************************  ]
BatteryLevel:           86
BatteryThreshold:       10
BootTime:               1753699505
CompileVersion(com.hughsie.libjcat):0.2.2
CompileVersion(com.hughsie.libxmlb):0.3.22
CompileVersion(org.freedesktop.Passim):0.1.10
CompileVersion(org.freedesktop.fwupd):1.9.31
CompileVersion(org.freedesktop.gusb):0.4.9
CpuArchitecture:        x86_64
CpuModel:               Intel Core™ Ultra 5 125H
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 41 (Sway)
DistroVariant:          sway
DistroVersion:          41
FwupdSupported:         True
HostBaseboardManufacturer:Framework
HostBaseboardProduct:   FRANMECP02
HostBiosMajorRelease:   03
HostBiosMinorRelease:   04
HostBiosVendor:         INSYDE Corp.
HostBiosVersion:        3.04
HostEnclosureKind:      a
HostFamily:             Laptop
HostFirmwareMajorRelease:ff
HostFirmwareMinorRelease:ff
HostProduct:            Laptop 13 (Intel Core Ultra Series 1)
HostSku:                FRANDPCP02
HostVendor:             Framework
KernelCmdline:          pci=realloc=off rd.driver.blacklist=nouveau,nova_core modprobe.blacklist=nouveau,nova_core
KernelName:             Linux
KernelVersion:          6.15.8-200.vanilla.fc41.x86_64
LidState:               open
PassimDownloadSaving:   1434994
PowerState:             ac-charging
[... 508 lines cut ...]
│     GUID:               00cf42fa-f402-5c7c-8767-4e626b24cd8f ← UEFI\CRT_FEE3D28E3DA7EA45A996D9B5261A35C1D204E42E
│   
└─USB4 Retimer:
      Device ID:          8f92f06a1ff9413dbdd56b2337683e333ab56574
      Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
      Current version:    24.82
      Vendor:             Intel (THUNDERBOLT:0x8087)
      GUIDs:              74fb1b87-53e6-5115-bd02-a9fefd06ed9a ← THUNDERBOLT\VEN_8087&DEV_0D9C
                          92ca5f32-c2d9-5bda-8765-a467b169e2fd ← TBT-80870d9c-retimer0-0:3.1
      Device Flags:       • Internal device
                          • Updatable
                          • Device stages updates
                          • Unsigned Payload
```
</details>

**Steps to Reproduce**
Be on version 2141, update to 2518

**Expected behavior**
No error

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Dont remember. It is Fedora 41, the one recommended by Framework at the time.

tmuehlbacher, 2025-07-30:
Could just be 68ce789667461b206a4e9ff5f38f794bace3bb07?

jabcross, 2025-07-31:
Update: Doing `fwupdmgr refresh --force` apparently solves it. But I had to dig around a bit to find this out.

JohnAZoidberg, 2025-08-04:
> Update: Doing `fwupdmgr refresh --force` apparently solves it. But I had to dig around a bit to find this out.

oh I think the original version we uploaded had the wrong format. So if you had an index that wasn't updated for a while,  that might have been it.
@quinchou77 fixed it quite quickly, so you must've been pretty unlucky.