fwupd/fwupd #9092: Match replugged Firehose devices

lvoegl, 2025-08-01:
**Describe the bug**
When switching _USB_ QCDM devices into EDL mode, they are successfully added to the `qc-firehose` plugin, but the modem-manager plugin fails with `device did not come back` due to the device ids mismatching.
```
12:20:14.892 FuEngine             /org/freedesktop/ModemManager1/Modem/8 added FuMmQcdmDevice:
  Name:                 QUECTEL Mobile Broadband Module
  Summary:              Quectel EG25-G/EC25 modem
  Protocol:             com.qualcomm.firehose
  ...
  PhysicalId:           /sys/devices/pci0000:00/0000:00:13.0/usb1/1-1/1-1.3
  BackendId:            /org/freedesktop/ModemManager1/Modem/8
  ...

12:20:14.892 FuPlugin             backend_device_added(modem_manager)
12:20:14.892 FuDevice             using 296ac3c0ba7d9ab8257a6bad5ebd2b8df243e1eb for /sys/devices/pci0000:00/0000:00:13.0/usb1/1-1/1-1.3
12:20:14.893 FuPlugin             emit added from modem_manager: 296ac3c0ba7d9ab8257a6bad5ebd2b8df243e1eb

12:20:38.044 FuDeviceList         waiting 5000ms for replug

12:20:38.825 FuEngine             /sys/devices/pci0000:00/0000:00:13.0/usb1/1-1/1-1.3 added FuUsbDevice:
  Name:                 Gobi Wireless Modem (QDL mode)
  Flags:                can-emulation-tag
  ...
  PhysicalId:           1-1.3
  BackendId:            /sys/devices/pci0000:00/0000:00:13.0/usb1/1-1/1-1.3
  ...

12:20:38.834 FuDevice             using 88a187f1bf16fcf94a6b0648feff8fcdc92e33ab for 1-1.3
12:20:38.836 FuPlugin             emit added from qc_firehose: 88a187f1bf16fcf94a6b0648feff8fcdc92e33ab
```

**Expected behavior**
The modem-manager device should be matched and reused for the qc-firehose plugin to successfully wait for a replug, just like with mhi-pci devices:
```
11:35:26.713 FuDeviceList         waiting 5000ms for replug
11:35:26.794 FuEngine             auto-removing backend device RM520N-GL [5b82190ce5dcf46fdcfed3fbf42b96a95996f361]
11:35:26.798 FuDeviceList         waiting 5000ms for RM520N-GL device removal
11:35:26.798 FuEngine             resetting update motd timeout
11:35:28.947 FuDeviceList         found compatible device 5b82190ce5dcf46fdcfed3fbf42b96a95996f361 recently removed, reusing item from plugin modem_manager for plugin qc_firehose
```

I've looked at setting a counterpart, but I don't think that works in that case.

hughsie, 2025-08-01:
Ohh it's VID_05C6&PID_9008 again! What about something like this?

    diff --git a/plugins/modem-manager/fu-mm-qcdm-device.c b/plugins/modem-manager/fu-mm-qcdm-device.c
    index c28a0e40b..9ebd1dab6 100644
    --- a/plugins/modem-manager/fu-mm-qcdm-device.c
    +++ b/plugins/modem-manager/fu-mm-qcdm-device.c
    @@ -128,6 +128,9 @@ fu_mm_qcdm_device_init(FuMmQcdmDevice *self)
     {
            fu_udev_device_add_open_flag(FU_UDEV_DEVICE(self), FU_IO_CHANNEL_OPEN_FLAG_READ);
            fu_udev_device_add_open_flag(FU_UDEV_DEVICE(self), FU_IO_CHANNEL_OPEN_FLAG_WRITE);
    +       fu_device_add_instance_id_full(FU_DEVICE(self),
    +                                      "USB\\VID_05C6&PID_9008",
    +                                      FU_DEVICE_INSTANCE_FLAG_COUNTERPART);
            fu_device_add_protocol(FU_UDEV_DEVICE(self), "com.qualcomm.firehose");
     }

hughsie, 2025-08-04:
> It does work if I temporarily remove the remove_id == 0 check

I think this is because remove_delay isn't being set in `FuMmQcdmDevice`. If you move the `fu_device_set_remove_delay()` from `fu_mm_mhi_qcdm_device_init` to the baseclass `fu_mm_qcdm_device_init` I think it might work. I'll do a PR as I'm pretty confident this should work.

lvoegl, 2025-08-04:
But the backend id of an `FuMmMhiQcdmDevice` should be quite similar so I don't understand exactly what it's removing.

lvoegl, 2025-08-04:
> If the modem device disappears completely, I think it's fine.

Thinking about it again I agree. Similar to Fastboot, we only send a detach command which makes the device disappear from ModemManager anyways.

hughsie, 2025-08-04:
Right, I think we're on the same page. I've updated hughsie/modems-qcdm -- can you give that a go again please?