fwupd/fwupd #9118: Razer blade stealth cannot boot after fwupd updates

sydhds, 2025-08-09:
**Describe the bug**

I have the following message when booting my Razer laptop after fwupd did some updates (dbx updates):

```
istRT: Volume Full                                                                                    Could not creage MokL           
Could not create MokListXRT: Volume Full
Could not create SbatLevelRT: Volume Full
Could not create MokListTrustedRT: Volume Full
Something has gone seriously wrong: import_mok_state() failed: Volume Full
```

I cannot boot on a bootable usb key (ubuntu usb key):

```
Could not create MokListRT: Volume Full
Could not create MokListXRT: Volume Full
Could not create SbatLevelRT: Volume Full
Could not create MokListTrustedRT: Volume Full
Something has gone seriously wrong: import_mok_state() failed: Volume Full
```

This is on a Razer Balde Stealth 2028 (with Intel i7-7500U - 8Gb).

**fwupd version information**

If I remember correcltly, updates was done on Ubutun 25.04 +  snap fwupd version 2.0.13 and after updating DBX.

**Additional questions**

I've tried disabling secure boot but the same message is displayed.

Anything I can do to unbrick my laptop?

sydhds, 2025-09-01:
> > I've tried disabling secure boot but the same message is displayed.
> 
> Can you get into the BIOS and reset all the settings to the "factory defaults"? That should clear the NVRAM.

There was no such option in the BIOS but I've opened my laptop and remove the CMOS battery for a while. My laptop is booting fine now.

Can I provide more info in order to resolve or mitigate this issue?

sydhds, 2025-09-04:
fwupdtool hwids : 

```
Loading…: 0%
Loading…: 1%
Loading…: 2%
Loading…: 5%
Loading…: 6%
Computer Information
--------------------
BiosVendor: Razer
BiosVersion: 8.02
BiosMajorRelease: 8
BiosMinorRelease: 2
FirmwareMajorRelease: 02
FirmwareMinorRelease: 00
Manufacturer: Razer
Family: 1A586752
ProductName: Blade Stealth
ProductSku: 
EnclosureKind: 9
BaseboardManufacturer: Razer
BaseboardProduct: Razer
Hardware IDs
------------
{143c957c-633f-5bb6-b598-d04e92da5665}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{bdd0b0f9-0b66-54eb-910c-23074304bcc1}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{931b4737-2418-5071-a95e-2a8c7210a8aa}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{bf1c8664-e001-5478-9972-9bd1332776d2}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{c5a29236-b7ea-5b36-9d17-2662546c0109}   <- Manufacturer + Family + ProductName + ProductSku
{f4e7bbbc-8d01-5fe8-a7b4-dc7dbf4e461b}   <- Manufacturer + Family + ProductName
{7309630c-f88d-5b15-8e76-7f524edcb7d5}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{606fa93b-4464-599e-b717-7c9236774e0a}   <- Manufacturer + ProductSku
{0c6d20ad-512b-5bff-8e16-4c74692c6d7e}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{66923440-8c37-53af-adb8-efc450f2a7ca}   <- Manufacturer + ProductName
{66b19135-087f-52df-8a73-dbb5f9f26783}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{17ade54a-387d-5683-ad3c-129dbfc7dbe7}   <- Manufacturer + Family
{ca8eb16f-1bb0-5ee3-97f3-55d7780ff66c}   <- Manufacturer + EnclosureKind
{f4c7f548-1139-5890-b261-e8940863536c}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{d4d9d9d7-c86c-53d9-bb5b-3d351772a8f3}   <- Manufacturer
Extra Hardware IDs
------------------
{0297b6fc-4308-59cf-8d48-f962909e5de9}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{a50d240d-de17-5a74-a5b6-ce4c84596966}   <- Manufacturer + Family + ProductName + BiosVendor
{5e0aa2ca-0d55-54f2-84d9-b4e9935dbcff}   <- Manufacturer + BiosVendor
```

fwupdtool get-report-metadata:

```
Loading…: 0%
Loading…: 1%
Loading…: 2%
Loading…: 3%
Loading…: 4%
Loading…: 5%
Loading…: 6%
Loading…: 7%
Loading…: 8%
Loading…: 9%
Loading…: 0%
Loading…: 10%
Loading…: 11%
Loading…: 12%
Loading…: 13%
Loading…: 28%
Loading…: 30%
Loading…: 31%
Loading…: 32%
Loading…: 33%
[... 232 lines cut ...]
│                         • Can tag for emulation
│     
└─TPM:
      Device ID:          1d8d50a4dbc65618f5c399c2ae827b632b3ccc11
      Current version:    5.0.4.16642
      Vendor:             Infineon (TPM:IFX)
      GUIDs:              9787c20b-de99-5e3b-8a72-e4fba23d93e1 ← TPM\VEN_IFX&DEV_0001
                          1a10e308-1ceb-5b61-bbd0-07461469ab8f ← TPM\VEN_IFX&MOD_SLB9665
                          f2957cd7-63f3-56bb-b7ef-c7c60c31c21e ← TPM\VEN_IFX&DEV_0001&VER_2.0
                          5cec2f57-3b14-5f7d-9436-882896e3923d ← TPM\VEN_IFX&MOD_SLB9665&VER_2.0
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device can recover flash failures
                          • Signed Payload
                          • Can tag for emulation
```

OS version: Ubuntu 25.04
Fwupd: 

```
snap list | grep fw
fwupd                           2.0.14                          7567      latest/candidate  richard-hughes  -
```

hughsie, 2025-09-04:
Many thanks for providing that -- I've added all Razer blade stealth models to the "do not attempt" list.

closed 2025-09-04 by hughsie