fwupd/fwupd #9126: Security policy violation when updating Intel ME firmware on Lenovo T14 Gen 1

jcharaoui, 2025-08-11:
**Describe the bug**

I'm unable to install an Intel ME firmware upgrade on my Lenovo T14 Gen 1 (type 20S1).

**Steps to Reproduce**

Run `sudo fwupdmgr update`

**Expected behavior**

Firmware installed after reboot.

**fwupd version information**

```
BatteryLevel:           78
BatteryThreshold:       25
BootTime:               1754951215
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.fwupd): 2.0.8
CpuArchitecture:        x86_64
CpuModel:               Intel Core™ i5-10210U CPU @ 1.60GHz
DisplayState:           connected
DistroId:               debian
DistroName:             Debian GNU/Linux
DistroPrettyName:       Debian GNU/Linux 13 (trixie)
DistroVersion:          13
FwupdSupported:         True
HostBaseboardManufacturer: LENOVO
HostBaseboardProduct:   20S1S5RP00
HostBiosMajorRelease:   01
HostBiosMinorRelease:   20
HostBiosVendor:         LENOVO
HostBiosVersion:        N2XET42W (1.32 )
HostEnclosureKind:      a
HostFamily:             ThinkPad T14 Gen 1
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 10
HostProduct:            20S1S5RP00
HostSku:                LENOVO_MT_20S1_BU_Think_FM_ThinkPad T14 Gen 1
HostVendor:             LENOVO
KernelName:             Linux
KernelVersion:          6.12.38+deb13-amd64
LidState:               open
PlatformArchitecture:   x86_64
PowerState:             ac-charging
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.8
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.12.38+deb13-amd64
DeviceId:               1ef2f004def59a66de0db75401014c567a8273ea
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               1f1e4ba4a7d41080bc25eb528f93465c4ab89bbd
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
[... 377 lines cut ...]
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b 78 a8 4a af 2b 2a fc 4e a7 9c f5 cc 8f 3d 38 03 / 7f ff 04 00
Boot0021* LENOVO CLOUD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri(https://download.lenovo.com/pccbbs/cdeploy/efi/boot.efi)
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b ad 38 cc bb f7 ed f0 4d 95 9c f4 2a a7 4d 36 50 / 03 18 3b 00 68 74 74 70 73 3a 2f 2f 64 6f 77 6e 6c 6f 61 64 2e 6c 65 6e 6f 76 6f 2e 63 6f 6d 2f 70 63 63 62 62 73 2f 63 64 65 70 6c 6f 79 2f 65 66 69 2f 62 6f 6f 74 2e 65 66 69 / 7f ff 04 00
Boot0022  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
      dp: 03 0a 25 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b ae a2 09 0a df de 21 4e 8b 3a 5e 47 18 56 a3 54 06 / 7f ff 04 00
Boot0023  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
      dp: 03 0a 25 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b 91 af 62 59 56 44 9f 41 a7 b9 1f 4f 89 2a b0 f6 06 / 7f ff 04 00
Boot0024* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 0b 01 / 7f ff 04 00
Boot0025* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 0b 00 / 7f ff 04 00
Boot0026* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b 91 af 62 59 56 44 9f 41 a7 b9 1f 4f 89 2a b0 f6 / 7f ff 04 00
Boot0027* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b ae a2 09 0a df de 21 4e 8b 3a 5e 47 18 56 a3 54 / 7f ff 04 00
```

**Additional questions**

- Operating system and version: Debian 13
- Have you tried rebooting? Yes
- Is this a regression? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? I don't know

jcharaoui, 2025-08-12:
I've got the `fwupd` package at 2.0.8-3 here. I'm not sure how to determine what version of the "shim" I'm running.

```
$ mokutil --list-sbat-revocations
sbat,1,2024010900
shim,4
grub,3
grub.debian,4
```

mrhpearson, 2025-08-14:
OK - I'll have to check out what Debian uses. Been meaning to try out Deb 13 so it's a good excuse.

Can you confirm the update works with secure boot disabled please?

closed 2025-08-27 by hughsie