fwupd/fwupd #9137: Debian 13: fwupdmgr update does not honor DisableShimForSecureBoot

mschmitt, 2025-08-14:
**Describe the bug**

`fwupdmgr update` does not honor `DisableShimForSecureBoot` in the `uefi_capsule` section in `fwupd.conf`

We successfully used  `fwupdmgr update` before on Ubuntu 22.04 and Debian 12, with this entry in `/etc/fwupd/uefi_capsule.conf`.

**Steps to Reproduce**

The system uses `systemd-boot` and performs all Secureboot signatures using a locally-generated platform key. Grub and Shim are not installed, artifacts of them may however be lingering in `/boot` and are not touched by `efibootmgr`. SECTPMCTL, which appears in the data collection below, is published here: https://github.com/telekom-mms/sectpmctl

On running `fwupdmgr update`, I get the error that Shim-Signed is not installed. (Can't reproduce the wording right now without downgrading firmware.)

The `uefi_capsule` section has been added to `fwupd.conf`:

```
        echo "Disabling grub-shim for fwupd."
        crudini --set /etc/fwupd/fwupd.conf uefi_capsule DisableShimForSecureBoot true
```

If `fwupdtool update` is used instead of `fwupdmgr update`, the update succeeds.

We successfully used  `fwupdmgr update` before on Ubuntu 22.04 and Debian 12, with the identical entry in `/etc/fwupd/uefi_capsule.conf`:

```
	echo "Disabling grub-shim for fwupd."
	crudini --set --existing=section /etc/fwupd/uefi_capsule.conf uefi_capsule DisableShimForSecureBoot true
```

Signing of `/usr/libexec/fwupd/efi/fwupdx64.efi.signed` using the platform key is hooked into the package manager.

**Expected behavior**

`fwupdmgr update` uses the locally-signed `/usr/libexec/fwupd/efi/fwupdx64.efi.signed` directly instead of attempting to use the shim.

**fwupd version information**

Please provide the version of the daemon and client.

```shell
# fwupdtool get-report-metadata
CompileVersion(org.freedesktop.fwupd): 2.0.8
CpuArchitecture:        x86_64
CpuModel:               Intel 11th Gen Intel Core™ i7-11850H @ 2.50GHz
DisplayState:           connected
DistroId:               debian
DistroName:             Debian GNU/Linux
DistroPrettyName:       Debian GNU/Linux 13 (trixie)
DistroVersion:          13
FwupdSupported:         True
HostBaseboardManufacturer: LENOVO
HostBaseboardProduct:   20YRS0J600
HostBiosMajorRelease:   01
HostBiosMinorRelease:   24
HostBiosVendor:         LENOVO
HostBiosVersion:        N37ET55W (1.36 )
HostEnclosureKind:      a
HostFamily:             ThinkPad P15 Gen 2i
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 0c
HostProduct:            20YRS0J600
HostSku:                LENOVO_MT_20YR_BU_Think_FM_ThinkPad P15 Gen 2i
HostVendor:             LENOVO
KernelName:             Linux
KernelVersion:          6.12.41+deb13-amd64
LidState:               open
PlatformArchitecture:   x86_64
PowerState:             ac-fully-charged
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.8
[... 891 lines cut ...]
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img-6.12.38+deb13-amd64
├── initrd.img-6.12.41+deb13-amd64
├── lost+found
├── System.map-6.12.38+deb13-amd64
├── System.map-6.12.41+deb13-amd64
├── vmlinuz-6.12.38+deb13-amd64
└── vmlinuz-6.12.41+deb13-amd64
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting? -> Sure
- Is this a regression? -> No idea
- Are you using an NVMe disk? -> Yes
- Is secure boot enabled? -> Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? -> No

superm1, 2025-08-14:
Share your fwupd.conf please.

mschmitt, 2025-08-14:
While I was doing a firmware downgrade, it suddenly worked using `fwupdmgr`. Is this a side effect because `fwupdtool` succeeded once and did things in `efibootmgr`, or was I short of a reboot for `fwupd.service` to pick up the configuration? I'll reinstall the system and give it another try.

superm1, 2025-08-14:
Fwupd.service needs to be reset to pick up this specific configuration item.

mschmitt, 2025-08-14:
Yep, it works now. I was so lost between the new config location and updating my packages to reflect it, that the final shutdown/boot cycle that's usually implied after installation was actually missing. Thanks everyone!