fwupd/fwupd #9150: framework-qmk: failed to wait for detach replug

JohnAZoidberg, 2025-08-20:
**Describe the bug**
I can't find a pattern but sometimes fwupd doesn't notice that the devices comes back in bootloader mode.
I don't think it's a timeout issue, if I run lsusb while fwupd is still waiting for the device to come  back, I can see that it's already back.

**Steps to Reproduce**
`fwupdmgr update` or `fwupdmgr install` with Framework 16 QMK keyboard.

**Expected behavior**
I expect the update to finish.

**fwupd version information**

Built from source:

```
> fwupdmgr --version
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.14
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.10
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.6
runtime   org.freedesktop.Passim        0.1.10
runtime   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.fwupd.source  2.0.13-135-g5f3d5a53a
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.15.6-200.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.14
```

```shell
> fwupdtool get-report-metadata
                                          fwupdtool hwids
Loading…                 [*************************************  ]
WARNING: This package has not been validated, it may not work properly.
Waiting…                 [*************************************  ]
BatteryLevel:           89
BatteryThreshold:       10
BootTime:               1755678809
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.Passim): 0.1.10
CompileVersion(org.freedesktop.fwupd): 2.0.14
CompileVersion(org.freedesktop.fwupd.source): 2.0.13-135-g5f3d5a53a
CpuArchitecture:        x86_64
CpuModel:               AMD Ryzen 9 7940HS w/ Radeon 780M Graphics
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 42 (Workstation Edition)
DistroVariant:          workstation
DistroVersion:          42
EfivarsNvramFree:       35187
EfivarsNvramUsed:       77366
FwupdSupported:         False
HostBaseboardManufacturer: Framework
HostBaseboardProduct:   FRANMZCP09
HostBiosMajorRelease:   03
HostBiosMinorRelease:   06
HostBiosVendor:         INSYDE Corp.
HostBiosVersion:        3.06 NV0A
HostEnclosureKind:      a
HostFamily:             16in Laptop
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            Laptop 16 (AMD Ryzen 7040 Series)
HostSku:                FRANMZCP09
HostVendor:             Framework
KernelCmdline:          modprobe.blacklist=nouveau,nova_core rd.driver.blacklist=nouveau,nova_core
[... 376 lines cut ...]
      Summary:            NVM Express solid state drive
      Current version:    61503000
      Vendor:             Sandisk (PCI:0x15B7)
      Serial Number:      215176441401
      Problems:           • Device requires AC power to be connected
      GUIDs:              887c0622-74ea-5036-99bd-c6c63082bd37 ← NVME\VEN_15B7&DEV_5011
                          b029d0ec-3f85-5cb7-a403-85c6f3bb4d83 ← NVME\VEN_15B7&DEV_5011&SUBSYS_15B75011
                          58b8f9eb-a2a8-5755-abb9-61c92edb716c ← WDC PC SN810 SDCPNRY-1T00
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version: Fedora 42
- Have you tried rebooting?
  -  Yes
- Is this a regression?
  - I don't think so, I saw it ocassionally before, but then I increased the timeout. Now I realize it's not a timeout issue.

hughsie, 2025-08-20:
I think I need a `fwupd -vv` log showing the behaviour please.

JohnAZoidberg, 2025-08-21:
Here are the logs from two invocations of ` fwupdtool install framework_ansi_0.31.cab --allow-reinstall -vv`
One time it failed, one time it succeeded.

[framework-qmk-replug-fail.txt](https://github.com/user-attachments/files/21913237/framework-qmk-replug-fail.txt)

[framework-qmk-replug-success.txt](https://github.com/user-attachments/files/21913239/framework-qmk-replug-success.txt)

JohnAZoidberg, 2025-08-21:
The logs shows that the device comes back in bootloader mode as RP2 in 3 seconds, much less than the 15s timeout.

caffineehacker, 2025-11-25:
Here's my `udevadm monitor` output

```
~ ❱ sudo udevadm monitor
Place your finger on the fingerprint reader
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[5765.105438] change   /devices/pci0000:00/0000:00:08.1/0000:c4:00.3/usb1/1-4/1-4.1 (usb)
KERNEL[5765.196954] change   /devices/pci0000:00/0000:00:08.1/0000:c4:00.3/usb1/1-4/1-4.1 (usb)
UDEV  [5765.203241] change   /devices/pci0000:00/0000:00:08.1/0000:c4:00.3/usb1/1-4/1-4.1 (usb)
UDEV  [5765.209658] change   /devices/pci0000:00/0000:00:08.1/0000:c4:00.3/usb1/1-4/1-4.1 (usb)
```

And the full log of the update with -vvv. I didn't see any other option in --help or the man page for more verbosity than that

[sudo fwupdmgr update.txt](https://github.com/user-attachments/files/23753845/sudo.fwupdmgr.update.txt)

hughsie, 2026-03-02:
Hmm, has anyone seen this recently or can we close?

caffineehacker, 2026-03-02:
Still reproducible 100% of the time for me. Latest test on version 2.0.19 of fwupd

JohnAZoidberg, 2026-03-15:
I also tried `sudo DBUS_SYSTEM_BUS_ADDRESS=unix:path=/dev/null fwupdtool reinstall aa1f48fb-ae6f-5f3f-96eb-1ce0002cdae3` for forcing the failure but that didn't work at all because even the manual mounting needs dbus (i think udisks2) to work.

closed 2026-03-16 by hughsie