fwupd/fwupd #9152: System Firmware update not happening on Dell Vostro 3520

yehudamber, 2025-08-24:
**Describe the bug**
`fwupdmgr` keeps showing there is an update for System Firmware, but when I apply the update and reboot no firmware update happens, and `fwupdmgr get-results` shows the error:
`failed to run update on reboot: expected 1.34.0 and got 1.30.0`.

**Steps to Reproduce**
with power connected:
`fwupdmgr update`
then `y` and again `y`

after reboot:
`fwupdmgr get-results`
then choose 'System Firmware'

**Expected behavior**
System Firmware should be updated to version 1.34.0.

**fwupd version information**
```shell
sudo fwupdtool get-report-metadata
```
```
Loading…                 [********************                   ]22:29:02.457 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-eDP-1/drm_dp_aux0: failed to setup: failed to read device firmware header: failed to write 7 bytes to 15: Input/output error
Loading…                 [*********************                  ]22:29:02.495 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/host1/target1:0:0/1:0:0:0/block/sda: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x20, ascq 0x00
Waiting…                 [*************************************  ]
BatteryLevel:           100
BatteryThreshold:       10
BootTime:               1756065340
CompileVersion(com.hughsie.libjcat): 0.2.0
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.fwupd): 2.0.7
CpuArchitecture:        x86_64
CpuModel:               Intel 12th Gen Intel Core™ i5-1235U
DisplayState:           connected
DistroId:               ubuntu
DistroName:             Ubuntu
DistroPrettyName:       Ubuntu 25.04
DistroVersion:          25.04
FwupdSupported:         True
HostBaseboardManufacturer: Dell Inc.
HostBaseboardProduct:   X8R8F
HostBiosMajorRelease:   01
HostBiosMinorRelease:   1e
HostBiosVendor:         Dell Inc.
HostBiosVersion:        1.30.0
HostEnclosureKind:      a
HostFamily:             Vostro
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 07
HostProduct:            Vostro 3520
HostSku:                B94
HostVendor:             Dell Inc.
KernelName:             Linux
KernelVersion:          6.14.0-28-generic
LidState:               open
PlatformArchitecture:   x86_64
PowerState:             ac-fully-charged
RuntimeVersion(com.hughsie.libjcat): 0.2.0
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.7
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.14.0-28-generic
DeviceId:               ee94f8e437ac3be6602c7f3916939dbb63f7556f
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
[... 859 lines cut ...]
├── initrd.img-6.14.0-28-generic
├── initrd.img.old -> initrd.img-6.14.0-27-generic
├── lost+found  [error opening dir]
├── memtest86+ia32.bin
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── System.map-6.14.0-27-generic
├── System.map-6.14.0-28-generic
├── vmlinuz -> vmlinuz-6.14.0-28-generic
├── vmlinuz-6.14.0-27-generic
├── vmlinuz-6.14.0-28-generic
└── vmlinuz.old -> vmlinuz-6.14.0-27-generic

11 directories, 314 files
```

**Additional questions**

- Operating system and version: Ubuntu 25.04
- Have you tried rebooting?
- Is this a regression?
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? Yes.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

hughsie, 2025-08-26:
> /boot
> ├── config-6.14.0-27-generic
> │   └── EFI
> │       ├── BOOT
> │       └── ubuntu
> │           ├── BOOTX64.CSV
> │           ├── fw
> │           ├── fwupdx64.efi
> │           └── shimx64.efi

I was expecting to see some firmware in the `fw` folder. If you deploy an update again with `fwupdmgr update` and then say "N" to the reboot, what does `sudo tree /boot` and `sudo efivar -l | grep fw` say?

yehudamber, 2025-08-26:
Now it suddenly refuses to update. When I run `fwupdmgr update` I get:
```
Devices with the latest available firmware version:
 • KBG50ZNS256G NVMe KIOXIA 256GB
 • System Firmware
Devices with no available firmware updates: 
 • Touch Controller Sensor
 • SBAT
 • TPM
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
```

I've tried `fwupdmgr clear-results` and even `sudo fwupdtool clear-history`, but nothing helps.

yehudamber, 2025-08-26:
> Also, what's the output of `fwupdmgr get-devices`? Maybe the firmware was auto-demoted from LVFS?

```
Dell Inc. Vostro 3520
│
├─Touch Controller Sensor:
│     Device ID:          c835c98b789afe71db885e3cfd625ad073ac99af
│     Summary:            Touchpad
│     Current version:    0x00051f06
│     Vendor:             Goodix inc. (HIDRAW:0x27C6)
│     GUID:               73c9bbd2-d84d-529b-924f-acd21f991244 ← HIDRAW\VEN_27C6&DEV_0D43
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Unsigned Payload
│                         • Can tag for emulation
│   
├─12th Gen Intel Core™ i5-1235U:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000437
│     Vendor:             Intel
│     GUIDs:              ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│                         d49f8285-de61-572f-ae6d-b83795576416 ← CPUID\PRO_0&FAM_06&MOD_9A&STP_4
│     Device Flags:       • Internal device
│   
├─Alder Lake-UP3 GT2 [UHD Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               83e870f1-00cb-5e3a-bc8c-4a7b8d27efb0 ← PCI\VEN_8086&DEV_4628
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
│   
├─KBG50ZNS256G NVMe KIOXIA 256GB:
│     Device ID:          c430a03ca2a65dfe2412ff950c79c51f6aec1317
│     Summary:            NVM Express solid state drive
│     Current version:    11200109
│     Vendor:             KIOXIA Corporation (PCI:0x1E0F)
│     Serial Number:      836C80TMEY5U
│     GUIDs:              6cefd2de-46a3-5fb7-acb1-25ef14bec6de ← NVME\VEN_1E0F&DEV_000C
│                         52a1c489-1299-5fa5-a97f-384dde8c5fa2 ← NVME\VEN_1E0F&DEV_000C&SUBSYS_1E0F0001
│                         22755fde-6467-597f-a58d-5bc06fd62b36 ← STORAGE-DELL-111195
│                         aac4a8cc-a312-11eb-bcbc-0242ac130002
│                         be08a1d4-26cb-54ed-82fe-4dbebedc27a0 ← KBG50ZNS256G NVMe KIOXIA 256GB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│                         • Can tag for emulation
│   
├─SBAT:
│     Device ID:          6469856584e2f5873b2f148302e46c9313c7d054
│     Summary:            Generation number based revocation mechanism
│     Current version:    1.7.4
│     Vendor:             OS:ubuntu
│     GUID:               9947c120-b97b-52d6-9d83-738a1c342e23 ← UEFI\OS_ubuntu&VAR_SbatLevelRT
│     Device Flags:       • Updatable
│                         • Needs a reboot after installation
│                         • Signed Payload
│   
├─System Firmware:
│ │   Device ID:          ee94f8e437ac3be6602c7f3916939dbb63f7556f
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    1.30.0
│ │   Minimum Version:    1.30.0
│ │   Vendor:             Dell (DMI:Dell Inc.)
│ │   Update State:       Success
[... 50 lines cut ...]
│     Update State:       Success
│     GUID:               e72475a2-b8f0-4217-bbea-3304b187ae54
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─UEFI Device Firmware:
│     Device ID:          d357520c99a2564819b8b609d0f44cb6561b85d1
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    287310089
│     Minimum Version:    287310088
│     Vendor:             DMI:Dell Inc.
│     Update State:       Success
│     GUID:               aac4a8cc-a312-11eb-bcbc-0242ac130002
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
```

hughsie, 2025-08-27:
The update was automatically demoted as it was failing for the majority of people. I assume Dell will re-upload a new update that works better.

[image]