fwupd/fwupd #9156: Upload broken?

quintuple-lained, 2025-08-25:
**Describe the bug**
when choosing to upload the anonymous results to the Linux Vendor Firmware Service and choosing yes after reviewing the data it states "server rejected report: invalid data, expected HsiResult"

**Steps to Reproduce**
run fwupdmgr security
choose Y on upload question
review payload
choose Y again when asked to proceed to upload
get error:
"server rejected report: invalid data, expected HsiResult"

**Expected behavior**
upload report without issue

**fwupd version information**
fwupdmgr --version
compile   info.libusb                   1.0.29
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.13
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.16.0
runtime   org.freedesktop.fwupd         2.0.13

fwupdtool get-report-metadata (this one doesnt seem to exist?)
fwupdtool hwids (neither does this one?)

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

nix, package "fwupd" and enabling "service.fwupd.enable = true;"

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

~/sysflake main* ⇡ 15.19s 01:13:54 ❄️impure ❯ fwupdmgr get-devices --show-all-devices
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Micro-Star International Co., Ltd. MS-7B85
│
├─Unknown Device:
│     Device ID:          57e35fbedc32c4e5e69c55e3d7810f7ae6f1427a
│     Vendor:             PNP:SAM
│     GUID:               5918ec37-541a-5935-b828-0be047f63991 ← DRM\VEN_SAM&DEV_082B
│     Device Flags:       • Can tag for emulation
│   
├─Unknown Device:
│     Device ID:          5f7dee69f5c6dc34f6af5a8ecff65b51ce478670
│     Vendor:             PNP:DEL
│     Serial Number:      1V88T1CD0DDL
│     GUID:               f819f4a5-e272-504e-a4a5-9bb8e64d127f ← DRM\VEN_DEL&DEV_405B
│     Device Flags:       • Can tag for emulation
│   
├─Unknown Device:
│     Device ID:          a6c25fea64794a797f39d32a3277659be669963d
│     Vendor:             PNP:AUS
│     Serial Number:      LBLMQS022989
│     GUID:               856a8d0f-7c44-5c64-b3da-97e058cae4e7 ← DRM\VEN_AUS&DEV_24AC
│     Device Flags:       • Can tag for emulation
│   
├─Unknown Device:
│     Device ID:          a29de8124928f7dc31db026069ac1d601a56056f
│     GUID:               2d9ac22e-0ffe-5691-9e21-f9fecce8761c ← GPIO\ID_AMDIF030:00
│     Device Flags:       • Can tag for emulation
│   
├─AMD Ryzen 9 5900X 12-Core Processor:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Vendor:             Advanced Micro Devices, Inc.
[... 168 lines cut ...]
│                         279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│                         9d131a0c-a606-580f-8eda-80587250b8d6 ← USB\VID_046D&PID_AAAA
│     Device Flags:       • Updatable
│                         • Supported on remote server
│                         • Signed Payload
│                         • Can tag for emulation
│   
└─cMobile Line:
      Device ID:          5cf308a7e1cc37b5f13b56199e68628ae9dd7432
      Summary:            SCSI device
      Current version:    1100
      Vendor:             Intenso (USB:0x090C, SCSI:Intenso)
      GUIDs:              9f56c909-7ac6-5741-81ec-5693e7fe6ef7 ← BLOCK\VEN_090C&DEV_1000
                          cd097405-7ce0-50b1-ab9f-95805cf1081b ← SCSI\VEN_Intenso &DEV_cMobile Line    
                          89c840e7-f1af-520e-a8d4-5c30e6ca3d1c ← SCSI\VEN_Intenso &DEV_cMobile Line    &REV_1100
      Device Flags:       • Can tag for emulation

</details>

**Additional questions**

- Operating system and version:
nixos unstable
- Have you tried rebooting? not yet
- Is this a regression? unsure

hughsie, 2025-08-26:
Can you attach the output of `fwupdmgr security --json` please

quintuple-lained, 2025-08-26:
[output.json](https://github.com/user-attachments/files/21990126/output.json)

hughsie, 2025-08-26:
Sure enough, one of them doesn't have an HsiResult:

      "AppstreamId" : "org.fwupd.hsi.Kernel.Lockdown",
      "HsiLevel" : 0,
      "HsiResultSuccess" : "enabled",

... presumably because it's also `FWUPD_SECURITY_ATTR_FLAG_MISSING_DATA`. PR incoming.

hughsie, 2025-08-29:
Fixed in https://gitlab.com/fwupd/lvfs-website/-/commit/67fb1d791f3e70ea1248972fa2f8965a78772485 instead. Give me ~1h to deploy then please try again.

closed 2025-08-29 by hughsie

quintuple-lained, 2025-09-01:
only got around to it now, yes, upload works now, thank you