fwupd/fwupd #9169: Request to add get usb bcd function from FuDevice

jasonhouang, 2025-08-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

> The bcd information contains usb device release number in binary-coded decimal, I have demand to get the device release number to display like in bootloader.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

> There could have the interface `guint16 fu_device_get_bcd(FuDevice *self)` like `guint16 fu_device_get_pid(FuDevice *self)`

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

> Define command to get device version by each vendor.

**Additional context**
Add any other context or screenshots about the feature request here.

hughsie, 2025-08-28:
Isn't this `fu_usb_device_get_release()`? Internally it's `desc.bcdDevice`

jasonhouang, 2025-08-29:
Yes @hughsie, that's what I was looking for, sorry I didn't find it, I close this issue now.

closed 2025-08-29 by jasonhouang