fwupd/fwupd #9173: Support S3 authentication

arturkow2000, 2025-08-29:
**Is your feature request related to a problem? Please describe.**
[Dasharo](https://www.dasharo.com) has paid subscription feature providing binary firmware packages subscribers. Artifacts are served over S3 which requires non-standard authentication methods.

**Describe the solution you'd like**
Add support for S3 authentication to fwupd.

**Describe alternatives you've considered**
Using proxy to accept secret/access key in HTTP BASIC auth header and converting this SigV4.

hughsie, 2025-08-29:
I'm not against it, but I have no idea how to implement this with libcurl. I'm happy to review a PR.

superm1, 2025-08-29:
Good opportunity to see what copilot does to give us something to review. If it can't figure it out though someone will need to contribute a PR.

superm1, 2025-08-29:
@arturkow2000 can you have a try with the PR mostly written by copilot and see how it works?  If things work we'll give it a closer review to see if it's high enough quality to merge.

superm1, 2025-08-29:
Yeah it's my first time trying it too for a PR, so we're learning together how well it can work 😛