fwupd/fwupd #9203: snap fwupd: intel-cvs: insufficient permission to open the devices

CragW, 2025-09-05:
**Describe the bug**
The confined snap fwupd cannot open/write the sysfs devices.  Issue is gone if the Snap version of fwupd was installed in `devmode`.

**Steps to Reproduce**
```
$ sudo snap install fwupd
$ sudo fwupd.fwupdmgr install ./84_90_42_01.CAB
Writing…                 [                                       ]
failed to write-firmware: failed to open /sys/devices/platform/INTC10B6:00/i2c-15/i2c-INTC10DE:00/cvs_ctrl_data_pre: Permission denied
```

**Expected behavior**
Install firmware updates successfully.

**fwupd version information**

```
$ fwupd.fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.14
compile   com.hughsie.libxmlb           0.3.21
compile   io.snapcraft.fwupd            7567
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.11.0-26-generic
runtime   org.freedesktop.fwupd         2.0.14
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): snap

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupd.fwupdmgr get-devices --show-all-devices
Dell Inc. Dell Pro 14 Plus PB14250
│
├─Camera:
│     Device ID:          bb28dbf199e10aa0349a2498463b32d7dc603ec1
│     Summary:            Computer Vision Sensing Camera
│     Current version:    84.88.42.1
│     Vendor:             USB:0x06CB
│     GUID:               ea64304f-03c1-55bc-9ff4-485ba80f5790 ← I2C\NAME_INTC10DE:00&VID_06CB&PID_0701
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device stages updates
│                         • Device can recover flash failures
│                         • Signed Payload
│                         • Can tag for emulation
│
├─Core™ Ultra 7 268V:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             Intel
│     GUIDs:              7e515fa5-57fb-557c-a30e-b27e38b18a30 ← CPUID\PRO_0&FAM_06&MOD_BD
│                         5f87d1bb-65ae-59e0-82cd-1c846105f12d ← CPUID\PRO_0&FAM_06&MOD_BD&STP_1
│     Device Flags:       • Internal device
│
├─Internal SPI Controller:
│     Device ID:          b04e387fb80d2b91f37a4d0c7b21461c451775e1
│     Summary:            Memory Technology Device
│     Vendor:             Intel (PCI:0x8086)
│     GUIDs:              af783c5c-0390-55a5-ac35-7a6c43c6dbec ← MTD\NAME_BIOS
[... 190 lines cut ...]
│
└─UEFI Device Firmware:
      Device ID:          233d877a828a10c790eac180e3d563c99cbb15a3
      Summary:            UEFI System Resource Table device (updated via NVRAM)
      Current version:    1627857217
      Minimum Version:    1627857217
      Vendor:             DMI:Dell Inc.
      Update State:       Success
      GUID:               d36c38ce-03b0-4cc2-b8fc-275ee12a6199
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
      Device Requests:    • Message

```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2025-09-05:
I think you need to open an issue in the canonical bugtracker really -- there's nothing we can fix here IIUC.

superm1, 2025-09-05:
@valentindavid could you help here?

zyga, 2025-09-05:
The snapd PR is approved and should be released in snapd 2.72

CragW, 2025-09-05:
@zyga, and @bboozzoo, thanks for your time reviewing PR.