fwupd/fwupd #9205: Goodix FP reader 27c6:609c disappeared from `get-devices`

dimitris-personal, 2025-09-05:
**Describe the bug**
The fingerprint reader no longer shows up in `fwupdmgr get-devices`.

**Steps to Reproduce**
```shell
$ fwupdmgr get-devices --show-all-devices |grep -i finger
Idle…: 0%
```

**Expected behavior**
Reader has previously appeared even when its firmware is up to date.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdtool get-report-metadata
fwupdtool hwids
```

Installed via standard `dnf` on Fedora 42 Workstation

<details>

```
$ sudo fwupdtool get-report-metadata
Place your right index finger on the fingerprint reader
Waiting…                 [*************************************  ]21:15:34.047 FuDeviceLocker       failed to close device: cannot close device, refcount already zero
21:15:34.047 FuDeviceLocker       failed to close device: cannot close device, refcount already zero

BatteryLevel:           65
BatteryThreshold:       10
BootTime:               1757101984
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.23
CompileVersion(info.libusb): 1.0.29
CompileVersion(org.freedesktop.Passim): 0.1.10
CompileVersion(org.freedesktop.fwupd): 2.0.14
CpuArchitecture:        x86_64
CpuModel:               AMD Ryzen 7 7840U w/ Radeon 780M Graphics
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 42 (Workstation Edition)
DistroVariant:          workstation
DistroVersion:          42
EfivarsNvramFree:       102136
EfivarsNvramUsed:       87189
FwupdSupported:         True
HostBaseboardManufacturer: Framework
HostBaseboardProduct:   FRANMDCP07
HostBiosMajorRelease:   03
HostBiosMinorRelease:   10
HostBiosVendor:         INSYDE Corp.
HostBiosVersion:        3.16
HostEnclosureKind:      a
HostFamily:             Laptop
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            Laptop 13 (AMD Ryzen 7040Series)
HostSku:                FRANDGCP07
HostVendor:             Framework
KernelCmdline:          amdgpu.dcdebugmask=0x10
KernelName:             Linux
KernelVersion:          6.16.5-200.fc42.x86_64
LidState:               open
PassimDownloadSaving:   0
PlatformArchitecture:   x86_64
PowerState:             ac
RuntimeVersion(com.hughsie.libjcat): 0.2.3
[... 403 lines cut ...]
│       Device ID:        452837bd48f66d3512d9f3c62b4d390b5d0d135d
│       Current version:  F907.14.07
│       Vendor:           Texas Instruments, Inc. (USB:0x0451)
│       GUIDs:            d1e34fb2-db1e-56e4-8a20-43886fe028ff ← USB\VID_047D&PID_80CC&PD_01
│                         5e441847-4783-5bd2-901b-6c3aa3905cfa ← USB\VID_047D&PID_80CC&REV_0714&PD_01
│       Device Flags:     • Updatable
│                         • Signed Payload
│     
└─USB3.0 Hub:
      Device ID:          b46022468451d79b1c327f101949ab71372371b6
      Current version:    38.0
      Vendor:             Intel (USB:0x8087)
      GUID:               ca7b924f-024b-5b8b-a6e4-df758c7cf4ef ← TBT-001b000a
      Device Flags:       • Updatable
                          • Signed Payload
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version: Fedora 42 Workstation
- Have you tried rebooting? N/A (will comment)
- Is this a regression? Yes

hughsie, 2025-09-07:
Can you please get us the full output of `sudo fwupdtool get-devices --plugins goodixmoc -vv` please. Also, if you can downgrade fwupd and find the last version tat it enumerated correctly then that would also help a lot. Thanks.

dimitris-personal, 2025-09-07:
Here's the `sudo fwupdtool get-devices --plugins goodixmoc -vv` output:

[get-devices-output.txt](https://github.com/user-attachments/files/22199016/get-devices-output.txt)

dimitris-personal, 2025-09-07:
Downgraded with `sudo dnf --enablerepo updates-archive downgrade fwupd*`:

```
$ fwupdmgr --version
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.13
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.10
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.23
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.16.5-200.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.13
```

and the reader is visible again:

```
$ fwupdmgr get-devices --show-all-devices |grep -i finger
Idle…: 0%
├─Fingerprint Sensor:
│     Summary:            Match-On-Chip fingerprint sensor
```

Just for completeness, upgraded again:

```
$ fwupdmgr --version
compile   info.libusb                   1.0.29
compile   org.freedesktop.fwupd         2.0.14
compile   com.hughsie.libxmlb           0.3.23
compile   org.freedesktop.Passim        0.1.10
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.23
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.16.5-200.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.14
```

And the fingerprint is again not seen.