fwupd/fwupd #9214: snap fwupd: failed to submit test report

CragW, 2025-09-08:
**Describe the bug**
snap fwupd 2.0.14 or the latest edge both can not submit test report.

**Steps to Reproduce**
```
$ sudo snap install fwupd --channel=latest/stable
$ fwupd.fwupdmgr refresh --force
$ fwupd.fwupdmgr install {system-firmware}
reboot

$ fwupd.fwupdmgr report-history
No reports require uploading
```

The history does exist, but no `RemoteID` was found.
```
$ fwupd.fwupdmgr report-history -vv
(pkttyagent:6212): GLib-GIO-DEBUG: 22:29:55.540: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(/snap/fwupd/7567/bin/fwupdmgr:6177): GLib-GIO-DEBUG: 22:29:55.556: _g_io_module_get_default: Found default implementation dummy (GDummyProxyResolver) for ‘gio-proxy-resolver’
(/snap/fwupd/7567/bin/fwupdmgr:6177): GLib-GIO-DEBUG: 22:29:55.557: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(/snap/fwupd/7567/bin/fwupdmgr:6177): Fwupd-DEBUG: 22:29:55.558: Emitting ::status-changed() [idle]
(/snap/fwupd/7567/bin/fwupdmgr:6177): FuMain-DEBUG: 22:29:55.587: 1 devices with history
(/snap/fwupd/7567/bin/fwupdmgr:6177): Fwupd-DEBUG: 22:29:55.593: da09124945f291ef6e9e9a50ef324a067716a70a has no RemoteID
(/snap/fwupd/7567/bin/fwupdmgr:6177): Fwupd-DEBUG: 22:29:55.595: da09124945f291ef6e9e9a50ef324a067716a70a has no RemoteID
(/snap/fwupd/7567/bin/fwupdmgr:6177): Fwupd-DEBUG: 22:29:55.597: da09124945f291ef6e9e9a50ef324a067716a70a has no RemoteID
No reports require uploading
```

```
$ fwupd.fwupdmgr get-history
Dell Inc. Latitude 7350
│
└─System Firmware:
  │   Device ID:          da09124945f291ef6e9e9a50ef324a067716a70a
  │   Previous version:   1.17.0
  │   Update State:       Success
  │   Last modified:      2025-09-08 13:49:48
  │   GUID:               d42287a8-51f4-4b9a-8f7c-bd2c6d25d7a7
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─(null) Update:
        New version:      1.17.0
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        The vendor did not supply any release notes.
```

**Expected behavior**
Upload the report successfully.

**fwupd version information**

<details>

<summary>fwupd.fwupdtool get-report-metadata</summary>

```shell
Loading…                 [******************                     ]14:23:38.257 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:07.0/0000:02:00.0/0000:03:02.0/0000:04:00.0/usb5/5-2/5-2.3/5-2.3.3/5-2.3.3:1.2/0003:046D:C52B.000C/hidraw/hidraw9: failed to subclass open: failed to open /dev/hidraw9: Operation not permitted
Waiting…                 [*************************************  ]
BatteryLevel:           100
BatteryThreshold:       10
BootTime:               1757339897
CompileVersion(com.hughsie.libjcat): 0.2.0
CompileVersion(com.hughsie.libxmlb): 0.3.21
[... 638 lines cut ...]
│     Vendor:             THUNDERBOLT:0x8087
│     GUIDs:              74fb1b87-53e6-5115-bd02-a9fefd06ed9a ← THUNDERBOLT\VEN_8087&DEV_0D9C
│                         dac6f95d-7c5b-52b5-8c79-a0436a5ef784 ← TBT-80870d9c-retimer0-0:1.1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Device stages updates
│                         • Unsigned Payload
│                         • Can tag for emulation
│   
└─YJ3JK:
      Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
      Vendor:             PNP:AUO
      GUID:               340848af-fb81-553f-a1d4-9b81b5539548 ← DRM\VEN_AUO&DEV_CFA4
      Device Flags:       • Internal device
                          • Can tag for emulation

```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes
- Is this a regression? Yes

hughsie, 2025-09-08:
What does `fwupd -vv` say post install post reboot?

superm1, 2025-09-08:
We had something really similar to this outside of snap too.  I don't remember the fix though.  Can you repro without snap too in a venv?

hughsie, 2025-09-08:
Ya, --force shouldn't have been required. None of those releases have a RemoteID set and `--force` just says _fsck it, lets assume the LVFS_

hughsie, 2025-09-08:
Ohh hang on -- does it work without `--force` if you don't use reinstall -- i.e. if you do a downgrade or upgrade?

hughsie, 2025-09-09:
That must have taken some time, so thanks. I'll investigate now.