fwupd/fwupd #9239: LVFS CDN serves firmware-08239-stable.xml.gz with wrong checksum (c46c9f47… vs expected 4f0ea29c…)

hkumarsaikia, 2025-09-13:
Description:
On Ubuntu 22.04 with fwupd 1.7.9, refreshing LVFS metadata fails with a checksum error.

The file https://cdn.fwupd.org/downloads/firmware-08239-stable.xml.gz consistently downloads with SHA1:

c46c9f4788df09e6bb5829d5da4d4f4b64ed63a6  firmware-08239-stable.xml.gz

But fwupd expects:

4f0ea29c9d4f26f5a3536aa0c1c9296d558a53b6

As a result:

$ sudo fwupdmgr --force refresh
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 4f0ea29c...

Steps to Reproduce:

Run on Ubuntu 22.04.4 LTS, fwupd 1.7.9 (/usr/bin/fwupdmgr)

Clear caches:

sudo rm -rf /var/cache/fwupd/* /var/lib/fwupd/* ~/.cache/fwupd/*

Run:

curl -L -o fwupd_meta.gz https://cdn.fwupd.org/downloads/firmware-08239-stable.xml.gz
sha1sum fwupd_meta.gz

Observe mismatch with expected checksum.

What I tried:

Tested on normal ISP connection and mobile hotspot → same wrong checksum.

Used curl with cache-buster query and Cache-Control: no-cache header → still wrong checksum.

So this seems to be a stale or corrupted file at the CDN origin.

System Info:

$ apt policy fwupd
fwupd:
  Installed: 1.7.9-1~22.04.3
  Candidate: 1.7.9-1~22.04.3

Expected Behavior:
The CDN should serve firmware-08239-stable.xml.gz matching checksum 4f0ea29c9d4f26f5a3536aa0c1c9296d558a53b6.

Actual Behavior:
CDN serves a different file (c46c9f47…) and fwupd refuses to refresh metadata.

hughsie, 2025-09-13:
You need to file this issue with Ubuntu, 1.7.x is long EOL. The 1.7.x release assumes the metadata is less than 1MB in size, and it's now more than that. There are only so many workarounds we can do.

hughsie, 2025-09-13:
I've merged https://gitlab.com/fwupd/lvfs-website/-/commit/051df8ea49610b6611aff3415de6117d3e57bddd which is a bit more drastic. It might mean that anyone running 1.7.x misses a security update, but there's not much we can do about that.

closed 2025-09-13 by hughsie