fwupd/fwupd #9240: Fail to upgrade UEFI dbx: failed to load /boot/efi/EFI/Boot/BootX64.efi

jlory, 2025-09-13:
**Describe the bug**

I'm trying to update the UEFI dbx on Ubuntu 22.04 / Dell XPS 9305, everything is up to date ( dell firmware / ubuntu / fwupdmgr from snap )

I get that error after running ```sudo fwupdmgr upgrade --force```:

```
══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 20241101 to 20250507?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ Some insecure versions of BiosFlashShell and Dtbios by DT Research Inc were  ║
║ added, due to a security vulnerability that allowed an attacker to bypass    ║
║ UEFI Secure Boot.                                                            ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Verifying…               [                                       ]
Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/Boot/BootX64.efi: Error opening file /boot/efi/EFI/Boot/BootX64.efi: Permission denied
```

**Steps to Reproduce**

```
julien@xps:~$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [   -                                   ]
Successfully downloaded new metadata: Updates have been published for 4 of 9 local devices
julien@xps:~$ sudo fwupdmgr upgrade --force
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 20241101 to 20250507?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ Some insecure versions of BiosFlashShell and Dtbios by DT Research Inc were  ║
║ added, due to a security vulnerability that allowed an attacker to bypass    ║
║ UEFI Secure Boot.                                                            ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Verifying…               [                                       ]
Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/Boot/BootX64.efi: Error opening file /boot/efi/EFI/Boot/BootX64.efi: Permission denied
```

**Expected behavior**
It should have worked, it worked a month ago.

**fwupd version information**
```
julien@xps:~$ sudo fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.14
compile   com.hughsie.libxmlb           0.3.21
compile   io.snapcraft.fwupd            7567
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.8.0-79-generic
runtime   org.freedesktop.fwupd         2.0.14
```

```shell
julien@xps:~$ sudo fwupd.fwupdtool get-report-metadata
Waiting…                 [*************************************  ]
BatteryLevel:           96
BatteryThreshold:       10
[... 737 lines cut ...]
├── initrd.img-6.8.0-78-generic
├── initrd.img-6.8.0-79-generic
├── initrd.img.old -> initrd.img-6.8.0-78-generic
├── lost+found
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-6.8.0-78-generic
├── System.map-6.8.0-79-generic
├── vmlinuz -> vmlinuz-6.8.0-79-generic
├── vmlinuz-6.8.0-78-generic
├── vmlinuz-6.8.0-79-generic
└── vmlinuz.old -> vmlinuz-6.8.0-78-generic

14 directories, 316 files
```

**Additional questions**

- Operating system and version: Ubuntu 22.04.5 LTS
- Have you tried rebooting? Yes
- Is this a regression? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2025-09-13:
You should probably file this with Canonical -- it's probably a snap issue.

t3kn, 2025-10-23:
I have the same problem with same operating system but different upgrade:

```
══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 20250507 to 20250902 ?                                 ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ Some insecure versions of the IGEL bootloader were added, due to a security  ║
║ vulnerability that allowed an attacker to bypass UEFI Secure Boot.           ║
║                                                                              ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Verifying…               [                                       ]
Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/Boot/BootX64.efi: Error opening file /boot/efi/EFI/Boot/BootX64.efi: Permission denied
```  
and different fwupd version:

```
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.16
compile   com.hughsie.libxmlb           0.3.21
compile   io.snapcraft.fwupd            7611
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.8.0-85-generic
runtime   org.freedesktop.fwupd         2.0.16
``` 

@jlory did you find a solution ?

jlory, 2025-11-04:
> You should probably file this with Canonical -- it's probably a snap issue.

Sorry, late to reply but can you give more details on why do you think it's a snap issue?

>@jlory did you find a solution ?

Not yet, but I'm still looking into it.

dinobagio, 2025-11-06:
Hi @hughsie 
I was facing the exact same issue.
I read about the thought relation to snap so I removed fwupd from snap and installed it with apt.
Same result.
Or quite the same, before it was proposing to update and then fail with:
```Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/ubuntu/shimx64.efi: Error opening file /boot/efi/EFI/ubuntu/shimx64.efi: Permission denied```
Now it says:
```WARNING: UEFI ESP partition not detected or configured```
and I can see in journalctl
```Nov 06 13:34:12 nuc fwupd[1588138]: 13:34:12.781 FuPluginUefiCapsule  cannot find default ESP: user specified ESP /boot/efi/ not found```
```
$ sudo fwupdmgr get-devices
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Intel(R) Client Systems NUC10i7FNH
[...]
├─System Firmware:
│ │   Device ID:          8e9e912973de0e703b990d27e9d1c667f7c7a118
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    66
│ │   Minimum Version:    66
│ │   Vendor:             Intel(R) Client Systems (DMI:Intel Corp.)
│ │   Update State:       Success
│ │   Update Error:       Not updatable as UEFI ESP partition not detected
│ │   GUID:               f4cd40f8-4947-4069-bd4d-d5987adefc79
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │   Device Requests:    • Message
```

yet,

```
$ sudo /usr/bin/fwupdtool esp-list
Loading…                 [************************************** ]
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/efi/EFI/ubuntu/grubx64.efi
/boot/efi/EFI/ubuntu/shimx64.efi
/boot/efi/EFI/ubuntu/mmx64.efi
/boot/efi/EFI/ubuntu/BOOTX64.CSV
/boot/efi/EFI/ubuntu/grub.cfg
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/BOOT/mmx64.efi
```

And here is my info
```
$ efibootmgr -v
BootCurrent: 0000
Timeout: 1 seconds
BootOrder: 0000,0002,0003
Boot0000* Ubuntu        HD(1,GPT,a251f581-910b-4115-ba0a-3ec7e741fb03,0x800,0x100000)/File(\EFI\UBUNTU\SHIMX64.EFI)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 00 10 00 00 00 00 00 81 f5 51 a2 0b 91 15 41 ba 0a 3e c7 e7 41 fb 03 02 02 / 04 04 34 00 5c 00 45 00 46 00 49 00 5c 00 55 00 42 00 55 00 4e 00 54 00 55 00 5c 00 53 00 48 00 49 00 4d 00 58 00 36 00 34 00 2e 00 45 00 46 00 49 00 00 00 / 7f ff 04 00
Boot0002* UEFI: PXE IPv4 Intel(R) Ethernet Connection (10) I219-V       PciRoot(0x0)/Pci(0x1f,0x6)/MAC(1c697aaae11e,0)/IPv4(0.0.0.00.0.0.0,0,0)0000424f
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 06 1f / 03 0b 25 00 1c 69 7a aa e1 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f
Boot0003* UEFI: PXE IPv6 Intel(R) Ethernet Connection (10) I219-V       PciRoot(0x0)/Pci(0x1f,0x6)/MAC(1c697aaae11e,0)/IPv6([::]:<->[::]:,0,0)0000424f
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 06 1f / 03 0b 25 00 1c 69 7a aa e1 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0d 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f
```
```
$ sudo fwupdtool get-report-metadata
Loading…                 [****                                   ]13:38:06.026 FuPluginUefiCapsule  cannot find default ESP: user specified ESP /boot/efi/ not found
Loading…                 [*************************************  ]
WARNING: UEFI ESP partition not detected or configured
[... 41 lines cut ...]
RuntimeVersion(org.freedesktop.gusb):0.4.8
RuntimeVersion(org.kernel):6.8.0-87-generic
DeviceId:               8e9e912973de0e703b990d27e9d1c667f7c7a118
  pre:
      CapsuleApplyMethod:nvram
      EspPath:
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x42
DeviceId:               362301da643102b9f38477387e2193e57abaa590
  pre:
      EfivarNvramUsed:  77774
      LinuxLockdown:    integrity
      Pcr0_SHA256:      fe1d48a9b3400c1025fe2f662f5fafcf57d0af625632b5045387b6a50975b6af
      TpmEventLog:      0x00000007 7022cba477314f53b645370963385b80bdc6261a26fd56d63c7cb1b6316eca56 [Qm9vdCBHdWFyZCBNZWFzdXJlZCBTLUNSVE0A]
      0x00000008 d4720b4009438213b803568017f903093f6bea8ab47d283db32b6eabedbbf155 [HvtrVAwdVUCkrU70vxe4Og==]
      0x00000001 1db27d83c40fc2845737be67f7814052b3f197dd22087cf33f5bf4dd87c09736 [AADX/wAAAAAAAAkAAAAAAA==]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: fe1d48a9b3400c1025fe2f662f5fafcf57d0af625632b5045387b6a50975b6af
      TpmFamily:        2.0
      EfivarNvramUsed:  77774
      SecureBoot:       Enabled
      UEFIUXCapsule:    Disabled
```

dinobagio, 2025-11-11:
> I suggest debugging your issue on the apt version - it will be easier by removing snapd from the picture.  Please drop your configuration file change.  If you can still reproduce the issue:
> 
> 1. Do you have multiple ESPs on your disk?  
> 2. Or multiple disks that could contain an ESP? 
> 3.  Attach `udisksctl dump` output
> 4. Attach  `sudo fwupdtool esp-list -v` output.  

Just a quick answer before proceeding all you asked for.

Yes I realized afterwards I had another ESP, because I'm using incus (ex: lxd) with zfs storage backend and it appears it creates a zd0 block device with an ESP to provide to uefi VM.

```/dev/zd0p1   2048   206847   204800  100M EFI System```

Might it affect fwupd ?

dinobagio, 2025-11-11:
SNAP seems to respond to this need then:
```
$ sudo snap install fwupd
fwupd 2.0.16 from Richard Hughes installed
```
I attach both files again but with snap

[esp-list.log](https://github.com/user-attachments/files/23481439/esp-list.log)
[udisksctl.log](https://github.com/user-attachments/files/23481440/udisksctl.log)

EDIT: it updates OK now with snap 😲 

```
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database e14c88dc
│       Current version:  20250902
│       Minimum Version:  20250902
│       Vendor:           UEFI:Microsoft
│       Install Duration: 1 second
│       GUIDs:            f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│                         d07ff664-b0e1-5f4e-a723-d7fbcbfcb94f ← UEFI\CRT_3CD3F0309EDAE228767A976DD40D9F4AFFC4FBD5218F2E8CC3C9DD97E8AC6F9D&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│                         • Can tag for emulation
```
So weird.
might it be because of rm -rf /etc/fwupd ?

latest status
```
Devices with the latest available firmware version:
 • UEFI CA
 • UEFI dbx
Devices with no available firmware updates:
 • CT1000P2SSD8
 • KEK CA
 • Option ROM UEFI CA
 • SBAT
 • System Firmware
 • Thunderbolt host controller
 • Windows Production PCA
```

jlory, 2025-11-12:
```
snap --version
snap          2.72
snapd         2.72
series        16
ubuntu        22.04
kernel        6.8.0-87-generic
architecture  amd64
```

bboozzoo, 2025-11-12:
AFAICT following https://uefi.org/specs/UEFI/2.11/03_Boot_Manager.html#removable-media-boot-behavior and https://uefi.org/specs/UEFI/2.11/13_Protocols_Media_Access.html#directory-structure the convention is `\EFI\BOOT\BOOT{machine type short-name}.EFI`, with that in mind `EFI/Boot/BootX64.efi` does not match. Though that other page mentions that at least from the load perspective the names are case-insensitive. 

@jlory do you know what created the  `EFI/Boot/BootX64.efi` path in your system?

bboozzoo, 2025-11-12:
I've proposed https://github.com/canonical/snapd/pull/16244 which grants read access to /boot/efi/EFI and everything under that path. I think security reviewers should be fine with this.

bboozzoo, 2025-11-13:
Funny, didn't expect the issue to be closed by a commit in another project.

Anyways, I've landed a tweak in snapd. It should be available in snapd edge later today or tomorrow assuming Launchpad does not fail to build to build the snap again.@jlory if you can, run `snap refresh --edge snapd` tomorrow and see if that resolves the issue.  The version in edge should look like ` 2.72+g242.<sha>` (the number following `+g` could be higher, but looking at the current git log I'm expecting it to be at least 242).