fwupd/fwupd #9246: Thinkpad T580 - fwupdmgr attemps update, and looses the device during the update

nFishr, 2025-09-14:
**Describe the bug**
Running `fwupdmgr update`  shows an update for the Thunderbolt Host Controller of my Lenovo T580 laptop.
After downloading the update it prints that it has not found any supported devices: `No supported devices found`. The update was not installed.

Background:
My Thinkpad T580 (Type 20L9) Thunderbolt host controller has currently version 14.00. It should be [at least 20.00](https://pcsupport.lenovo.com/de/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t580-type-20l9-20la/solutions/ht508988) to prevent loosing that controller.

End of the output of `sudo fwupdtool update --verbose --verbose`
```shell
10:21:28.353 FuEngine             ignoring 0.1.32 < 0.1.43
10:21:28.353 FuEngine             ignoring 0.1.30 < 0.1.43
10:21:28.353 FuEngine             ignoring 0.1.29 < 0.1.43
10:21:28.353 FuMain               current version is 0.1.43: 0.1.28=older, 0.1.27=older, 0.1.21=older, 0.1.20=older, 0.1.19=older, 0.1.18=older, 0.1.16=older, 0.1.15=older, 0.1.14=older, 0.1.37=older, 0.1.36=older, 0.1.34=older, 0.1.33=older, 0.1.32=older, 0.1.30=older, 0.1.29=older
10:21:28.353 FuEngine             72021ca5-cd76-5b37-85ca-d39e7184bf77 was not found: indexed string '72021ca5-cd76-5b37-85ca-d39e7184bf77' was unfound
10:21:28.356 FuEngine             5bed8b5c-e5c2-547b-8806-6603930184ee matched 1 components
10:21:28.356 FuRelease            remote lvfs has kind=download and so marking as trusted
10:21:28.357 FuRelease            remote lvfs has kind=download and so marking as trusted
10:21:28.357 FuRelease            remote lvfs has kind=download and so marking as trusted
10:21:28.357 FuRelease            remote lvfs has kind=download and so marking as trusted
10:21:28.358 FuRelease            remote lvfs has kind=download and so marking as trusted
10:21:28.358 FuEngine             5bed8b5c-e5c2-547b-8806-6603930184ee matched 5 releases
10:21:28.358 FuEngine             f69aca31-6f9e-5b23-93ec-9683b02d66eb was not found: indexed string 'f69aca31-6f9e-5b23-93ec-9683b02d66eb' was unfound

╔══════════════════════════════════════════════════════════════════════════════╗
║ Thunderbolt host controller von 14.00 auf 20.00 aktualisieren?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ N27TF20W update PD                                                           ║
║                                                                              ║
║ • Patch bundle upgrade to F407.06.29                                         ║
║                                                                              ║
║ Bug Fixes Key 	Summary                                                       ║
║                                                                              ║
║ TPS65982-14138 	988CE/SNK was occasionally entering an incorrect state when  ║
║ the far-end/SRC prevented the SNK from doing a transmission by holding the   ║
║ CC line at SinkTxNG state, thus preventing the device from resending         ║
║ pending PD commands on the CC line.                                          ║
║                                                                              ║
║ TPS65982-14485 	988CE/SNK was occasionally setting the Overcurrent flag in   ║
║ register 0x5F (Data Status Register) when FRS was enabled and the far end    ║
║ was a source capable dongle/dock, due to a spurious interrupt upon           ║
║ disconnect.                                                                  ║
║                                                                              ║
║ TPS65982-4314 	VCONN acceptable range for cables is 4.75V-5.5V               ║
║                                                                              ║
║ Thunderbolt host controller und alle angeschlossenen Geräte sind während     ║
║ der Aktualisierung möglicherweise nicht nutzbar.                             ║
╚══════════════════════════════════════════════════════════════════════════════╝
Operation durchführen? [Y|n]: Y
10:21:29.770 FuProgress           raw timing data was { 0,001, 0,000, 0,001, 0,008, 0,064 } -- steps were set as [ 1 1 3 1 94 ] but should have been [ 2 0 1 10 86 ] at ../libfwupdplugin/fu-context.c:1153
10:21:29.771 FuProgress           raw timing data was { 0,001, 0,001, 0,013, 0,001 } -- steps were set as [ 1 72 26 1 ] but should have been [ 8 4 80 9 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1083
Wird geschrieben …       [*******************                    ]10:21:29.777 Fwupd                downloading https://fwupd.org/downloads/49d43a0b6da0905014b62523bad96844da42865dc82ff2049f775c7b2931617c-Lenovo-ThinkPad-P52s-Thunderbolt-Firmware-N27TF20W-Secured.cab
10:21:29.777 pxbackend            px_manager_get_proxies_sync: url=https://fwupd.org/downloads/49d43a0b6da0905014b62523bad96844da42865dc82ff2049f775c7b2931617c-Lenovo-ThinkPad-P52s-Thunderbolt-Firmware-N27TF20W-Secured.cab online=1
10:21:29.777 pxbackend            px_manager_get_proxies_sync: Proxy[0] = direct://
10:21:29.777 pxbackend            px_manager_get_proxies_sync: url=none://fwupd.org:65535 online=1
10:21:29.777 pxbackend            px_manager_get_proxies_sync: Proxy[0] = direct://
10:21:29.802 Fwupd                Emitting ::status-changed() [downloading]
Herunterladen …          [               /                       ]10:21:30.443 Fwupd                download progress: 100%
Herunterladen …          [***************************************]10:21:30.725 Fwupd                download progress: 6%
Herunterladen …          [**                                     ]10:21:30.725 Fwupd                download progress: 10%
Herunterladen …          [***                                    ]10:21:30.727 Fwupd                download progress: 16%
Herunterladen …          [******                                 ]10:21:30.727 Fwupd                download progress: 24%
Herunterladen …          [*********                              ]10:21:30.727 Fwupd                download progress: 25%
Herunterladen …          [*********                              ]10:21:30.735 Fwupd                download progress: 31%
Herunterladen …          [************                           ]10:21:30.736 Fwupd                download progress: 37%
Herunterladen …          [**************                         ]10:21:30.738 Fwupd                download progress: 44%
10:21:30.738 Fwupd                download progress: 45%
Herunterladen …          [*****************                      ]10:21:30.745 Fwupd                download progress: 49%
Herunterladen …          [*******************                    ]10:21:30.745 Fwupd                download progress: 55%
Herunterladen …          [*********************                  ]10:21:30.747 Fwupd                download progress: 61%
Herunterladen …          [***********************                ]10:21:30.747 Fwupd                download progress: 69%
[... 655 lines cut ...]
      GUID:               c271910d-0695-4b4b-bccf-83550d87a59a
      Geräte-Bitschalter: • Internes Gerät
                          • Aktualisierbar
                          • System benötigt externe Stromquelle
                          • Benötigt einen Neustart nach der Installation
                          • Das Gerät ist während der Dauer der Aktualisierung nutzbar
      Geräteanfragen:     • Mitteilung
```

</details>

**Additional questions**

- Operating system and version
Fedora42 Workstation, up to date

- Have you tried rebooting?
yes. Also loaded BIOS Optimized defaults, but no change.

- Is this a regression?
I think so:
- #6643 seems to have the same problem for T580
- #5975 had also a Thunderbolt problem, but for T480 (same generation, smaller one)
 
Many thanks in advance!

nFishr, 2025-09-14:
Also maybe worth mentioning: 

- Fedora42 using two days ago -> fwupd Version 2.0.14 -> same problem as in 2.0.16
- yesterday update -> fwupd Version 2.0.15 -> no update shown at all
- today -> fwupd Version 2.0.16 -> same result as in 2.0.14

nFishr, 2025-09-14:
And LVFS does not have the latest update of the thunderbolt controller firmware for T580:

- Currently on LVFS: [N27TF18W](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN27TF_AssistMode.firmware)
- But on Lenovo Site: [N27TF22W](https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t580-type-20l9-20la/downloads/DS502888)

In contrast T480 has newest update on LVFS: [N24TF23W_Secured](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24TF.firmware)

nFishr, 2025-09-14:
I also tried latest Linux Mint 22.2 ISO, but same result. Version used there:

```shell
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         1.9.30
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.14.0-29-generic
runtime   org.freedesktop.fwupd         1.9.30
``

hughsie, 2025-09-16:
> downloading https://fwupd.org/downloads/49d43a0b6da0905014b62523bad96844da42865dc82ff2049f775c7b2931617c-Lenovo-ThinkPad-P52s-Thunderbolt-Firmware-N27TF20W-Secured.cab

For some reason your T580 is downloading the update file for the P52s! It's a good job fwupd refuses to install the update. 

@mrhpearson for that firmware I see:

[image]

It looks like the same AppStream ID was used for both updates -- hence why fwupd is confused why there are 2 different `20.0` releases in the same stream. I think because the update was uploaded so long ago we were not doing the kinds of dedupe checks we now do. Can you tell me what the codes are supposed to be for each update please? e.g. N27TF20W for P52s and then what should be the code for the T580. etc. Thanks.

mrhpearson, 2025-09-17:
Hi @hughsie ,

The T580 and P52s share the same BIOS so I think that's why some of the confusion (they share the same N2F identifier). 

Won't the different GUIDs differentiate between the P52s TBT FW update (https://fwupd.org/lvfs/components/7481) and the T580 update (https://fwupd.org/lvfs/components/7478) even if they have the same AppID?

Mark

hughsie, 2025-09-17:
> Won't the different GUIDs differentiate between the P52s TBT FW update (https://fwupd.org/lvfs/components/7481) and the T580 update (https://fwupd.org/lvfs/components/7478) even if they have the same AppID

Nope; the LVFS will mash them together and merge the two GUIDs into one component. Do you want me to split them up with something like com.lenovo.ThinkPadN27TF.thunderbolt.p52s.firmware perhaps?

mrhpearson, 2025-09-17:
Ah.
Yeah - I was going to ask the FW team to do that, but if you're able to, then that would be great.
I think I can demote/change/promote myself too - just haven't actually done the steps before.

Mark

mrhpearson, 2025-09-22:
Best to wait until we hear back from the FW team on this one. HW is the similar for T580 and P52s, but the TBT controller seems to have separate updates so I'm hoping we get guidance from them.

verpejas, 2025-10-02:
You may have been hit by the infamouse Thinkpad tunderbolt eeprom issue, and thats why it refuses to update? I just done a batch of ~9 X1 Yoga G3's, replacing the thunderbolt eeprom chips on all of them. On one laptop i was able to attempt the update using fwupd but it would loose the device, so maybe it is a hardware issue?

nFishr, 2025-10-03:
Thanks for providing further information 👍

> You may have been hit by the infamous Thinkpad Thunderbolt eeprom issue, and that's why it refuses to update? I just done a batch of ~9 X1 Yoga G3's, replacing the thunderbolt eeprom chips on all of them. On one laptop i was able to attempt the update using fwupd but it would loose the device, so maybe it is a hardware issue?

Currently, my laptop works as expected: charging works, using USB devices with expected high speeds works, on both USB-C ports. I would like to keep it this way, hence the interest in updating the firmware.

As of now my summary is: fwupd evolved to contain more safety checks (a good thing). T580 and P52s are different laptops (different UIDs), but share same (exactly same?) components, and fwupd wants to load a P52s firmware for a T580 - that checks are triggered. Currently the next step is waiting for internal ticket LO-4003.