fwupd/fwupd #9261: fwupdmgr conflicting with gs_usb CAN driver

goberhammer, 2025-09-16:
**Describe the bug**
When plugging in an USB-CAN device managed by the gs_usb driver, fwupdmgr seems to interfere with the device initialization (or maybe runtime firmware upload?). De device will be listed by`ip link`, but trying to configure it via e.g. `ip link set can0 type can bitrate 1000000` returns an error like `RTNETLINK answers: Device or resource busy`.

The behavior is a bit random, sometimes replugging the device several times makes it work.
Disabling fwupdmgr with `sudo systemctl stop fwupd` before plugging the device prevents the problem.

When the problem occours, this is the dmesg log:

```
[32092.937322] usb 1-1.1.1.2: new full-speed USB device number 18 using xhci_hcd
[32093.023136] usb 1-1.1.1.2: New USB device found, idVendor=1d50, idProduct=606f, bcdDevice= 0.00
[32093.023140] usb 1-1.1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[32093.023143] usb 1-1.1.1.2: Product: USB2CAN v3.2
[32093.023145] usb 1-1.1.1.2: Manufacturer: InnoMaker
[32093.023146] usb 1-1.1.1.2: SerialNumber: [REDACTED]
[32093.029723] gs_usb 1-1.1.1.2:1.0: Configuring for 1 interfaces
[32093.043959] usb 1-1.1.1.2: usbfs: process 18088 (fwupd) did not claim interface 1 before use
```

**Steps to Reproduce**
Insert a gs_usb managed USB-CAN adapter

**Expected behavior**
Adapter should be configurable via ip link, but is not responding

**fwupd version information**

```
sudo fwupdtool get-report-metadata
Loading…                 [*********************                  ]17:14:30.124 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/drm_dp_aux0: failed to setup: failed to read device firmware header: checksum mismatch in device response header (header specified: 0x0, actual: 0xAB)
Loading…                 [*********************                  ]17:14:30.125 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
17:14:30.125 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Waiting…                 [*************************************  ]
BatteryLevel:           100
BatteryThreshold:       10
BootTime:               1758010017
CompileVersion(com.hughsie.libjcat): 0.2.0
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.fwupd): 2.0.7
CpuArchitecture:        x86_64
CpuModel:               Intel Core™ i7-9750H CPU @ 2.60GHz
DisplayState:           connected
DistroId:               ubuntu
DistroName:             Ubuntu
DistroPrettyName:       Ubuntu 25.04
DistroVersion:          25.04
FwupdSupported:         True
HostBaseboardManufacturer: Dell Inc.
HostBaseboardProduct:   XMC3F
HostBiosMajorRelease:   01
HostBiosMinorRelease:   2a
HostBiosVendor:         Dell Inc.
HostBiosVersion:        1.42.1
HostEnclosureKind:      a
HostFamily:             Precision
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            Precision 7540
HostSku:                926
HostVendor:             Dell Inc.
KernelCmdline:          spl.spl_hostid=0x00bab10c kvm.enable_virt_at_load=0
KernelName:             Linux
KernelVersion:          6.14.0-29-generic
LidState:               open
PlatformArchitecture:   x86_64
PowerState:             ac-fully-charged
RuntimeVersion(com.hughsie.libjcat): 0.2.0
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.7
[... 214 lines cut ...]
│                         • Can tag for emulation
│   
└─Unifying Receiver:
      Device ID:          fdb5c2807386a6014025a352dec974ac08d4ab84
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.03_B0009
      Vendor:             Logitech, Inc. (USB:0x046D, HIDRAW:0x046D)
      Install Duration:   18 seconds
      GUIDs:              aa995882-da42-574a-8338-c8dfdca447e3 ← UFY\VID_046D&PID_C52B
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          cc4cbfa9-bf9d-540b-b92b-172ce31013c1 ← USB\VID_046D&PID_AAAC
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Signed Payload
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 25.04 Plucky
- Have you tried rebooting? Yes
- Is this a regression? Don't know

hughsie, 2025-09-16:
So something like https://www.inno-maker.com/product/usb-can/ perhaps? I'm not familiar with how gs_usb works, so any information you can share would be useful. The dmesg log might also be a clue. Can you attach the **full** output of `fwupdtool get-devices -vv` to this issue please -- then we can work out what plugin is trying to configure the device.

goberhammer, 2025-09-17:
Thank for the quick reply :-)

The adapter is exactly the one you mentioned.
Attached  the requested information. Please feel free to ask any question, I'm glad to help in any way possible.

[devices.txt](https://github.com/user-attachments/files/22385929/devices.txt)

hughsie, 2025-09-17:
I think https://github.com/fwupd/fwupd/pull/9263 will help (at least with the dmesg warning) but the issue is that the device also exports a valid DFU interface -- so fwupd is actually doing the right thing. Is there any chance you could try that PR and see if it fixes things?